hzeller/rpi-rgb-led-matrix #419: Display two images with a command at a time

beikimajid, 2017-11-01:
This is common in led-matrix borad.You can split the screen into multiple sections and display in each part of the image.
How to display two images or more together with [led-image-viewer] & -L parameter such that Both images appear at the same time one at the top and one at the bottom of the screen.
thanks

hzeller, 2017-11-01:
You have to write your own program that does that or modify the led-image-viewer.

closed 2017-11-01 by hzeller

beikimajid, 2017-11-02:
Hi hzeller
How to do this?
How to split the screen and use it as argument?
I will be grateful if guide

hzeller, 2017-11-02:
Internally, you have a canvas, so you can display anything you want on it.
So you can load an image and show it on the top left corner and then load another image but display it starting from the middle.
So you don't split the screen, you just use different parts of the screen to display different things in your program.

The library provides everything you need, but you need to get into the programming part. If you just start out programming, you might find it easier to start doing that in python (there are samples in bindings/python/samples/).