hzeller/rpi-rgb-led-matrix #422: 1/2 scan

beikimajid, 2017-11-06:
Hi
Has anyone worked with P10 16x32 1/2 scan and wrote transformer?

MaitreDede, 2017-11-06:
Hi,
Are you sure it is a 1/2 scan ? I have a panel with same pin layout, and it is a 1/32 scan...

beikimajid, 2017-11-07:
Yes, it is 1/2 scan & full color.I write transformer for 1/4 scan and it works well but 1/2 scan not work.
The problem is that when the power-on command is given to one led, it turns on multiple led.
Please help me.

MaitreDede, 2017-11-07:
Can you link the website you bought it ?
I have also a kind of panel that seams to work like yours, so I would like to check if I can compare them...

beikimajid, 2017-11-07:
I will send a video showing the difference in content playback.
right side : p10 smd 1/2 scan    left side: p10 3in1 1/4 scan.
https://drive.google.com/file/d/1d9WVBw-9kEY_exqLg5i0x3gwwSDc-3fe/view?usp=sharing

link the website my bought it :  https://www.aliexpress.com/item/Freeshipping-outdoor-p10-smd-led-display-module-16x32-32x16-160x320-320x160-1-2-scan-high-brightenss/32669606028.html

MaitreDede, 2017-11-07:
Do you plug them together or use them separatly ?

beikimajid, 2017-11-08:
The results are the same for both

MaitreDede, 2017-11-08:
My question is how do you plug your panels to your raspberry pi :
- the first panel connected to the second, and the second to the pi ?
- both connected to the same pi (the parallel config) ?
- each panel connected to its own pi ?

beikimajid, 2017-11-08:
Hi MaitreDede.
Your questions do not really make a difference.
What transformer is written is important.

I coonect 1/2 scan(right side) panel to pi then connect 1/2 to 1/4(left side) with this command:
./demo -D 1 runtext16.ppm --led-chain=2 --led-rows=16
also I use two pi and connect to each panel separately.

Please talk about transformer & why in 1/2 scan multiple led lit for a coordinate & how to write mapping for it

MaitreDede, 2017-11-09:
Ok. I think that in the first case, you can't connect the panels together. They have incompatible scan mode.
For example : addresses for a 1/4 scan are 0,1,2,3 and 1/8 has also 4,5,6,7. When displaying 4-7, it is unterstood by the 1/8 as 4-7, but the 1/4 will understand them as 0-3. So you will have an ugly display.
I think @hzeller will confirm this.

So if you plug them on their own pi, yes, we can dig into your display problem when using your 1/2 panel.
I have a 1/32 panel with only A/B (other 1/32 panels has ABCDE). To understand how it works, I wrote a simple program that drive directly the GPIO, to understand the pin signal sequence to send.
If you are interested in trying to figure out, I may write some docs to help you...

If it is a transformer problem, I can't help you (yet). :)

beikimajid, 2017-11-09:
I will be grateful if you send sample program.
Has anyone no idea about mapping table for 1/2 scan? This panel(1/2) work with hd2016 as well.

MaitreDede, 2017-11-09:
My program source is here : https://github.com/MaitreDede/RaspberryPi.Libs/blob/master/CoremanDriverPoc/Program.cs
It is dotnet core, you need a bit of configuration on your dev computer then deploy to you pi.

It allows to [drive manually](https://github.com/MaitreDede/RaspberryPi.Libs/blob/master/CoremanDriverPoc/Program.cs#L132-L143) each GPIO pin. It uses the [pigpio](http://abyz.me.uk/rpi/pigpio/) library (available on apt). There is also the [working code](https://github.com/MaitreDede/RaspberryPi.Libs/blob/master/CoremanDriverPoc/Program.cs#L412-L481) for my panel.

This program is really simple : no pwm, no colors, just on/off.

If you prefer using C/C++, libpigpio has a [C interface](http://abyz.me.uk/rpi/pigpio/cif.html).

beikimajid, 2017-11-14:
Does anyone have an idea?
Someone has a pixel mapping table for 1/2 scan like this:

this is 1/4 scan mapping table
[image]

angelogoncalve, 2017-11-14:
Hello Dear. I hope you are fine. I have a panel P6 32x32 outdoor 1/8 scan rate which repeat a pixel for example (0,0) have a pixel in the same collumn but eigth rows above (9,0). I think I will need configure the matrix for rows=16 and chain= 2 and use a transformer to do the translation in rows from each pixels but I do not get the wrigth transformer to do that. Can you give some ideas to do the rigth transformer for this panel from china? See the link below:

https://ibb.co/mCZNwv

How can I scroll text in a panel P6 16x32 indoor 1/8 scan rate? 

If you see in the center of the panel is ok the the entire square but in the top has the part superior of the square and bottom the part inferior of the square.

How can I fix this please?

I am using the command: sudo ./demo -r 32 -c 1 -D 0

But with scan rate 1/8 in my panel 32x32 I need build a correct transformer in the hzeller library and run a new command: sudo ./demo -r 16 -c 2 -D 0

He made prepare for 32x32 panel but with a scan rate 1/16 (indoor panels). The scan rate should be a parameter to pass in the configuration of the options of the matrix that is wonderful if it is possible. 

I need your help please.

Thank you for your atention and your help.

Best Regards,

Ângelo Gonçalves.

beikimajid, 2017-11-18:
hello
I find pixel mapping for 1/2 scan with HD2016 and D10 controller.
[image]
Can anyone help me in writing transformer?
Please help me.

KentWalker, 2017-11-19:
You could try this for starters:

```
new_y = y & 8;
test_y = y & 5;
if(test_y == 0) new_y += 2;
if(test_y == 1) new_y +=6;
if(test_y == 5) new_y +=4;
if(y & 16) new_y +=1;

new_x = x % 8;
if(y & 2) new_x += 8;
else new_x = 7 - new_x;
new_x += (x & 8) * 16;

```

beikimajid, 2017-11-20:
Hello 
this transformer **does not** work for 1/2 scan  correctly.

KentWalker, 2017-11-20:
The transform should work for the LED map supplied based on 32 led segment blocks.
You will need to modify for different LED segment block sizes.

To test if the transform is going to work we need to make sure only single pixels are lit for a given coordinate to start with.  You need to try with no transformer enabled in your code to test.

Looking at your table we can see for (1,1) led 72 should be lit.  Since our pixel coordinates are zero based we first need to subtract 1 from our LED number so pixel (x,y) maps to the correct LED.  
Finding the x,y transform coordinates is defined by the segment block (sb) size which may be 8, 16 or 32, depending on how the panel is laid out.
This will give us LED (x,y) where x = (led % sb), y = (led / sb).
So for led = 71; if sb = 8, try pixel (7,8); if sb = 16, try pixel (7,4) and if sb = 32, try pixel (7,2). 

Only one of these pixel coordinates should give a lit LED at (0,0).  If more than one LED lights you will need to find the right combination of rows, chains and SUB_PANELS_ (in the framebuffer.cc file, try defining SUB_PANELS_ =  4) that will light only one LED.  Note that if you change anything in the framebuffer.cc file you will need to recompile the library first before compiling your main code. 

Once you have the segment block size you can test other pixel locations, particularly those on the block boundaries.  It should be relatively easy to then generate a transform that will work for your panel. 

Finally check the transformer returns the correct values for width and height, you may have to double or halve to suit your panel/row/width/SUB_PANEL_ combination.

Let me know how you get on.

hzeller, 2018-01-21:
Please check out the new option `--led-multiplexing` https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing
It does the mapping built into the library (and can also be used from Python.)
(You need to do a fresh checkout and compile, this just went in).

hzeller, 2018-01-23:
With the AB thing, it could also be `--led-row-addr-type=1`

closed 2018-02-17 by hzeller