hzeller/rpi-rgb-led-matrix #423: How to expand image using led image viewer

Den214, 2017-11-06:
Hello,

I am trying to expand my image to take up my entire display. When I display an image using the led image viewer tool, the image is shown in a 32x32 section of my whole display either in the top left corner of the display by default or if I use -C on the command line it is centered. I am assuming this is just the normal operation of led image viewer. 

I would like to know if there is a way to make the image take up the entire display and how? Is there an easy way to implement this?

Here is how my display is chained. It is twelve 32x32 panels chained together to make a 128x96 display.
(Really I am using six 64x32 panels but these are essentially two 32x32 panels each.)

[<][<][<]
[>][>][>]
[<][<][<]<-----Pi output

hzeller, 2017-11-06:
What are the commandline flags you are giving? If you only give it a chain length of 9 then the image viewer will see a long display with 288x32 Pixel and scale the image to fit in that height.
You also need to write a pixel transformer that translates it to 96x96, and enable it in the image viewer.

In this case though I strongly suggest to use three parallel chains as described in the wiring (and is available as PCB to solder up). Not only you don't have to worry about the transformer but your refresh rate will be higher.

Den214, 2017-11-06:
My mistake, the panels are chained in this formation consisting of 12 panels chained not 9:

[<][<][<][<]
[>][>][>][>]
[<][<][<][<]<-----Pi output

But regardless yes, I will try both options. 
Thanks for the help I appreciate it, will comment again if I run into another issue.

hzeller, 2017-11-06:
With 12 panels you really want to use the parallel chains, as a single chain will result in more visible flicker.

Den214, 2017-11-06:
Will this also apply for video viewer as well? Was planning to play around with that as well once I got this working how I want it to.

hzeller, 2017-11-06:
Yes, the video viewer will also benefit from the parallel chains.

Den214, 2017-11-06:
Ok. Since my current display is not a perfect square though, and after breaking up my single chain of 12 panels into 3 parallel chains of 4 panels each, the image displayed will still have some empty space between the left and right side of the image and the edge of the display correct?

This would have to be fixed by altering the transformer?

hzeller, 2018-01-21:
Yes, with a transformer, you can do arbitrary pixel mapping.

closed 2018-01-21 by hzeller