hzeller/rpi-rgb-led-matrix #425: Rasbian9 Problem

mmft1359, 2017-11-08:
Hello 
I download Rasbian9 and follow project documentations to make and run led-image-viewer. after successful  make,i execute demo -D0 and led-image-viewer ... But nothing is displayed.
I did not have this problem in Rasbian8.
Can anyone help me?!

hzeller, 2017-11-08:
- What is the wiring you are using ? If  you are not using the default wiring (e.g. Adafruit HAT), make sure to give the correct -`-led-gpio-mapping`
- Did you follow all the recommendations mentioned in the  [Troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting) ?

mmft1359, 2017-11-13:
Hello
My wiring is exactly the same as Rasbian 8. in Rasbian 8 everything is OK but  I will not respond when I work on Rasbian 9 with the same settings and wiring.
It is noteworthy that when I run program on Rasbian9, the CTRL+C does not work at runtime, and I have to use the CTRL+Z and then kill the process.

hzeller, 2017-11-13:
The newer Raspbian look different w.r.t to GPIO memory map detection, but I fixed that a couple of days after Raspbian 9 came out. Are you using a recent checkout of rpi-rgb-led-matrix or an older checkout (a couple of months old).
If it is a recent checkout, then they might have changed around things again and I'll have a look. If you had an older checkout, just do a fresh copy.

hzeller, 2017-11-13:
Also see #383

mmft1359, 2017-11-17:
I tried the new version and my problem was solved.
Thank you.

closed 2017-11-17 by mmft1359

esbennn, 2018-01-17:
I'm currently having this problem. 

Installing as detailed in [here](https://learn.adafruit.com/raspberry-pi-led-matrix-display/software), and configuring a single 32x32 matrix, nothing shows up using Stretch from 27/11 17. 
Popping in an sd card with Jessie, installing the same way and using the same configuration, it works as expected.

I have followed the clues in the troubleshooting section, ie disabling sound, 1-Wire and GUI.

Any ideas?

hzeller, 2018-01-17:
Do not use any Adafruit documentation if you are using this library. Use the documentation that comes with this library, and it will work.
Also, only use a fresh checkout if you are using Debian9, apparently yours is more than half a year old if it does not work with 9.

esbennn, 2018-01-29:
Just got back from vacation!
@hzeller my god, you are right. Seems like Adafruit absolutely do not keep their repo updated. 
Using yours, it works on Stretch. 
Thanks, by the way - this library is great!