hzeller/rpi-rgb-led-matrix #426: repeat pixel in panel P6 32x32 outdoor 1/8 scan rate

angelogoncalve, 2017-11-08:
Hello Dear Zeller. I hope you are fine. I have a panel P6 32x32 outdoor 1/8 scan rate which repeat a pixel for example (0,0) have a pixel in the same collumn but eidth rows above (9,0). I think I will need a transformer but I to do the translation in rows from each pixels but I do not have the wrigth transformer to do that. Can you give some ideas to do the rigth transformer for this panel from china? Thank you for your atention. Best Regards, Ângelo Gonçalves.

angelogoncalve, 2017-11-08:
eigth* rows

angelogoncalve, 2017-11-13:
Hello Dear Zeller. I hope you are fine. I have a panel P6 32x32 outdoor 1/8 scan rate which repeat a pixel for example (0,0) have a pixel in the same collumn but eigth rows above (9,0). I think I will need configure the matrix for rows=16 and chain= 2 and use a transformer to do the translation in rows from each pixels but I do not get the wrigth transformer to do that. Can you give some ideas to do the rigth transformer for this panel from china? See the link below:

https://ibb.co/mtCLia

How can I scroll text in a panel P6 16x32 indoor 1/8 scan rate? 

Thank you for your atention.

Best Regards,

Ângelo Gonçalves.

angelogoncalve, 2017-11-13:
The picture that I saw in the panel for demo 0 is:

https://ibb.co/mCZNwv

hzeller, 2018-02-17:
There is now a `--led-multiplexing` parameter which might already solve your problem. Can you try that ?

closed 2018-02-17 by hzeller