hzeller/rpi-rgb-led-matrix #429: Framebuffer copy hdmi to matrix rgb

openscoreboard, 2017-11-14:
Hello:
Is it possible used rpi-fb-matrix without adafruit hat?
My proyect is a sport scoreboard and want copy hdmi to matrix rgb led.

https://youtu.be/tvCH6AX_kB0

Thanks!!

hzeller, 2017-11-14:
There is the https://github.com/adafruit/rpi-fb-matrix project that uses this library as the backend.
I have updated their code so that they use a more recent version of the rpi-rgb-led-matrix. Like with this library, you choose the hardware mapping with the `--led-gpio-mapping=...` flag. So if you are using the [active 3 adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3), you just set this to `--led-gpio-mapping=regular --led-parallel=3 --led-chain=...`

openscoreboard, 2017-11-14:
Thanks hzeller!!

closed 2017-11-14 by openscoreboard