hzeller/rpi-rgb-led-matrix #430: Led panel P8 16x32 bad scan

getvladimir4, 2017-11-14:
Hello Sirs,
I told you that I have correctly tested the panles led p10 16x32, p5 32x64 and everything was successful. But when testing the led panel p8 16x32 does not work it apparently requires a different scan. Could you help me how to solve?

P5 32x64 (ok)
https://www.youtube.com/watch?v=DA7VOTb2M0o
P8 16x32 (bad)
https://www.youtube.com/watch?v=DA7VOTb2M0o

[image]
[image]

P.S:. all the panels that I have are hub75

angelogoncalve, 2017-11-14:
Dear Vladimir. First what is the scan rate for your panel 16x32? The hzeller library is prepare for indoor panels and for panels with 16x32 the scan rate is 1/8. Perhaps tour panel is outdoor and have a scan rate 1/4 or 1/2. You need build a correct transformer for your new scan rate. I think it is better speak with Zeller. He is expertise in this area. I am try to do that with a 32x32 panel with a scan rate 1/8 (see https://ibb.co/mCZNwv) but I did not fix yet and I need help too. He made prepare for 32x32 panel but with a scan rate 1/16. The scan rate should be a parameter to pass in the configuration of the options of the matrix that is wonderful if it is possible. Best Regards, Ângelo.

angelogoncalve, 2017-11-14:
your*

closed 2017-11-23 by getvladimir4

MikkelMiksen, 2026-04-07:
Does anyone a have fix yet? please