hzeller/rpi-rgb-led-matrix #431: text-example code crashes?

TanerBursoy, 2017-11-16:
Hi,

Thanks for the great job and library you did.

I'm using text-example demo code for my own application.

However, besides my own code, when I use the original text-example.cc code, it crashes.

I type terminal to sudo ./text-example four or fıve times and then raspberry stucks. (start and stop)

Therefore, I have to power off raspberry and start it again. 

Can you help me please?

Thanks.

hzeller, 2017-11-23:
Did you compile the latest version of this software @TanerBursoy ?
If you are running a newer version of Raspbian, you need to make sure to also run a fresh checkout of rpi-rgb-led-matrix, as they changed the way Pi versions can be detected.

TanerBursoy, 2017-11-23:
Hi Hzeller,
I downloaded the fresh checkout of rpi-rgb-led-matrix and the problem solved.
Thanks you very much.

closed 2017-11-23 by hzeller