hzeller/rpi-rgb-led-matrix #432: Parallel LEDs with Adafruit HAT?

mattdesl, 2017-11-17:
I get an error trying to use Adafruit HAT with three parallel LED chains, the library says only 1 is supported with this mapping. Can somebody explain why this mapping is limited? Is there a way to support 3 rows with this setup or something similar?

I have an unusual configuration: a triangle of 6 32x32 Adafruit LED panels (5mm pitch) with an Adafruit RGB HAT and Raspberry Pi 2 model B. 3 panels on the bottom row, 2 on the middle, and 1 on top. I’m working in C++ and comfortable writing my own transformer for this, or even fudging things a bit... it doesn’t need to align perfectly.

hzeller, 2017-11-17:
There is only one connector on the Adafruit HAT, so this is why you only can have one chain.

hzeller, 2017-11-17:
... so I suggest to solder up an adapter that allows three chains:
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

hzeller, 2017-11-17:
If you want to do three rows and only have a Adafruit HAT, then just make a longer chain and wrap that around so that you have such a set-up, then you can write your transformer that does the mapping
( https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates ).

However, if you have the parallel chains, your refresh rate is of course higher, so this is why it is a good idea to use an adapter with three chains.

closed 2017-11-17 by hzeller

mattdesl, 2017-11-17:
Thanks for the quick reply – I’ll give it a go!

mattdesl, 2017-11-21:
I'm leaving this here in case anybody else is running into the same problem. My code/solution is really hacky and in-elegant, but it works well for my case, and the performance seems suitable.

I was having trouble getting the transformer to work, so I just wrote a simple function to re-map the pixels before setting them.

https://gist.github.com/mattdesl/7e075159e768edc7a4a46d3981c2e7ce

hzeller, 2017-11-21:
So, if you already have the code to remap a pixel, it is a tiny step to make this a transformer with the advantage that you don't pay the price of pixelmapping every time you set a pixel, as it is just a converted to a static look-up map at startup, so I encourage you to add this little step.

What were the troubles you had with transformers ?