hzeller/rpi-rgb-led-matrix #433: entry of unicode characters

marwellit, 2017-11-17:
hi

I have successfully linked up some 16x32 rgb panels and all is working well
I'm using runtext.py with a mod to pull the text that is displayed from a file

I'm having trouble getting Unicode characters (specifically the GB pound sign) into a file to be displayed on the matrix

i can echo the characters into  a file

eg

echo £hello£ > testtext

but when its read in for drawtext i get

  File "runtext.py", line 30, in run
    len = graphics.DrawText(offscreen_canvas, font, pos, 12, textColor, my_text)
  File "rgbmatrix/graphics.pyx", line 44, in rgbmatrix.graphics.DrawText (rgbmatrix/graphics.cpp:2056)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xa3 in position 0: ordinal not in range(128)

any advice would be most welcome

thanks
Dave

Wqrld, 2017-11-24:
i think that you need to save the file as ascii(should be possible) and we need to change a encoding string somewhere

hzeller, 2017-11-24:
The c++ api expects strings as UTF8 encoding. Not sure how Python is handling that internally, but if you know Python, you might be able to use that.
Otherwise I suggest to use c++ directly, see the text example.

closed 2018-02-17 by hzeller