hzeller/rpi-rgb-led-matrix #434: how build a correct transformer 1:4 scan rate for 16x32 panel

vitthalrm, 2017-11-18:


angelogoncalve, 2017-11-18:
Hello Dear. I hope you are fine. I have a panel P6 32x32 outdoor 1/8 scan rate which repeat a pixel for example (0,0) have a pixel in the same collumn but eigth rows above (9,0). I think I will need configure the matrix for rows=16 and chain= 2 and use a transformer to do the translation in rows from each pixels but I do not get the wrigth transformer to do that. Can you give some ideas to do the rigth transformer for this panel from china? See the link below:

https://ibb.co/mCZNwv

How can I scroll text in a panel P6 16x32 indoor 1/8 scan rate? 

My problem is my 32x32 panel has 1/8 scan rate and when I run a demo (a square rotating) from the hzeller library is like this:

If you see in the center of the panel is ok the the entire square but in the top has the part superior of the square and bottom the part inferior of the square.

How can I fix this please?

I am using the command: sudo ./demo -r 32 -c 1 -D 0

But with scan rate 1/8 in my panel 32x32 I need build a correct transformer in the hzeller library and run a new command: sudo ./demo -r 16 -c 2 -D 0

I need your help please.

Thank you for your atention and your help.

Best Regards,

Ângelo.

hzeller, 2018-01-22:
Can you try the new [`--led-multiplexing`](https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing) option ? It supports these kind of panels and also works from Python.

closed 2018-02-17 by hzeller