hzeller/rpi-rgb-led-matrix #436: Resize fonts

openscoreboard, 2017-11-18:
I have matrix 32x32 running examples ok
Is it posible to change the font`s size? (By sample 10x20.bdf)
I need full size and now the font is a 60% of 
my matrix
Thanks!

angelogoncalve, 2017-11-18:
from rgbmatrix import graphics
font = graphics.Font()
font.CharacterWidth(font_size)

I have a panel P6 32x32 outdoor 1/8 scan rate which repeat a pixel for example (0,0) have a pixel in the same collumn but eigth rows above (9,0). I think I will need configure the matrix for rows=16 and chain= 2 and use a transformer to do the translation in rows from each pixels but I do not get the wrigth transformer to do that. Can you give some ideas to do the rigth transformer for this panel from china? See the link below:

https://ibb.co/mCZNwv

How can I scroll text in a panel P6 16x32 indoor 1/8 scan rate? 

My problem is my 32x32 panel has 1/8 scan rate and when I run a demo (a square rotating) from the hzeller library is like this:

If you see in the center of the panel is ok the the entire square but in the top has the part superior of the square and bottom the part inferior of the square.

How can I fix this please?

I am using the command: sudo ./demo -r 32 -c 1 -D 0

But with scan rate 1/8 in my panel 32x32 I need build a correct transformer in the hzeller library and run a new command: sudo ./demo -r 16 -c 2 -D 0

I need your help please.

Thank you for your atention and your help.

Best Regards,

Ângelo.

openscoreboard, 2017-11-23:
The font bdf is static size
How can you generate a new file with font more big....?
I tired change a file bdf with nano edit  but is really dificult
Any tool?

hzeller, 2017-11-23:
You can use the [otf2bdf](https://www.math.nmsu.edu/~mleisher/Software/otf2bdf/) tool to convert any ttf font into an appropriately sized bdf tool.

(also mentioned in the [Getting started](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#getting-started) section)

openscoreboard, 2017-12-03:
Otf2bdf work for me!!
Thanks

closed 2017-12-03 by openscoreboard

angelogoncalve, 2017-12-03:
Can you disponibilize the code?

Thanks.

abscooters, 2019-03-14:
the otf2bdf link is dead

Is there a way to resize the fonts Bigger?

I have four 32 x 64 panels set up as a 64 x 128 sign with U-mapper and can't figure out how to fill the whole sign height(64) with scrolling text.  have downloaded craploads of bdf fonts and not finding the right solution

hzeller, 2019-03-14:
Looks like this is there http://sofia.nmsu.edu/~mleisher/Software/otf2bdf/

hzeller, 2019-03-14:
Also this should work https://github.com/jirutka/otf2bdf

abscooters, 2019-03-14:
Thank you, I will take a closer look.  

Also found  dumping text to a 64 pixel high image in gimp works, and then just export it to a ppm and use the scrolling image.  Then I can use virtually any font/color etc....

Thank you for this incredible software for the matrix!!

baku104788, 2020-05-25:
> 
> 
> Thank you, I will take a closer look.
> 
> Also found dumping text to a 64 pixel high image in gimp works, and then just export it to a ppm and use the scrolling image. Then I can use virtually any font/color etc....
> 
> Thank you for this incredible software for the matrix!!

Would it be possible for you to share this? I just want a font to fill the screen / will be scrolling the text.  Attempting to use the otf2bdf tool above ran into (multiple) dead ends.

hzeller, 2020-05-25:
What were the problems you encountered with the otf2bdf tool ?

It is so much easier to use the text with font directly, then use the text scroller.
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#text-scroller

baku104788, 2020-05-25:
> What were the problems you encountered with the otf2bdf tool ?

It didn't want to run on windows - I couldn't get the free text dll to install. I'm just looking for a single font to use for scrolling on a 32height panel.  I'll be honest I didnt' invest a huge amount of effort into the dll but I did find the prebuilt one and windows just gave me an error when I tried to import it. I'm creating a (basic) mqtt client so I can control the output over mqtt and use for notifications with my  home automation - the code I have for the scrolling is your code already - thank you for that :)

hzeller, 2020-05-25:
You have the Raspberry Pi where you can simply run the otf2bdf :)
But there are certainly tools on Windows that output a bdf font format, which is the most basic format, so it should be supported by all font editors.

forever80-dev, 2021-11-13:
Hello, is there a way to automatically center the text?