hzeller/rpi-rgb-led-matrix #439: Linux framebuffer driver

MaitreDede, 2017-11-20:
Hi,
I would like to know if anyone has made (or plans to make) a linux kernel framebuffer driver to expose a `/dev/fb` device that directly drives the led matrix ?
Thanks.

hzeller, 2017-11-20:
Adafruit has a program that copies over the framebuffer to the LED library https://github.com/adafruit/rpi-fb-matrix

MaitreDede, 2017-11-20:
I saw it, but I was wondering if there is a way to have a "led matrix framebuffer", to launch apps (and maybe X) and display them directly on led matrix, instead of having a services that does the copy...

For example, launche a Qt app (compiled for [embedded devices](http://doc.qt.io/qt-5/embedded-linux.html)) and display it directly on the matrix.

I have a sample to build qt for the pi and launch an app on EGL.

In the worst case, I will copy the dispmanx display to ledmatrix (I do it already in my [opengl poc](https://github.com/MaitreDede/RaspberryPi.Libs/tree/master/Samples/OpenGlToMatrix)),but I was just asking, in case :)

hzeller, 2017-11-20:
You could just make a Qt backend that writes to the matrix.

But even the regular framebuffer copy rpi-fb-matrix is not too bad, you can definitely use it to show some things with sufficient refresh, so a Qt app might just work directly with it, as it can use that framebuffer.

MaitreDede, 2017-11-20:
An other way, is it possible to create a virtual framebuffer, launch app/X on it, then copy display data to matrix ?

hzeller, 2017-11-20:
It is just software to write, so, yes, everything is possible :)

closed 2018-01-21 by hzeller