hzeller/rpi-rgb-led-matrix #440: Rotate panel 90 degrees to portrait mode :help-wanted

bradcornford, 2017-11-20:
Hi there, 

I've been using your package and its works wonderfully. I've been building a Tetris game using it, see: https://github.com/bradcornford/Rgb-Matrix-Tetris. I'm currently using a 32x64 panel, with the configuration options as follows:

```
    'hardware': 'adafruit-hat',
    'rows': 32,
    'chain_length': 2,
    'parallel': 1,
    'pwm_bits': 11,
    'brightness': 25,
    'lsb_nanoseconds': 130,
    'gpio_slowdown': 1,
    'disable_hardware_pulsing': False,
    'rgb_sequence': 'RGB'
```

At the moment the panel is in landscape mode, i.e. 64 columns, 32 rows. Is there a way in which i can flip the panel to be in portrait mode? i.e. 32 columns, 64 rows. I couldn't seem to find anything in your documentation around this, and I wondered if there was a better way to achieve this than pixel re-mapping in my game?

hzeller, 2017-11-20:
Yes, you can use the rotation transformer for that. Have a look at [`examples-api-use/demo-main.cc`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/demo-main.cc#L1160) to see an example how the RotateTransformer is applied:

```
matrix->ApplyStaticTransformer(RotateTransformer(rotation))
```

hzeller, 2017-11-20:
Just looked at your code and see you are using Python. In that case, there is no easy way to reach that from the language binding, but you can hack it into the matrix library, roughly here:
  https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L184

... and then recompile the Python binding, it should work from there.

bradcornford, 2017-11-20:
Ah yes, sorry, should have mentioned I was using Python. Thanks for taking a look at this, I'll give that a go and let you know how i get on.

bradcornford, 2017-11-22:
Yeah that works as expected. Thank you for your help!

GOBish, 2018-01-26:
I'm looking into doing the same thing (rotating to portrait orientation) in python and I think the location of the line of code you referenced above (line 184) has changed now due to some code updates. Could you please point me in the right place now? 
Any chance of adding the rotation as an option in the shell commands for the python examples? That would be really awesome. Thanks again!

hzeller, 2018-01-26:
You can still add a transformer manually at the end of the constructor [before line 206](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L206)

Having said that, if someone wants to tackle the 'make transformers available in Python' issue, comment on issue #101 and I can tell you what I have in mind implementation-wise so that you can work on a pull request.

GOBish, 2018-01-26:
Thanks for the reply. I'm having a little trouble getting it to work by adding it manually to the python code.
@bradcornford , any way you can post the code you added to make this work? Thanks

GOBish, 2018-03-17:
Now that CanvasTransformer has been replaced with PixelMapper - how do i rotate by 90 degrees in python?

hzeller, 2018-03-17:
So this is just possible by giving in the RGBMatrixOptions inside the Python program the `Rotate` argument with angle, e.g.
```
options = RGBMatrixOptions()
options.pixel_mapper_config = "Rotate:90"
```

If you are using `samplebase.py`, you also can give this with the `--led-pixel-mapper` flag.

Note, you need to check out the latest version and recompile the Python binding so that it is available.

oliverdarvall, 2019-05-02:
Does the pixel_mapper have a limit in terms of the board size that it can rotate in Python ?

I have a 128x32 configuration that I set to Rotate:90 , but it only displays on the first 32x32 boards then. If I rotate by 270 then it only displays on the last 32x32 boards. 

Any ideas ? Easy fix ? Thanks !

hzeller, 2019-05-02:
do you have an example of what you do?

oliverdarvall, 2019-05-06:
> do you have an example of what you do?

Initially I had 8  32x16 panels that were connected into 2 parallel chains of 5 each, thus ending up with 128x32 effectively.

The panels however were physically installed like this:
[image]
Thus installed in Portrait orientation, instead of Landscape

I then tried rotating everything (in Python) using the "Rotate:90" option (or 270), but as mentioned the display rotated, but only the first 32x32 or last 32x32 panels displayed anything, the rest were blank.

I eventually dismantled the display and physically orientated the panels until it displayed as I wanted. 

However, I would really prefer to install the panels in Portrait orientation and using my font I would then display a single digit per panel. Is this possible to achieve using the "Rotate" options ? Am I perhaps understanding the orientation of the panels incorrectly ?

Thanks !

`
#!/usr/bin/env python
from rgbmatrix import graphics
from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

options = RGBMatrixOptions()

options.hardware_mapping = "regular"
options.rows = 16
options.cols = 32
options.chain_length = 4
options.parallel = 2
options.row_address_type = 0
options.multiplexing = 3
options.pwm_bits = 11
options.brightness = 100
options.pwm_lsb_nanoseconds = 150
options.led_rgb_sequence = "RGB"
options.pixel_mapper_config = "Rotate:90"
options.gpio_slowdown = 2
options.disable_hardware_pulsing = False
options.drop_privileges = False

self.matrix = RGBMatrix(options = options)        

font = graphics.Font()
font.LoadFont("my.bdf") # Used otf2bdf to create big font that fills a 32x16 completely with a digit 

self.matrix.Clear()
graphics.DrawText(self.matrix, font, 1, self.matrix.height-1, graphics.Color(255, 255, 0), "12345678")

time.sleep(10)
`

oliverdarvall, 2019-05-17:
> > do you have an example of what you do?
> 
> Initially I had 8 32x16 panels that were connected into 2 parallel chains of 4 each, thus ending up with 128x32 effectively.
> 
> The panels however were physically installed like this:
> [image]
> Thus installed in Portrait orientation, instead of Landscape
> 
> I then tried rotating everything (in Python) using the "Rotate:90" option (or 270), but as mentioned the display rotated, but only the first 32x32 or last 32x32 panels displayed anything, the rest were blank.
> 
> I eventually dismantled the display and physically orientated the panels until it displayed as I wanted.
> 
> However, I would really prefer to install the panels in Portrait orientation and using my font I would then display a single digit per panel. Is this possible to achieve using the "Rotate" options ? Am I perhaps understanding the orientation of the panels incorrectly ?
> 
> What I would like to achieve is the following:
> [image]
> 
> Thanks !
> 
> ```
> #!/usr/bin/env python
> from rgbmatrix import graphics
> from rgbmatrix import RGBMatrix, RGBMatrixOptions
> import time
> 
> options = RGBMatrixOptions()
> 
> options.hardware_mapping = "regular"
> options.rows = 16
> options.cols = 32
> options.chain_length = 4
> options.parallel = 2
> options.row_address_type = 0
> options.multiplexing = 3
> options.pwm_bits = 11
> options.brightness = 100
> options.pwm_lsb_nanoseconds = 150
> options.led_rgb_sequence = "RGB"
> options.pixel_mapper_config = "Rotate:90"
> options.gpio_slowdown = 2
> options.disable_hardware_pulsing = False
> options.drop_privileges = False
> 
> matrix = RGBMatrix(options = options)        
> 
> font = graphics.Font()
> font.LoadFont("my.bdf") # Used otf2bdf to create big font that fills a 32x16 completely with a digit 
> 
> matrix.Clear()
> graphics.DrawText(matrix, font, 1, matrix.height-1, graphics.Color(255, 255, 0), "12345678")
> 
> time.sleep(10)
> ```

Any idea guys ?

MrRSquared, 2019-09-22:
Hello,
 I want to use 3 32x64 panels to create a classroom display using the following setup.  I am programming in python.   I am wondering if it is possible to use pixel mapper to flip the third panel to portrait.  If it is possible, any idea how we do it?
Thanks.
Mr. R^2
[classroomMatrix](https://user-images.githubusercontent.com/25676667/65393895-1e310b00-dd55-11e9-856c-53c2d8950c54.png)

hzeller, 2019-09-22:
Yes, with pixelmapper you can map any geometry into any other.
If you read the [pixel-mapper.h](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/pixel-mapper.h) header, you have everything you need. You just have to implement a method `MapVisibleToMatrix()` in which you express which visible (x, y) coordinate is mapped to which matrix (x, y).

(you need to implement that in C++, but then you can use the mapper in Python).

closed 2019-09-22 by hzeller