hzeller/rpi-rgb-led-matrix #441: 32:32 with 1:8 scan multiplexing problem

vitthalrm, 2017-11-20:
I have command sudo ./demo -r 32 -c 2 -D 3 output like
[image]
How to solve

lukasmalkmus, 2017-12-11:
Exact same issue here, but I was not able to get a proper transformer running.

darknessii, 2017-12-24:
Try this ugly Transform Set, it works up to 8 Panel only on one Parallel output.

void UArrangementTransformer::TransformCanvas::SetPixel(
  int x, int y, uint8_t red, uint8_t green, uint8_t blue) {

int new_x = 0;
int new_y = 0;

if (y <= 7) {
  new_x = ((x / 16) * 32) + (x % 16);
  if ((y & 8) == 1) {
   new_x += 16;
  }
  new_y = ((y / 16) * 8) + (y % 8);
}else {
  if ((y >= 16) && (y <= 23)) {
    new_x = ((x / 16) * 32) + (x % 16);
    if ((y & 8) == 1) {
      new_x += 16;
    }
    new_y = ((y / 16) * 8) + (y % 8);
  }else {
    new_x = ((x / 16) * 32) + (x % 16);
    y -= 8;
    if ((y & 8) == 0) {
     new_x += 16;
    }
    new_y = ((y / 16) * 8) + (y % 8);
  }
}
delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

darknessii, 2017-12-24:
./demo -D 3 --led-rows=16 --led-chain=2

hzeller, 2018-01-21:
Please check out the new option `--led-multiplexing` https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing
It does the mapping built into the library (and can also be used from Python.)
(You need to do a fresh checkout and compile, this just went in).

lukasmalkmus, 2018-02-14:
Can confirm. The new multiplexing options fixed my issue with the same display. I'm not talking in behalf of the issue author but I think this can be closed. @hzeller @angelogoncalve

darknessii, 2018-02-14:
The --led-multiplexing option does not work for me. 

Only my Transfomset works fine. i did some optimisation.

void UArrangementTransformer::TransformCanvas::SetPixel(
  int x, int y, uint8_t red, uint8_t green, uint8_t blue) {

int new_x = 0;
int new_y = 0;

//Panel SMD3535 RGB OS-P6-32x32-8S
if ((y <= 7) || ((y >= 16) && (y <= 23)) || ((y >= 32) && (y <= 39)) || ((y >= 48) && (y <= 55))) {
  new_x = ((x / 16) * 32) + (x % 16);
  if ((y & 8) == 1) {
   new_x += 16;
  }
  new_y = ((y / 16) * 8) + (y % 8);
  }else {
	new_x = ((x / 16) * 32) + (x % 16);
	y -= 8;
	if ((y & 8) == 0) {
		new_x += 16;
	}
	new_y = ((y / 16) * 8) + (y % 8);
  }
delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse

[image]

**Not working.....**

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse --led-multiplexing=0

[image]

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse --led-multiplexing=1

[image]

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse --led-multiplexing=2

[image]

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse --led-multiplexing=3

[image]

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse --led-multiplexing=4

[image]

Also the option  --led-row-addr-type  dose not work.

But thanks for the transform Set Option. this works fine.

lukasmalkmus, 2018-02-14:
Hm. Have you tried something like `./demo --led-rows=32 --led-cols=32 --led-chain=2 --led-parallel=2 --led-multiplexing=1 -D3` Thats how it worked for me.

darknessii, 2018-02-14:
Thank for the information, but it does not work with my panel.
it looks also funny...

./demo --led-rows=32 --led-cols=32 --led-chain=2 --led-parallel=2 --led-multiplexing=0 -D3
[image]

./demo --led-rows=32 --led-cols=32 --led-chain=2 --led-parallel=2 --led-multiplexing=1 -D3
[image]
 
./demo --led-rows=32 --led-cols=32 --led-chain=2 --led-parallel=2 --led-multiplexing=2 -D3
[image]

./demo --led-rows=32 --led-cols=32 --led-chain=2 --led-parallel=2 --led-multiplexing=3 -D3
[image]

./demo --led-rows=32 --led-cols=32 --led-chain=2 --led-parallel=2 --led-multiplexing=4 -D3
[image]

darknessii, 2018-02-14:
The Option 2 looks like the best, but not in the right order.

./demo --led-rows=32 --led-cols=32 --led-chain=2 --led-parallel=2 --led-multiplexing=2 -D3

and has to be reorderd. But how? Looks like a puzzle.

[image]

[image]

hzeller, 2018-02-17:
This looks related to #445 ? Let's continue there to not have too many parallel threads open.
You seem to have found a pixel mapping for that, so it would be good if you could provide a pull request with your mapping in multiplex-transformers.cc.

closed 2018-02-17 by hzeller