hzeller/rpi-rgb-led-matrix #442: Problems with P10 outdoor panel mapping

xer0design, 2017-11-22:
Hey!

I've been using 4 P8 32x16 panels for the past while with absolutely no issues, it's been great. I just bought 10 P10 32x16 panels to replace them and I'm having a lot of issues getting them working right, the mapping is all over the place and I can't seem to fix it.

See picture for an example of old and new panels showing what should be the same thing
https://ibb.co/i15DL6

The panels seem to be divided in half, with 4 sections from top to bottom, which can be clearly seen in demo 0
https://ibb.co/h5r9tR

I've tried looking through the posts and I see a lot of suggestions to use -L, which if used with various combinations for led-chain and led-rows, does seem to make it slightly better but still completely wrong.

Any help would be greatly appreciated, thanks.

angelogoncalve, 2017-11-22:
Hi. First what is the scan rate for your panel 16x32? The hzeller library is prepare for indoor panels and for panels with 16x32 the scan rate is 1/8. Perhaps tour panel is outdoor and have a scan rate 1/4. You need to 
[newTransformer.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/1497543/newTransformer.txt)
build a correct transformer for your new scan rate. I think it is better speak with Zeller. He is expertise in this area. 

Best Regards, Ângelo.

hzeller, 2018-01-21:
Please check out the new option `--led-multiplexing` https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing
It does the mapping built into the library (and can also be used from Python.)
(You need to do a fresh checkout and compile, this just went in).

closed 2018-02-17 by hzeller

xer0design, 2018-03-01:
I know the issue is closed but I'm only just getting back to these panels now so I thought I'd post the fix that I eventually found that worked for me.

They're 1/4 scan rate P10 panels. With the newest version of the code, these are the flags that work perfectly

`./demo -D0 --led-cols=16 --led-rows=16 --led-chain=2 --led-parallel=1 --led-brightness=30 --led-multiplexing=2
`

Multiply the chain length by 2, so 2 panels is --led-chain=4 etc.

Thanks for the help.