hzeller/rpi-rgb-led-matrix #443: LED PANEL 16X32 BAD SCANNING

getvladimir4, 2017-11-23:
hello masters!

I have the following problem for several weeks now. And the fact is that when executing the henner demos on the led panel 16x32 p8 I find it distorted as you can see the images and videos that I present below.
-------------------------------------------------------------------------------------------------------------------
*OK SCANNING in 16x32 P10
sudo ./demo -r 16 -c 1 -D 3
[image]
-------------------------------------------------------------------------------------------------------------------
*BAD SCANNING in 16x32 P8

sudo ./demo -r 16 -c 1 -D 3
[image]

sudo ./demo -r 16 -c 1 -D 1 runtext16.ppm
https://www.youtube.com/watch?v=zr0klS0OVoE

Apparently he has the following scan but I'm not sure about this.
[image]

Any suggestions will help me a lot. Thank you so much.
Please, could you help me solve this problem? It's very urgent!

hzeller, 2017-11-23:
You have to write yourself a transformer that maps the logical pixels to the physical ones.

Description in the [canvas.h](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/canvas.h#L51) header and in the [examples README](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates)

You can find a few [implementation examples here](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/transformer.cc).

hzeller, 2018-01-21:
Please check out the new option `--led-multiplexing` https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing
It does the mapping built into the library (and can also be used from Python.)
(You need to do a fresh checkout and compile, this just went in).on

closed 2018-02-17 by hzeller