hzeller/rpi-rgb-led-matrix #444: Libre Computer Board AML

beikimajid, 2017-11-25:
Hello
Can this board be a replacement for Raspberry Pi to run this project?

angelogoncalve, 2017-11-25:
Hello. What is the differences between Libre Computer Board AML and Raspberry Pi 3? The Libre Computer Board AM runs android things? Probably Libre Computer Board AML run this project with no problems because is similar as Raspberry Pi 3 and runs Linux/Debian OS I think.

hzeller, 2017-11-25:
Given that this library is using the features of the SoC of the Raspberry Pi and assumes certain memory mappings of GPIO pins, it is very unlikely that this board would work out of the box.

beikimajid, 2017-11-26:
Hi hzeller
link to product:
https://www.cnx-software.com/2017/06/30/libre-computers-le-potato-amlogic-s905x-development-board-goes-for-25-and-up-crowdfunding/

SoC – Amlogic S905X quad core ARM Cortex-A53 processor @ 1.5 GHz with penta core ARM Mali-450MP GPU, and Amlogic Video Engine 10
System Memory – 1GB or 2GB DDR3 RAM
Storage –  1x micro SD Card slot with UHS SDR104 support, eMMC module connector with HS400 support
Video Output – HDMI 2.0 Port, 3.5mm TRRS “AV” jack with CVBS (480i / 576i)
Audio Output – HDMI, AV jack (stereo audio), SPDIF output header, and I2S header
Connectivity – 10/100M Ethernet
USB – 4x USB 2.0 host ports
Expansion

    ADC + I2S Header
    40-pin Raspberry Pi header with I2C, SPI, PWM, 5V, 3.3V, and GPIO

Debugging –  UART via header for access to the serial console
Misc – IR Receiver, onboard jumper for HDMI CEC or GPIO 11 selection, status LEDs,
Power Supply – 5V via micro USB port
Dimensions – Raspberry Pi 3 form factor

hzeller, 2018-01-21:
they only claim to be compatible with the 'form factor' of the Raspberry Pi. If you use a different chip, you'll need to adapt the gpio code, but the rest should work.

closed 2018-01-21 by hzeller

zombodotcom, 2023-05-20:
@beikimajid did you ever get this working