hzeller/rpi-rgb-led-matrix #445: panel P6 32x32 outdoor 1/8 scan rate - mapping pixels wrigth transform

angelogoncalve, 2017-11-25:
Hello people. I nedd to have the wrigth transform for mapping pixels a panel P6 32x32 outdoor 1/8 scan rate.

I think I need to change trasnsformer.cc file in the UArrangementTransformer but I do not get the wright transformation for the pixels. 

In below is the transformation for a panel 16x32 with a scan rate 1:4:

void UArrangementTransformer::TransformCanvas::SetPixel(
  int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  int new_x = 0;
  int new_y = 0;
  if (y < 4 || (y > 7 && y < 12)) {
    new_x = (x * -1) + (24 * ((x / 8) + 1)) - 17;
    new_y = y;
  } else {
    new_x = x + (8 * ((x / 8) + 1));
    new_y = y - 4;
  }
  delegatee_->SetPixel(new_x, new_y, red, green, blue);
  
  /*
  if (x < 0 || x >= width_ || y < 0 || y >= height_) return;
  const int slab_height = 2*panel_height_;   // one folded u-shape
  const int base_y = (y / slab_height) * panel_height_;
  y %= slab_height;
  if (y < panel_height_) {
    x += delegatee_->width() / 2;
  } else {
    x = width_ - x - 1;
    y = slab_height - y - 1;
  }
  delegatee_->SetPixel(x, base_y + y, red, green, blue);
  */
}

Can anybody help me please to finish the project for the shchool.

Thank you for your atention.

darknessii, 2017-12-24:
hello guys, here I have an ugly code created for my p6 32x32 1/8 panel. it works for me. however, it only works for one --led-parallel=1 and at least 4 panels.

void UArrangementTransformer::TransformCanvas::SetPixel(
  int x, int y, uint8_t red, uint8_t green, uint8_t blue) {

// Panel SMD3535 RGB OS-P6-32x32-8S

int new_x = 0;
int new_y = 0;

if (y <= 7) {
  new_x = ((x / 16) * 32) + (x % 16);
  if ((y & 8) == 1) {
   new_x += 16;
  }
  new_y = ((y / 16) * 8) + (y % 8);
}else {
  if ((y >= 16) && (y <= 23)) {
    new_x = ((x / 16) * 32) + (x % 16);
    if ((y & 8) == 1) {
      new_x += 16;
    }
    new_y = ((y / 16) * 8) + (y % 8);
  }else {
    new_x = ((x / 16) * 32) + (x % 16);
    y -= 8;
    if ((y & 8) == 0) {
     new_x += 16;
    }
    new_y = ((y / 16) * 8) + (y % 8);
  }
}
delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

Example with 2 Panel:
(while :; do date +%T ; sleep 0.2 ; done) | sudo ./text-example -f ../fonts/8x13B.bdf -y8 -C140,0,0 --led-rows=16 --led-chain=4 --led-parallel=1 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=200 --led-no-hardware-pulse

darknessii, 2017-12-24:
Example with 4x 32x32 Panel U-arrangement (64x64)
./demo -D 0 --led-rows=16 --led-chain=8 --led-parallel=1 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse -L

angelogoncalve, 2017-12-26:
Hi the transformer for 32x32 panel with scan rate 1:8 can be these ones:

/*********************************/
/* P10outdoor Transformer Canvas */
/*********************************/
class P10outdoorTransformer::TransformCanvas : public Canvas {
public:
  TransformCanvas() : delegatee_(NULL) {}

  void SetDelegatee(Canvas* delegatee);

  virtual void Clear();
  virtual void Fill(uint8_t red, uint8_t green, uint8_t blue);
  virtual int width() const;
  virtual int height() const;
  virtual void SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue);

private:
  Canvas *delegatee_;
};

void P10outdoorTransformer::TransformCanvas::SetDelegatee(Canvas* delegatee) {
  delegatee_ = delegatee;
}

void P10outdoorTransformer::TransformCanvas::Clear() {
  delegatee_->Clear();
}

void P10outdoorTransformer::TransformCanvas::Fill(uint8_t red, uint8_t green, uint8_t blue) {
  delegatee_->Fill(red, green, blue);
}

int P10outdoorTransformer::TransformCanvas::width() const {
  return delegatee_->width() / 2;
}

int P10outdoorTransformer::TransformCanvas::height() const {
  return delegatee_->height() * 2;
}

void P10outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  if (x < 0 || x >= width() || y < 0 || y >= height()) return;
  /*int new_x, new_y;
  new_x = ((x / 32) * 64) + (x % 32);
  if ((y & 8) == 0) {
    new_x += 32;
  }  
  new_y = ((y / 16) * 8) + (y % 8);*/
  //fprintf(stdout, "value of new_x new_y :: %d %d\n ", new_x, new_y);
  
  /*int odd8_block =(((y % 16) / 8) +1) % 2; //to know if the pixel is in an odd 8pixel row
  int num_mod_x = (x / 32); //num of daisy chainned modules
  int new_x = (x + 32*num_mod_x + 32*odd8_block); 
  int new_y = (y % 8) + 8 *( y / 16);*/
  
  //The following code converts a 32x32 matrix into a 64x16 matrix for proper display.
  /*int new_x = x;
  int new_y = y;
  if (y < 8) 
    new_x = x + 32;
  else if (y < 16) 
    new_y = y - 8;
  else if (y < 24) 
  {
    new_x = x + 32;
    new_y = y - 8;
  }
  else 
    new_y = y - 16;*/
    
  //Panel SMD3535 RGB OS-P6-32x32-8S
  int new_x = 0;
  int new_y = 0;
  if (y <= 7) {
    new_x = ((x / 32) * 64) + (x % 32);
  if ((y & 8) == 0) {
    new_x += 32;
  }
  new_y = ((y / 16) * 8) + (y % 8);
  }else {
    new_x = ((x / 32) * 64) + (x % 32);
    //y -= 8;
  if ((y & 8) == 0) {
    new_x += 32;
  }
  new_y = ((y / 16) * 8) + (y % 8);
  }
  delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

/*************************/
/* P10outdoor Transformer */
/*************************/
P10outdoorTransformer::P10outdoorTransformer()
  : canvas_(new TransformCanvas()) {
}

P10outdoorTransformer::~P10outdoorTransformer() {
  delete canvas_;
}

Canvas *P10outdoorTransformer::Transform(Canvas *output) {
  assert(output != NULL);

  canvas_->SetDelegatee(output);
  return canvas_;
}

./demo -D 0 --led-rows=16 --led-chain=2 * number of panels --led-parallel=1

angelogoncalve, 2017-12-26:
The correct transformer that you send my friend for 32x32 panel with scan rate 1:8 is this:

//Panel SMD3535 RGB OS-P6-32x32-8S
  int new_x = 0;
  int new_y = 0;
  if (y <= 7) {
    new_x = ((x / 32) * 64) + (x % 32);
  if ((y & 8) == 0) {
    new_x += 32;
  }
  new_y = ((y / 16) * 8) + (y % 8);
  }
  else {
  if ((y >= 16) && (y <= 23)) {
    new_x = ((x / 32) * 64) + (x % 32);
    new_y = y - 8;
  if ((y & 8) == 0) {
    new_x += 32;
  }
  new_y = ((y / 16) * 8) + (y % 8);
  }
  else {
    new_x = ((x / 32) * 64) + (x % 32);
  if ((y & 8) == 0) {
    new_x += 32;
  }
  new_y = ((y / 16) * 8) + (y % 8);
  }
  }
  delegatee_->SetPixel(new_x, new_y, red, green, blue);

hzeller, 2018-01-21:
Please check out the new option `--led-multiplexing` https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing
It does the mapping built into the library (and can also be used from Python.)
(You need to do a fresh checkout and compile, this just went in).

darknessii, 2018-02-14:
The --led-multiplexing option does not work for me. 

Only my Transfomset works fine. i did some optimisation.

void UArrangementTransformer::TransformCanvas::SetPixel(
  int x, int y, uint8_t red, uint8_t green, uint8_t blue) {

int new_x = 0;
int new_y = 0;

//Panel SMD3535 RGB OS-P6-32x32-8S
if ((y <= 7) || ((y >= 16) && (y <= 23)) || ((y >= 32) && (y <= 39)) || ((y >= 48) && (y <= 55))) {
  new_x = ((x / 16) * 32) + (x % 16);
  if ((y & 8) == 1) {
   new_x += 16;
  }
  new_y = ((y / 16) * 8) + (y % 8);
  }else {
	new_x = ((x / 16) * 32) + (x % 16);
	y -= 8;
	if ((y & 8) == 0) {
		new_x += 16;
	}
	new_y = ((y / 16) * 8) + (y % 8);
  }
delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse

[image]

**Not working..... --led-multiplexing=**

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse **--led-multiplexing=0**

[image]

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse **--led-multiplexing=1**

[image]

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse **--led-multiplexing=2**

[image]

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse **--led-multiplexing=3**

[image]

sudo ./examples-api-use/demo -D 3 --led-rows=16 --led-chain=4 --led-parallel=2 --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse **--led-multiplexing=4**

[image]

Also the option  --led-row-addr-type  dose not work.

But thanks for the transform Set Option. this works fine.

hzeller, 2018-02-17:
So if the existing `--led-multiplexing=` values 0 .. 4 don't work, but you have figured out a transformer mapping for you, it would be great if you could send a pull request that adds these to [lib/multiplex-transformers.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-transformers.cc) - there might be other people that have the same panel so would be greatful for your contribution.

hzeller, 2018-03-17:
@darknessii I guess this can be closed now as this was the transformer you were working on ?

angelogoncalve, 2018-03-19:
Hello Dear Henner.

The --led-multiplexing=6 works fine. Thank you my friend for your help.

hzeller, 2018-03-19:
Thank @darknessii  who implemented this mapper, not me :)

closed 2018-03-19 by hzeller