hzeller/rpi-rgb-led-matrix #446: --led-rgb-sequence has no effect at all

devjb, 2017-11-26:
I have one 32x64 panel here that has G and B swapped. But no matter which --led-rgb-sequence I use, nothing changes at all. I am driving the panel with an old PI (1st gen), and thus I have --led-gpio-mapping=regular-pi1 set as well. But I do not think that it is related, because the colors are swapped nonetheless.

Any idea, why this could happen?

angelogoncalve, 2017-11-27:
You can do this:

from rgbmatrix import RGBMatrix, RGBMatrixOptions

def rgbmatrix_options():
      options = RGBMatrixOptions()
      options.led_rgb_sequence = "RBG"
      options.rows = 32
      options.chain_length = 2
      options.parallel = 1
      options.hardware_mapping = 'adafruit-hat'
      return options

options = self.rgbmatrix_options()
display = RGBMatrix(options=options)

hzeller, 2017-11-27:
Ah, are you using Python @devjb ?

devjb, 2017-11-27:
Thank you for your comment. But I do not user the python bindings and no adafruit hat. But if you suggest to set everything manually in RGBMatrix::Options instead of passing it by the parameters, I for sure could give this a try.

hzeller, 2017-11-27:
Mmh, the `--led-rgb-sequence=RBG` should work though.

hzeller, 2018-01-21:
did this work ?

closed 2018-01-21 by hzeller

devjb, 2018-01-21:
Not yet so far. I've had little time lately and did not further investigate. I got a pi 3  lastweek and will test the panel there again. I will reopen here if necessary.