hzeller/rpi-rgb-led-matrix #450: Brightness and power led flicker

Frank071, 2017-12-04:
Hi -- we have a setup with six (6) 64x32 panels and we saw the occasional flickering, meaning a short higher brightness. This seems to coincide with the red RPi3 led blinking as well (blink as in: shortly off). Now it gets weirder (in my unexperienced eyes) because with a specific image the red led is going haywire. It blinks in an erratic way, but this time the panels do not show the brightness flicker.
At first, when we only saw that incidental and correlated blinking, I looked up the meaning of the red led and could not found anything more than "it's the power led and (on the newer Pi's it blinks when the voltage drops below 4.6V". So I measured the voltage level, but that stays steady at 5,02x V. And the erratic blinking is also not what one would expect.
So my main question is: what can this blinking of the red led mean and how can it be related to the brightness flicker. I have already implemented all the documented tricks: isolcpus=3, disable sound module, etc. Help and/or pointers are appreciated....

angelogoncalve, 2017-12-05:
You can do this:

from rgbmatrix import RGBMatrix, RGBMatrixOptions

def rgbmatrix_options():
options = RGBMatrixOptions()
options.led_rgb_sequence = "RGB"
options.rows = 32
options.chain_length = 2 * 6
options.parallel = 1
options.hardware_mapping = 'adafruit-hat'
options.pwm_bits = 1
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

hzeller, 2017-12-05:
If the LED indicated power trouble, then you probably do indeed have power troubles.

Note that if you have 6 64x32 panels, your power supply will roughly need to supply  45 Ampere on average on a white display; and average means there are peaks of even higher current.

I suspect your power supply just glitches out very briefly  in these current peaks which causes the trouble. You might not be able to measure these glitches with a multimeter, so that might still show 5.02V but you don't see the 10ms glitches where it drops to 3V.

So what you need to do is
  * Make sure your power supply is beefy enough. For that set-up, if this is a P5 display, I'd probably get a 60A power supply for these number of panels.
  * Read the [A word about power](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power) section. The short current pulses are pretty taxing to power supplies and the supply lines have high inductance in relation to the pulse length, so I found that in some cases adding the capacitors is pretty needed for stability

Frank071, 2017-12-05:
Thanks Henner. I tried to find more information about the behaviour of the red led and from what I read I did not expect it to flash erraticly (but more a 'warning blinking'). Just for the record: the panels are fed per three, but it might be that the supply is not sufficient. I will check. 
It still leaves the question what causes the brightness spike, because I am back at the scenario that the brightness spike causes the voltage to drop (in stead of the other way around). Is there some way to monitor things to find out what is causing this?

@angelogoncalve - we neither use the AdaFruit Hat, nor python at the moment. But thanks for the input.

angelogoncalve, 2017-12-05:
You are wright Zeller the amperes that is need for 6 panels 64x32 pixels. For example for me I have two panels 16x32 and one raspberry and I need 5V/5A for all. For 6 panels 64x32 is: 6*4*(16)x32 = 6*4*2.5A=60A.

hzeller, 2018-01-12:
Did this get resolved with a solid power supply ?
To answer the monitor question: if it is voltage fluctuations, your best bet is an oscilloscope.

Since you have 6 panels: do you have them run in parallel (so 3x2 panels), or just 6 panels in a row ? It is strongly recommended to use as much parallelism as possible as it reduces CPU load and increases refresh rate. If you do have a long chain, there is another thing you can try, to uncomment [`#DEFINES+=-DDISABLE_RT_THROTTLE` in the lib/Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L121).

Frank071, 2018-01-12:
We are still working on this, dearly lacking as oscilloscope... We ordered different, more powerfull powersupplies, but we did not yet install those. Creating two parallel chains has some manufacturing disadvantages, but I understand your idea. I will also try your last option. And I will surely give feedback as soon as we found a solution.

closed 2018-02-17 by hzeller