hzeller/rpi-rgb-led-matrix #453: Help and Assistance regarding chained=3 16x32Rgb matrix from China(P8smd 160813-12...)

Arsenik4444, 2017-12-11:
Hi Sir, thanks a lot regarding this great work; please sir I'm having issues when running all Demos, all wiring look fine and i have attached a video and pictures of my issues, please may you assist with a right transformer, i'm a student and not too good in programing and i'm teaching myself in programing, please may you assist:
 sudo ./demo --led-rows=16 --led-chain=3 -D0 runtext16.ppm
sudo ./runtext.py --led-rows=16 --led-chain=3 --led-parallel=1
[image]
[image]
[image]

[image]
[image]

angelogoncalve, 2017-12-11:
It is because of the scan rate of the panel. Can you tell me thew scan rate of your panel (maybe is 1:4)? If yes you need to write a traansformer for mapping the pixels. I sugest see some issues that have similar problem as you have.

Arsenik4444, 2017-12-11:
Hi Sir thank you for your feedback, apologies for the question i did try to find out how i can see or identify the scan rate of the panel but i could not and i had a look in all the issues the other members had but i could not make really sense of the way the transformer are made as the library has been updated and all example are using the old library(sudo ./led-matrix), may you give me any guidance or a starting point as I'm new in programing to solve the issue, thank you Sir

angelogoncalve, 2017-12-11:
The scan rate you can see where you buy the panel, ask the manufacturer. First do a loop to see how you need to change the position of the pixels from your panel:

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

class Matrix(object):
    """
    Matrix Class initialization
    """

    def __init__(self):

        options = self.rgbmatrix_options()

        # Loads the matrix
        self.__display = RGBMatrix(options=options)

    def rgbmatrix_options(self):

        # Configuration for the led matrix - HZELLER Library
        self.width = 32
        self.height = 16
        chain = 1
        parallel = 1
        brightness = 100

        # Options for RGB Matrix Panel
        options = RGBMatrixOptions()
        options.brightness = brightness
        options.rows = self.height
        options.chain_length = chain
        options.parallel = parallel
        options.hardware_mapping = 'adafruit-hat'  # If you have an Adafruit HAT: 'adafruit-hat'
        options.inverse_colors = False
        options.led_rgb_sequence = "RGB"
        options.gpio_slowdown = 4  # Default 1
        options.pwm_lsb_nanoseconds = 130  # Default 130
        options.show_refresh_rate = 0  # Default 1
        options.disable_hardware_pulsing = True
        options.scan_mode = 1  # Default 0
        options.pwm_bits = 1  # Default 11
        options.daemon = 0
        options.drop_privileges = 0  # ROOT Privileges

        return options

    def pixel_draw(self, red, green, blue):
        
        for y in range(0, self.height):
             for x in range(0, self.width):
                  self.__display.SetPixel(x, y, red, green, blue)
                  time.sleep(1)

matrix = Matrix()
red = 0
green = 255 # If you want the pixels with green color
blue = 0
matrix.pixel_draw(red, green, blue)

Next see this issue https://github.com/hzeller/rpi-rgb-led-matrix/issues/313 and this one https://github.com/hzeller/rpi-rgb-led-matrix/issues/136
[newTransformer.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/1549443/newTransformer.txt)

Arsenik4444, 2017-12-12:
Good Day Sir, i just wanted to say thank you for your guide i will be going through the resources given by you and see what i will come up with, and will let you know, thank you.

angelogoncalve, 2017-12-12:
Good day. Some panels have a letter S, I think is for indicate the scan rate for example 4S scan rate 1:4 maybe is the scan rate for your panel. If you want to do the loop for the three panels 16x32 do this that I forget:

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

class Matrix(object):
"""
Matrix Class initialization
"""

def __init__(self):

    options = self.rgbmatrix_options()

    # Loads the matrix
    self.__display = RGBMatrix(options=options)

def rgbmatrix_options(self):

    # Configuration for the led matrix - HZELLER Library
    chain = 3
    parallel = 1
    self.width = 32 * chain
    self.height = 16 * parallel
    brightness = 100

    # Options for RGB Matrix Panel
    options = RGBMatrixOptions()
    options.brightness = brightness
    options.rows = self.height
    options.chain_length = chain
    options.parallel = parallel
    options.hardware_mapping = 'adafruit-hat'  # If you have an Adafruit HAT: 'adafruit-hat'
    options.inverse_colors = False
    options.led_rgb_sequence = "RGB"
    options.gpio_slowdown = 4  # Default 1
    options.pwm_lsb_nanoseconds = 130  # Default 130
    options.show_refresh_rate = 0  # Default 1
    options.disable_hardware_pulsing = True
    options.scan_mode = 1  # Default 0
    options.pwm_bits = 1  # Default 11
    options.daemon = 0
    options.drop_privileges = 0  # ROOT Privileges

    return options

def pixel_draw(self, red, green, blue):
    
    for y in range(0, self.height):
         for x in range(0, self.width):
              self.__display.SetPixel(x, y, red, green, blue)
              time.sleep(1)

matrix = Matrix()
red = 0
green = 255 # If you want the pixels with green color
blue = 0
matrix.pixel_draw(red, green, blue)

Arsenik4444, 2017-12-12:
Hi Sir Thank you for your reply, please may you be patient with me, with you hep i will be able to get this work as i manage to get to this point, i had a look on the back of the 16 x32Rgb matrix i'm using and i found the attached Images may be this will help,
for the script you send me to do ""the loop Test"", how can i use it i try to used it as a python scripts then I'm getting an error, may be I'm doing something wrong
[image]
[image]
[image]
the error: expected an indented block, at (self.rgbmatrix_options()),i still trying to figure out and be waiting for any suggestion and guide from your side, thank you.

angelogoncalve, 2017-12-12:
For indentation you can use an editor free PyCharm that works in windows or use the python2 editor that has the raspberry in the operating system (Debian) because if you see there is no this ({ }) to init and close a block in python. In python the indentation is do by number of backspaces. The scan rate for the square test that you do I think you have a panel outdoor with a scan rate 1:4 and the hzeller library is make for indoor panels that have a scan rate half of the rows of the panel: for this case number of rows / 2 = 16/2=8 (1:8) and your panel is 1:4 scan rate. You need to see the issues I send you to see how they make the transformer for mapping pixels in panels 16x32 pixels with a scan rate 1:4.

from rgbmatrix import RGBMatrix, RGBMatrixOptions

class Matrix(object):
      """
      Matrix Class initialization
      """
    

     def __init__(self):

           options = self.rgbmatrix_options()

           # Loads the matrix
           self.__display = RGBMatrix(options=options)

Arsenik4444, 2017-12-12:
Hi Sir, i did read all issues worked by others in the topic and the items you have send to me, thanks a lot, then i have see a transformer which is made and published by VolkerGoeschl and which may work for the panel i'm using but going through all files i do have few questions regarding the steps:
-what have to be done replacing or just adding the files in the corresponding location:
*where to add this to lib/transformer.cc: in the code
*where to add this to /include/transformer.h: in the code
*where to add this location and area:
rgb_matrix::CanvasTransformer *transformer = new rgb_matrix::P10outdoorTransformer();
matrix->SetTransformer(transformer);
*and after all done, what will be the final steps to activate the modification
thank you Sir.

angelogoncalve, 2017-12-12:
the issue 242 tells to put in the final of the files the transformer P10outdoorTransformer
[newTransformer.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/1552930/newTransformer.txt)
:
/lib/transformer.cc
/include/transformer.h
/lib/led-matrix.cc - lines 184 and 198 - ApplyStaticTransformer(P10outdoorTransformer);

remove all files that are in the dist-packages of the raspberry about the rgbmatrix library and compile the new files (I see that you not use the adafruit-hat to control the pixels and you use connection wires between the GPIO's of the raspberry and the IDC connector panel, that is no problem):

sudo HARDWARE_DESC=regular make install-python

Arsenik4444, 2017-12-13:
Hi Sir, thank you for your assistance i just have this fine question where i'm a bit confuse:
where this part of the code have to be added:
rgb_matrix::CanvasTransformer *transformer = new rgb_matrix::P10outdoorTransformer();
matrix->SetTransformer(transformer);
does is t use the same role with ApplyStaticTransformer(P10outdoorTransformer);?

angelogoncalve, 2017-12-13:
Yes you can use this lines:

rgb_matrix::CanvasTransformer *transformer = new rgb_matrix::P10outdoorTransformer();
matrix->SetTransformer(transformer);

but I do not know where are to put this lines (maybe in the end of /include/canvas.h file).

I see also that in the file /lib/led-matrix.cc - lines 184 and 198 - you uncomment these lines and add the new transformer P10outdoorTransformer - ApplyStaticTransformer(P10outdoorTransformer);

Arsenik4444, 2017-12-13:
Hi Sir i just wanted to say thank you for your assistance: when trying the  gausie Transform for P10 Outdoor Display, the Demo do not display anything, and i also tried different transformers in the forum but still, i think i will have to take some time to improve my programing and try to make more sense about how to solve this, i will also still checking if the could be any other update from others solution and Approach, 
it will be also great will Mr  Hzeller could design a debug or something which may be able or guys with little in programing could use as support,but thanks to you and Mr  hzeller, nice and great Job;
i have attached the image of the controller used for my display but raspberry pi would be a great experice, may be next time; thanks again...
[LEDOK6.6 controller.pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/1555667/LEDOK6.6.controller.pdf)

angelogoncalve, 2017-12-13:
I tryed also the gausie Transform for P10 Outdoor Display and yes not show any pixel of the panel for example with white color using the function matrix.SetPixel(0, 0, 255, 255, 255). Another transformation is the https://github.com/hzeller/rpi-rgb-led-matrix/issues/178 or the https://github.com/narioinc/rpi-rgb-led-matrix. Try the demo tests with each one to see what happens or with for example the function matrix.SetPixel(0, 0, 255, 255, 255). Can you disponibilize your code to see how it works with LEDOK6.6 and what is the language. The LEDOK6.6 works with outdoor panels? It is wonderful if it works with these type of panels my friend.

Arsenik4444, 2017-12-13:
Hi Sir, with the LEDOK6.6 some applications are used to control the Matrix and you're right that the LEDOK6.6 is used to control outdoor Display and the software or  interface used are: LedEditor or the LEDset but i'm not really sure about the language used to design these interface, i will try again tomorrow the new transformer and see what i will get,
 it will be really wonderful if this can work because the LEDOK6.6 (Controller) is not working(now faultly) and i have to Oder a new one, but Raspberry pi will be great if working and this may be also be used or controlled remotely;
i just wanted to ask you Sir, i have look in all directories but i could not see the function matrix.SetPixel(0, 0, 255, 255, 255);this may help for my progress; thanks a lot for your patient Sir.

angelogoncalve, 2017-12-14:
Please tell me about your tests with the transformers if you can and thank you. You can use this to test in python to see a pixel in the position (0, 0) with the green color :

from rgbmatrix import RGBMatrix
chain = 1
parallel = 1
rows = 16
matrix = RGBMatrix(rows, chain, parallel)
while True:
      x_column = 0
      y_row = 0
      red = 0
      green = 255
      blue = 0
      matrix.SetPixel(x_column, y_row, red, green, blue)

Arsenik4444, 2017-12-14:
Hi Sir, i tried the transformer from https://github.com/narioinc/rpi-rgb-led-matrix but when running the sudo ./demo -D 1 runtext.ppm i could see a message 96x16 and some other word in the pi command screen but nothing came up in the display even by changing the value of the Row and chain with  sudo ./demo --led-rows=16 --led-chain=3 -D0 runtext.ppm(Or runtext16.ppm),
But when running the Python demo sudo ./runtext.py there was a message but the same funny message as from all other used transformers i already tried(no change), but when i left the Lab i remembered some of the members in the Forum were talking about cleaning all python libraries, i will try to read again and see what will be the result but i think the https://github.com/narioinc/rpi-rgb-led-matrix should be the good transformer but there must be some changement to be made and make it work , but in the lab tomorrow i will again try and also test the python as you just guide me, i'm not too good in programing that why things are very slow but i have to get it work, thanks a lot Sir and apologies for the lat update, i will let you know after tomorrow tests, thank you; it's not easy...

angelogoncalve, 2017-12-14:
Yes it is need to clear all python libraries. Yes it is not easy my friend. Please when it is possible for you tell me about the tests but only do they as a first stage with one panel 16x32. Thank you for your help Dear Sir.

Arsenik4444, 2017-12-15:
Good day Sir, i have no word just to say thanks a lot you're a Genius, the Display i working well but for now I'm using a single matrix as you mention but one issue is that the matrix is divided in 2 as showned in the attached in the image, i also have include the first steps used to fix the issue this may be usefull to others:
*Delete /remove the rpi-rgb-led-matrix folder in the Pi
* and Clone the orginnal one from Mr hzeller:
cd
git clone https://github.com/hzeller/rpi-rgb-led-matrix
cd /home/pi/rpi-rgb-led-matrix/examples-api-use
make
*now i used the gausie transformer: https://github.com/gausie/rpi-rgb-led-matrix/tree/p10transformer
-/home/pi/rpi-rgb-led-matrix/include i just edited the transformer.h by including gausie // P10outdoorTransformer
-/home/pi/rpi-rgb-led-matrix/lib i just edited the transformer.cc by including gausie from /* P10outdoor Transformer Canvas */ to everything down
-/home/pi/rpi-rgb-led-matrix/lib  i just applied the P10outdoor Transformer in the led-matrix.cc: 
change this  // ApplyStaticTransformer(...); in line 184&line 198 to this ApplyStaticTransformer(P10outdoorTransformer()), all this editing and modification have to be done inside of the main function
-then:
cd /home/pi/rpi-rgb-led-matrix/examples-api-use
make
-all Demo are working with:
sudo ./demo --led-rows=8 --led-chain=1 -D0 runtext16.ppm 
anything else will distort the image or word in the Display Up or Down;
**please sir how can now we fix this issue of the matrix dived by 2:
And the Python still not work, after doing this steps:
cd /home/pi/rpi-rgb-led-matrix/bindings/python
make clean
make build-python
sudo make install-python
nothing has change still showing the same funny Hello World,
how can to stop the scrolling in the Python and Demo? 
Thanks a lot Sir i will be on the lab on Monday and will still be waiting for your feedback, thank you.
[image]
[image]

Arsenik4444, 2017-12-15:
Hi Sir got it working now for 1 and 3 matrix Display great, thanks a lot for your assistance and support i did almost give up;
*one matrix:
sudo ./demo --led-rows=8 --led-chain=2 -D1 runtext16.ppm
*3 matrix:
sudo ./demo --led-rows=8 --led-chain=6 -D1 runtext16.ppm
* There still few issue to be attend which are:
-Few pulsating Leds switching on in the display when running the demo and how can this be fixed?
-And 
[image]
[image]

how to use the Python to control the Display, 
i will be on the lab on Monday to continue with this thanks a lot Sir  a good weekend to you

angelogoncalve, 2017-12-15:
Hi Sir you are wrigth if you see the issues I send you for outdoor panels where the scan rate is half of the scan rate of indoor panels because they have more brigthness and they have protection for the water, you need to put --led-rows=rows/2 and --led-chain=2*number of panels conected in serie.

I forget to tell you that for the tests sorry my friend but I am so happy you can do it. I try with a 32x32 panel outdoor with a scan rate 1:8 and I do not get results. 

I think if I do what you do that will work but I need the wrigth transformer for 32x32 with 1:8 maybe the transformers of narioinc will work. Can you send me please your folder rpi-rgb-led-matrix with your files changed please to compile in my raspberry and test please my friend? Thank you very much for your help.

You have two choices to eliminate the flickers by software or hardware (the code below control the matrix using python using raspberry pi 3 and adafruit hat - use the PyCharm to help the indentation):

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

class Matrix(object):
"""
Matrix Class initialization
"""

def init(self):

options = self.rgbmatrix_options()

# Loads the matrix
self.__display = RGBMatrix(options=options)

def rgbmatrix_options(self):

# Configuration for the led matrix - HZELLER Library
chain = 3
parallel = 1
self.width = 32 * chain
self.height = 16 * parallel
brightness = 100

# Options for RGB Matrix Panel
options = RGBMatrixOptions()
options.brightness = brightness
options.rows = self.height
options.chain_length = chain
options.parallel = parallel
options.hardware_mapping = 'adafruit-hat'  # If you have an Adafruit HAT: 'adafruit-hat', 'regular' for wires, 'adafruit-hat-pwm' for flicker see the hardware image attach 
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4  # Default 1 # because the flicker
options.pwm_lsb_nanoseconds = 130  # Default 130
options.show_refresh_rate = 0  # Default 1
options.disable_hardware_pulsing = True
options.scan_mode = 1  # Default 0
options.pwm_bits = 1  # because the flicker - Default 11
options.daemon = 0
options.drop_privileges = 0  # ROOT Privileges

return options

# S
[image]
ee a pixel in the position (0, 0) with the green color
matrix = Matrix()
while True:
      x_column = 0
      y_row = 0
      red = 0
      green = 255
      blue = 0
      matrix.SetPixel(x_column, y_row, red, green, blue)

angelogoncalve, 2017-12-15:
To stop the scrolling in the Python and Demo you can use Ctrl + C or kill the process sudo kill -s python

angelogoncalve, 2017-12-15:
Have a nice weekend my friend.

angelogoncalve, 2017-12-15:
Do not forget also the enough power, for me I have two panels 16x32 and one raspberry and I need 5V/5A for all. For 3 panels 16x32 is: 3x(1x16)x32 = 3x2.5A = 7.5A.

angelogoncalve, 2017-12-15:
For the flicker see also this as I mentioned in the hardware image picture attach (https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker)

angelogoncalve, 2017-12-15:
If you see this issue https://github.com/hzeller/rpi-rgb-led-matrix/issues/331 you can put in the file /home/pi/rpi-rgb-led-matrix/examples-api-use/demo-main.cc in dded the following to demo-main.cc you can add in the line 1163 after the check for rotation:

matrix->ApplyStaticTransformer(rgb_matrix::P10outdoorTransformer());  // outdoor panel

angelogoncalve, 2017-12-15:
If you see this issue #331 you can put in the file /home/pi/rpi-rgb-led-matrix/examples-api-use/demo-main.cc in you can add in the line 1163 after the check for rotation:

matrix->ApplyStaticTransformer(rgb_matrix::P10outdoorTransformer()); // outdoor panel

angelogoncalve, 2017-12-15:
As the issue 331 for panel 32x32 with 1:8 scan rate the issue https://github.com/hzeller/rpi-rgb-led-matrix/issues/299 can be useful also and speaks also to put the new transformer in the function ApplyStaticTransformer. Also not forget for panel 32x32 1:8 with scan rate the demo test scroll text is: *one matrix:
sudo ./demo --led-rows=16 --led-chain=2 --led-parallel=1 -D1 runtext.ppm

angelogoncalve, 2017-12-15:
As the issue 331 for panel 32x32 with 1:8 scan rate the issue #299 can be useful also and speaks also to put in the file demo-main.cc the new transformer in the function ApplyStaticTransformer. Also not forget for panel 32x32 1:8 with scan rate the demo test scroll text is: *one matrix:
sudo ./demo --led-rows=16 --led-chain=2 --led-parallel=1 -D1 runtext.ppm

Arsenik4444, 2017-12-17:
Hi Sir, thank you for your feedback, i will be sending the file which i got working from my side tomorrow  when i get to the lab, thank you.

angelogoncalve, 2017-12-17:
Hi Dear Sir. Can be the all folder rpi-rgb-led-matrix please to see all changes files please, to see how fix the 32X32 panel with 1:8 scan rate. Thank you very much for your help my friend.

angelogoncalve, 2017-12-17:
I hope that works the changes with the transformer in C files for the scan rate for the panel, in the matrix functions in python also to control the pixels of the panel.

Arsenik4444, 2017-12-18:
Hi Sir, i have attached the link for the full working folder for now only the Demo is working:
https://drive.google.com/file/d/1dTVOYvaEbvJi0DRuctRJby_kOTGPHEZ8/view?usp=sharing
i manage to fix the Led problem it was just Ghosting LEd issue by running the Demo:
sudo ./demo --led-rows=8 --led-chain=6 --led-pwm-lsb-nanoseconds 300 -D0 runtext16.ppm
by increasing the Pulse with to 300, but i still have to read and find out how to integrate the Transformer to work with other demo, python..., thank you and good luck at your side; thank you.

angelogoncalve, 2017-12-18:
Thank you for your atention Dear Sir. Yes one possible to disappear with the ghost pixels is increasing the the pwm but there are others options also --led-gpio-slowdown 4 for raspberry pi 3 and --led-pwm-bits 1. For python as you work with images .ppm to scroll see if the image-scroller.py can work with your panel 16x32 my friend with your options matrix you mecioned in your comment above my friend.

angelogoncalve, 2017-12-18:
The directory of the python file for scroll a image is /home/pi/rpi-rgb-led-matrix/bindings/python/samples/image-scroller.py

angelogoncalve, 2017-12-19:
My friend I get it for 32x32 panel with scan rate 1:8 with your attach rpi-rgb-led-matrix with the addition of transformer using the issue 299 (narioinc has also this transformer) and apply these tranformation in led-matrix file. Other way is apply the transformer in the demo-main file as I told you before. For python, you can see this attach I send you for transform a text in a image .ppm and use the demo D1 to scroll the image, is a way to control the panel using python. Have you any way instead a mix between python and demo C file compiled to control the pixels of the panel, good is only with python my friend. Thank you for your help.
[16x32-rgb-display-with-raspberry-pi-part-2.pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/1570007/16x32-rgb-display-with-raspberry-pi-part-2.pdf)

Arsenik4444, 2017-12-20:
Hi Sir;thank you too for your help and i'm also happy to here that you could fix the issue with your 32×32 rgb run and make it work;
At my side when I will be get a chance I will have to try integrate the working  transformer to the python binding; which in the future I will see how to do a remote control rgb matrix;but thanks a lot for your assistance and to Mr Hezller for this great work and library;thank you.

angelogoncalve, 2017-12-20:
Hi Sir. Thank you for your atention. If you get the integration the working transformer to the python binding, please can you share my friend? I try but only works with Image python functions using the document I sent to you 16x32-rgb-display-with-raspberry-pi-part-2.pdf, and  it works using also the demo-main compiled to scroll the image with is no too good. Thank you for your help my friend.

angelogoncalve, 2018-01-21:
Hello Dear Henner Zeller. Nice job my friend, well done. I have two panels 32x32 with scan rate 1/8 with different mapping pixel.

One need this transformer (I have already put this C++ code and use in Python and works fine):

void P10outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
if (x < 0 || x >= width() || y < 0 || y >= height()) return;
int odd8_block =(((y % 16) / 8) +1) % 2; //to know if the pixel is in an odd 8pixel row
int num_mod_x = (x / 32); //num of daisy chainned modules
int new_x = (x + 32*(num_mod_x + odd8_block)); 
int new_y = (y % 8) + 8 *( y / 16);
delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

Or this:

void P10outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
if (x < 0 || x >= width() || y < 0 || y >= height()) return;
int new_x, new_y;
new_x = ((x / 32) * 64) + (x % 32);
if ((y & 8) == 0) {
  new_x += 32;
}  
new_y = ((y / 16) * 8) + (y % 8);
delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

To works with this transformer I need to put these flags:
--led-rows=16 --led-chain=2

To use the flag --led-multiplexing and not create the transformer in Python what is the correct program commandline flags for this transformer:
--led-rows=32 --led-multiplexing=1 or other combination of flags and values?

Another one need this transformer (I have already put this C++ code and use in Python and works fine):

void P10outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
if (x < 0 || x >= width() || y < 0 || y >= height()) return;
int new_x = x;
int new_y = y;
  
new_x = ((x / 16) * 32) + (x % 16);
if ((y/8)%2 == 1) {
  new_x += 16; 
}
new_y = ((y / 16) * 16) + (y % 8);
delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

To works with this transformer I need to put these flags:
--led-rows=32 --led-chain=2

To use the flag --led-multiplexing and not create the transformer in Python what is the correct program commandline flags for this transformer:
--led-rows=32 --led-multiplexing=1 or other combination of flags and values?

Thank you for your atention and help Dear Henner Zeller.

angelogoncalve, 2018-01-21:
I compare the your transformers with I have in Python with a 16x32 panel with 1/8 scan rate because I will try the tests tomorrow with your transformers in the each panel 32x32 with 1/8 scan rate and I think for:

The first case the correct program command line flags for the transformer:
--led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=1

The second case the correct program command line flags for the transformer:
--led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=2

Thanks for your atention.

angelogoncalve, 2018-01-23:
Each panel 32x32 with 1/8 scan rate I run with your library in Python and for:

The first case the correct program command line flags for the transformer:
--led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=1

The second case the correct program command line flags for the transformer:
--led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=2

But the second case I need also set the pixels columns and push eigth pixels columns to the top (y-8) and the others to the bottom (y+8) for half height panel. For the other half, do the same (y+-8) for the columns.

Dear Zeller is another way for the second case using only the flags? Thank you for your atention and your help.

hzeller, 2018-01-23:
[Instead of writing the same text to two issues #461 and #453 can you just please stick to one @angelogoncalve ?]

angelogoncalve, 2018-01-23:
Yes sure I will stick to only one Dear Zeller I am sorry for my mistake. Thank you for your atention.

closed 2018-02-17 by hzeller