hzeller/rpi-rgb-led-matrix #455: help required for continuous ticker display

nkpatro, 2017-12-12:
I have a 620*32 pixel display connected to RPI3 + Adafruit HAT. I am able to get independent text scrolling though font is problem for now.
I have multiple texts blocks that I want to tail behind one another. However, currently I am able to see only one text block at a time. i.e. the first text block scrolls from right to left. the second one starts only after the first is exited.
How do I achieve the tailing text blocks? if anyone already done that can help me with the code.

Thanks
Navin

angelogoncalve, 2017-12-12:
You can see this https://github.com/alexlukelevy/ticker-tape. Do not forget to use threads to stop a scroll when an request to a new scroll is make. You can use this to stop / kill a thread (https://eli.thegreenplace.net/2011/12/27/python-threads-communication-and-stopping).

hzeller, 2018-01-21:
Whenever you update your display make sure to either have enough letters 'in the queue' behind the text you're printing, or just do two write print operations. The [`DrawText()`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/graphics.h#L89) function returns the number of pixels it advanced, so you know where to place your next text.

This is more a generic programming question that this forum is not so much suited for. Closing for now.

closed 2018-01-21 by hzeller