hzeller/rpi-rgb-led-matrix #456: Two display servers on the same network implementation advice

gpulido, 2017-12-14:
Hello,
First of all I would like to apologize if this is not the proper channel to ask.
I'm developing a sport competition timer / display using a led-matrix panel.
Each "timer" is composed by a couple of led matrix, a raspib 3 and the rpi-rgb-led-matrix library to manage the display.
I managed to make it work properly showing local content to each display.
My problem is that I need to deploy at least two timers and to be capable to send the same content to each of them at the same time, and I don't know what is the best approach.
Each timer are under the same IP network so I can add any tcp / udp communication layer / protocol that could be needed.
The synchronization precision between each display has to be of least than one second (1ms would be awesome) as they are showing a countdown timer.

My idea is to have an android app or another device that would generate the data to display (the led matrix representation) and send it to each device at the same time, and here is where I'm not sure what to do.
One idea is to use some kind of custom message format and send it over UDP broadcast, so each timer would need a UDP broadcast listener that receives the message, inflate it and push it to the leds. I could send a message each ms with the leds representation.
My concern is the synchronization, and that the UDP broadcast could flood the network.
I have take a look to the pixelpush idea, but I haven't found the way to implement the protocol in android to use it.
Any suggestions of what is the best approach to such project?
Thank you in advance

angelogoncalve, 2017-12-14:
You can use the protocol socket TCP JSON Remote Procedure Call 2.0 (https://github.com/joshmarshall/jsonrpctcp).

angelogoncalve, 2017-12-14:
You can use the server in python and the client in android (java).

hzeller, 2017-12-14:
Don't use TCP, because then you have to send it one by one to each display.

If you use UDP multicast, you can send the image content in parallel to both displays. Here is an [simple-udp.cc](https://gist.github.com/hzeller/9ece7d7150c6d28ed1825eee85b5d243) example that shows a regular UDP server that receives the image content via UDP and shows it on the display. Change that so that you can receive multicast packets. This is of course limited to 64k/3 pixels because of the size limitation of a UDP packet, so if you need more, you can maybe modify this by just sending the text.

Then your server only has to send one packet per display update and they are shown at exactly the same time on both displays.

gpulido, 2017-12-14:
Hello,
That is what I thought. 
Add a "Multicast UDP listener layer" on each devices and using a multicast udp sender (on an android app for example). I made some test with another implementation on where I used UDP broadcast, and a custom message that is being processed on each side of the communication but I found it not very reliable (I don't really care if there are some package lost as far as there is at least one package / 5ms as the timer resolution is a second). Also it caused problems on the network (probably caused by the broadcast), and there where a delay caused by the process of the message on each raspi.

That is the cause for looking to a more "dummy" implementation on where the entire display is already send so the raspi doesn't need to do anything (just represent the pixels), and without flooding the network.

The first part I guess it could be done wit the simple-udc.cc sample and the second part using multicast instead of broadcast.
My panel is going to be 64x48 pixels = 3072 so it would be below 64k/3.
So just to be sure, I have just to compose a bmp image with the right size and send it over udp, is that right?
Thank you again for the help.

hzeller, 2017-12-14:
Essentially yes, you fill a buffer with RGBRGBRGB..., no header etc, just the raw data.

angelogoncalve, 2017-12-14:
Dear Henner Zeller. You are wright if use TCP because of the port, it is only possible to send the data one by one to each display.

angelogoncalve, 2017-12-14:
Dear GPulido and Zeller can you tell me please using the simple-udp.cc file how can I use it in the library and compile and show please with an example in python send a pixel at the same time, matrix.SetPixel(0, 0, 255, 255, 255), to two panels 16x32 pixels, matrix = RGBMatrix(16, 1, 1). Thank you for your atention.

hzeller, 2017-12-14:
The simple-udp.cc has in the comment header the description how to build it. Once you build it, you just run it with the usual flags you give to the LED matrix panel. It listens on UDP port 9999.

I don't know any Python to really give an answer, it will probably be something along the lines of:
```
import socket

width = 32   # or whatever the final display size is
height = 16
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sock.connect(("localhost", 9999))   # localhost if same machine; its IP address or name
data = bytearray(width * height * 3)
x = 3   # Let's set a pixel at (3,5)
y = 5
offset = (x + y * width) * 3
data[offset] = 255       # red
data[offset + 1] = 255 # green
data[offset + 2] = 255 # blue
sock.send(data)
```

angelogoncalve, 2017-12-14:
Thank you for your atention Zeller. I see this issue that is similar that you explain (https://github.com/hzeller/rpi-rgb-led-matrix/issues/152).

hzeller, 2018-01-21:
Looks like things are working ?

closed 2018-01-21 by hzeller

gpulido, 2018-01-21:
Hello @hzeller,
I'm sorry I haven't time to test it although I thing it will work properly.
I will do it asap and will give feedback.
Sorry for the delay.

gpulido, 2019-02-27:
Hello again, 
I has been a while and I had to delay the project for a year due personal things.
Now I'm back working on it and started to review the project.
I'm still need to syncronize several panels,  but the local "tools" that the libraries provide to draw text, load images etc to the matrixcanvas are more usable to have to design each glyph by hand. So I was thinking if there is any way to have a rgb-led-matrix library running WITHOUT any real panel attached (or a fake one) where I can control and "export" the matrix image content raw data to be send through UDP.

d3d9, 2019-02-27:
If there is no existing project with a large amount of usage of the library tools then it might be a good idea to skip that and create the data using a library like Pillow on its own; sending the image for example as a ppm file over the network, and using SetImage on the devices.    

If there already is a lot of code that would be too much to rewrite, it might be nice to have something like what you mentioned, some sort of fake panel so that this library can be used without something attached, and even better, so that it can be used on any device including e.g. windows machines.

In https://github.com/d3d9/rpi-rgb-led-matrix I added the ability to export the canvas as a ppm file, so you can get the image after you used functions to draw text, images etc., now it would be nice to have some way to completely ignore anything that goes further than that (gpio, ...). Maybe this ppm function is already enough to use as a base for your idea; as long as it runs on a raspberry pi (so that the library itself works) and with the correct settings so that you get a image of the correct size to send.

gpulido, 2019-02-27:
Hello @d3d9 
Thanks for the idea and link.
However I need the library to just keep exporting the "representation" to be send by udp for example.
That make me think on a solution where a Fake/ mock display can be plugged and that such display just open the socket and send the data through it and decide also if stop there or just chain the matrix panel (if any) attached phisically.
This way with just the same library, a raspi can acts as a generator of representation (server) and others raspis as dumb clients that just pick the data and represent it.
There is still a problem, the throughput, as the resolution / frecuency has to be reduced to adjust to the udp channel.

For my project (two synced displays), the other alternative is to send through UDP some kind of "representation of the data": json, protobuffer, and let the clients process the data, and generate the synchronization each one. My concern here is that I need about 1ms / 2ms of precision between all display and processing the data could add some work. However I think that a raspbib 3+ has enough horsepower to do both.
In the compelte picture, I want to run all in a tynicorelinux, with a bottle framework layer to control the timer, and probably serve an angular client app to communicate through rest with the framework.

hzeller, 2019-02-27:
Canvas is just an interface and the graphics operations work with canvases, so you can easily implement that.

gpulido, 2019-02-27:
Hello @hzeller I'm not sure what do you mean.
Which part should be easy? the drawing methods to the canvas ?

hzeller, 2019-02-27:
As I understand

  * You want to prepare graphics somewhere, using the provided graphics functions in rpi-rgb-led-matrix
  * Then want to send this frame-buffer via UDP somewhere to be displayed on a real panel
  * For that, you want to implement some sort of 'fakedisplay' that is not displaying anything, but just sends the framebuffer via UDP

So the good thing is that the graphics functions don't work on a RGBMatrix, but on the [`Canvas`-Interface](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/canvas.h#L21). So anything implementing Canvas can be used directly with the graphics functions, line drawing, fonts etc.

You now can implement your fake display by implementing the Canvas. You probably also want a method that flushes UDP. This whole class should only be a few lines of code to implement as each method is essentially a one-liner.
You need a framebuffer, where you set the right bytes with `SetPixel()`, `Clear()` is essentially a bzero() operation, width/height properties is just something you initialize in construction (when you size the UDP buffer) then `write()` to a UDP socket.

gpulido, 2019-02-27:
That exactly my use case :) Thanks for writing it down so clearly :) I don't know if that make sense or if it is efficient...

If I don't use the graphics functions (that are already working) I will have to replicate them to draw to my fake canvas that would send the data as raw representation.
However from what I can see (my c++ skills are limited ) this fake display need to be implemented somehow inside the library, to be "plug". 
I will take a look and try to generate a pr.

hzeller, 2019-02-27:
You can implement it outside the library, you just need the canvas include. `Canvas` is a public interface.

Then the rpi-rgb-led-matrix graphics functions will all work on your implementation of the UDPCanvas.

angelogoncalve, 2019-02-27:
You can use this it works for both:

https://github.com/BNNorman/rpi-rgb-led-matrix-animator