hzeller/rpi-rgb-led-matrix #460: SDL Events Ignored

irataxy, 2017-12-22:
I'm having an issue with the C++ library. I'm attempting to make a Tetris clone using the hardware and software listed below. I'm using SDL2 (https://www.libsdl.org/) to handle keyboard input, game timing, and output to a monitor (for testing). I have noticed the following:

* If I try to call CreateMatrixFromFlags() or CreateMatrixFromOptions outside of main() (e.g., in another class), the function never returns and the RGB matrix will not light up. I don't get any errors, its just like everything hangs. It works fine when I create the matrix inside of main() and pass it to another class, though. I combed through the examples and could not find an instance of creating the matrix outside of main().

*  When I create the matrix in main() and then enter the game loop in another class, SDL events (keyboard presses from a USB keyboard) are ignored. Everything works fine if I don't create the matrix. Are the USB ports on the Pi somehow disabled while the LED matrix is running?

Any help would be greatly appreciated!

Hardware:
Adafruit RGB Matrix HAT
Raspberry Pi 3 Model B
Adafruit RGB  16x32 Matrix
Hardware hack (HARDWARE_DESC=adafruit-hat-pwm)
One core refresh (isolcpus=3)
Disabled on-board sound (dtparam=audio=off)

Software:
Raspian Stretch Lite
SDL2 + SDL ttf

hzeller, 2017-12-22:
If the function hangs it could be that you are using an older library from a couple of months back ?

irataxy, 2017-12-22:
I downloaded the rpi-rgb-led-matrix project from github this week.

hzeller, 2017-12-22:
are you starting threads before you create the matrix ? Since the default is to go in daemon mode, which essentially does a `fork()`, all threads up to that point would be stopped.
So create the matrix as early as possible in main() before doing anything else, then start threads etc.

Speaking of which: by default, privileges are dropped from root to 'daemon' ,unless you specified `--led-no-drop-privs` or switched off that feature in the options. Maybe you don't have permissions to read the SDL events in that case ?

hzeller, 2017-12-22:
Did you get it to work ?

irataxy, 2017-12-22:
It just did! Thank you so much for your prompt and informative reply. I'm not creating any threads, but the privilege drop was the issue.

I now have the game running solid on a monitor and the LED display!

hzeller, 2017-12-22:
Cool!
I suggest to set the[`drop_privileges` in your default [`RuntimeOptions`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/led-matrix.h#L362)  you pass to CreateMatrixFromFlags() to `-1`, so that it can not accidentally be chosen by the user.

closed 2017-12-24 by irataxy