hzeller/rpi-rgb-led-matrix #461: How to make My new transformers(P10) working with Binding/Python/samples(Runtex.py)

Arsenik4444, 2017-12-24:
Hi Mr Hzeller, i just wanted to say thank you again for your Library and Help including other member in the group, i manage to make my 16x32 rgb 3 chains panel working all Demo but there still issues with the binding/python/samples(Runtex&others....), the display still displaying  funny please how can i get the new transformer included and working with binding/python/samples, thank you..

angelogoncalve, 2017-12-24:
Hi my friend. I see this issue (https://github.com/hzeller/rpi-rgb-led-matrix/issues/398) can be useful, I think is to put the tansformer after the line 180 but I do not how. If you know how can it is possible to do that please share your help my friend. Thank you for your atention.

angelogoncalve, 2017-12-24:
the file directory to put the transformer for work in python is /home/pi/rpi-rgb-led-matrix/bindings/python/rgbmatrix/core.pyx

angelogoncalve, 2017-12-26:
Hi the way to apply the transformer is to use Cython.
The module is just C++ code which is compiled in a way that Python understands.

angelogoncalve, 2017-12-29:
Hi. If you apply the SetPixel function that has the transformer in C file in Python using the SetPixel function for the matrix = RGBMatrix (number_rows/2, 2*number_of_panels, 1) that works and you can control the matrix using Python my friend.

Arsenik4444, 2017-12-31:
Hi Sir, thanks for your feedback and apologies for the late reply, i did not have a chance yet to continue with this project due to closed labs, i may attend to did around the 8/9 January 2018.
But did you manage to make it work at your side? if yes please may you send me your code than i can see and make sense of this, it may also be useful for me; thank you Sir.

angelogoncalve, 2017-12-31:
Hi Sir. You need to do this with your transformer and one panel 16x32 with scan rate 1:4:

from rgbmatrix import RGBMatrix

number_of_rows = 16
number_of_panels = 1
parallel = 1

display = RGBMatrix(number_of_rows/2, 2*number_of_panels, parallel)

def matrix_draw(x, y, red, green, blue):
        new_x = x
        new_y = y
        xoffset = 8 * (x / 8)
        int yoffset = ((y + 4) % 8) / 4
        new_x = x + 8 * yoffset + xoffset
        new_y = (y % 4) + 4 * (y / 8)
        display.SetPixel(new_x, new_y, red, green, blue)

while True:
    matrix_draw(0, 0, 0, 255, 0)

angelogoncalve, 2017-12-31:
The problem yet that I have is to scroll a text in python like have in examples in bydings (file runtext.py or image_scroller.py) using the transformation for the panel that we are using it not works so good enough I do not why my friend. If you get it please share your code. Thank you for your atention and help my friend.

angelogoncalve, 2017-12-31:
Hi. With scroll text also works with the transformation in python with success my friend.

Arsenik4444, 2018-01-08:
Hi Sir, thank you for your Feedback and congratulation, apologies for the Late Reply to your mail, the labs still closed and will open next year but please can you send me your folder or full code then i can have a look and see what you did and try make sense of it, thank you Sir

angelogoncalve, 2018-01-09:
Hi Sir. The transformation that you use in C++ use in python to set a pixel with the green color in the coordinate/point (x_column, y_row) = (0, 0) like this for your panel 16x32 with scan rate 1:4 (use the indentation correct using the PyCharm editor for windows to help for each block of Python code):

from rgbmatrix import RGBMatrix

number_of_rows = 16
number_of_panels = 1
parallel = 1

display = RGBMatrix(number_of_rows/2, 2*number_of_panels, parallel)

def matrix_draw(x, y, red, green, blue):
xoffset = 8 * (x / 8)
int yoffset = ((y + 4) % 8) / 4
new_x = x + 8 * yoffset + xoffset
new_y = (y % 4) + 4 * (y / 8)
display.SetPixel(new_x, new_y, red, green, blue)

while True:
x_column = 0
y_row = 0
red= 0
green = 255
blue = 0

matrix_draw(x_column, y_row, red, green, blue)

Arsenik4444, 2018-01-11:
Hi Sir, i have been going through the code you have send me but i just do have a question:
where can or were did you insert your Python transformer?
*/python/rgbmatrix/core.pyx and where?
*Runtex.py and where?
thank you and please i will be waiting for your feedback; thank you Sir.

angelogoncalve, 2018-01-11:
Hi Sir. First the transformer in C files is not need (you can put or not, this not change anything the library in python). Second you do not need change anything in the library in python. Third, you only create a python file and do the transformer in python for example for your panel as the same code that have the transformer.cc file for the function SetPixel.

Here is the code in Python for your transformer: 

For you set a pixel with the green color in the coordinate/point (x_column, y_row) = (0, 0) like this for your panel 16x32 with scan rate 1:4 (use the indentation correct using the PyCharm editor for windows to help for each block of Python code):

from rgbmatrix import RGBMatrix

number_of_rows = 16
number_of_panels = 1
parallel = 1

display = RGBMatrix(number_of_rows/2, 2*number_of_panels, parallel)

def matrix_draw(x, y, red, green, blue):
xoffset = 8 * (x / 8)
int yoffset = ((y + 4) % 8) / 4
new_x = x + 8 * yoffset + xoffset
new_y = (y % 4) + 4 * (y / 8)
display.SetPixel(new_x, new_y, red, green, blue)

while True:
x_column = 0
y_row = 0
red= 0
green = 255
blue = 0

matrix_draw(x_column, y_row, red, green, blue)

angelogoncalve, 2018-01-13:
Hi Sir. Where (the link from the manufacturer) did you buy your panel hub75 P10 outdoor scan rate 1/4 16x32 pixels RGB Led?

Arsenik4444, 2018-01-14:
Hi Sir, it's the university which did buy the panels but i will try to ask and check it for you; if i get the link i will send it, thank you

angelogoncalve, 2018-01-14:
Hi Sir, thnak you for your atention. I wait for your response about where did buy the university panels 16x32 please. Many thanks my friend.

hzeller, 2018-01-21:
Can you try the new `--led-multiplexing` option described in https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing ?
It is also supported in Python, so you might not need even need an own transformer.
(needs a fresh checkout and compile)

Arsenik4444, 2018-01-21:
Hi Mr hzeller; thank you for you feedback and attention; i will try that and give you a feedback when tested; thank you.

Arsenik4444, 2018-01-21:
Hi Mr Angelogoncalve, apologies regarding the P10 Matrix 16x32 Rgb i had to find the link for you where these were bought ; i could not find the link due to the fact that the person who ordered the matrix is not working anymore at the university; but i think someone in the forum may assist; tank you Sir.

angelogoncalve, 2018-01-21:
Hi Sir. Thank you for your atention. I will test also the new --led-multiplexing option for the panels 32x32 with scan rate 1/8 and give you the feedback. Many thanks for you my friend and for Mr Henner Zeller.

angelogoncalve, 2018-01-23:
Each panel 32x32 with 1/8 scan rate I run with your library in Python and for:

The first case the correct program command line flags for the transformer:
--led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=1

The second case the correct program command line flags for the transformer:
--led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=2

But the second case I need also set the pixels columns and push eigth pixels columns to the top (y-8) and the others to the bottom (y+8) for half height panel. For the other half, do the same (y+-8) for the columns.

Dear Zeller is another way for the second case using only the flags? Thank you for your atention and your help.

hzeller, 2018-01-23:
It is not really clear to me what you do. Do you have two different kinds of panels and want to connect them in the same chain ? Or are you talking about two different installations ?

If you are mixing panels in the same chain, then this is a special case you'd have to handle yourself.

I don't understand the part with the pixel movement. Can you provide some pictures that illustrates what you attempt to do ?

angelogoncalve, 2018-01-23:
Dear Henner Zeller. Thank you for your atention. Yes I have two different kinds of panels outdoors 32x32 with scan rate 1:8 each one. 

Yes I am talking about two different installations:

First case) One panel works well with this flags:
--led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=1

Second case) The second panel works almost well with this flags:
--led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=2

But the second case I need also set the pixels columns and push eigth pixels columns to the top (y-8) if y<16 and the others to the bottom (y+8) if y<8 for half height panel. For the other half, do the same (y+-8) for the columns:

if y < 8:
     new_y = y + 8
elif y < 16:
     new_y = y - 8
elif y < 24:
     new_y = y + 8
else:
     new_y = y - 8

Third case) I have also a panel 16x32 panel indoor with scan rate 1/8 that has the same problem as the second case:

--led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=0

if y < 8:
     new_y = y + 8
else:
     new_y = y - 8

Dear Zeller is another way for the second and third cases using only the flags values options? Thank you for your atention and your help.

hzeller, 2018-02-17:
You really only should connect one type of panel to the same Pi, but if you want you have to write your own transformer that handles different panels in the chains differently.

closed 2018-02-17 by hzeller

angelogoncalve, 2018-02-18:
Yes I am testing only one type of panel.

Case) The panel works almost well with this flags:
--led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=2

But it is need also set the pixels columns and push eigth pixels columns to the top (y-8) if y<16 and the others to the bottom (y+8) if y<8 for half height panel. For the other half, do the same (y+-8) for the columns:

if y < 8:
new_y = y + 8
elif y < 16:
new_y = y - 8
elif y < 24:
new_y = y + 8
else:
new_y = y - 8

For example to writing a number in the panel without the auxiliary function for the columns pixels, if for example I want to write the number "5" in the second case panel, the superior half of the number appears in the bottom of the panel, the inferior half of the number appears in the top of the panel, thats why I need also a function to put the columns pixels in the wrigth positions (y +-8).

Another case) I have also a panel 16x32 panel indoor with scan rate 1/8 that has the same problem as the second case:

--led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=0

Need also this:

if y < 8:
new_y = y + 8
else:
new_y = y - 8

Dear Zeller is another way for these cases using only the flags values options? Thank you for your atention and your help.