hzeller/rpi-rgb-led-matrix #463: mmap error -1 ?

fxw, 2018-01-04:
Hello guys,

I need help to connect a LED Matrix 32x16 on a Adafruit HAT with RPI2...

After compiling, i'm trying to launch the first test but it's a failed...

sudo ./led-matrix -r 32 -c 2 -t 60 -D 1 runtext.ppm 
mmap error -1

Any suggestions about debuging that ? Is it a problem with the Adafruit and all soldering made or may be the cable between the adafruit HAT and the LED matrix ? 

Many thanks for your help

hzeller, 2018-01-04:
The program has not been called `led-matrix` for a long time, but `examples-api-use/demo`, so I suspect you are using a very old version of this library.
If you are using a Raspberry Pi 3, that will not work as there are some detection issues regarding the GPIO mmap area; please do a fresh checkout and go from there.

If you have an Adafruit HAT and not use the native wiring, you need to give the command line option `--led-gpio-mapping=adafruit-hat` (see https://github.com/hzeller/rpi-rgb-led-matrix#if-you-have-an-adafruit-hat ).

fxw, 2018-01-05:
Hello,

Many thanks for your help, but it's not working with that also..

Below some details...

I'm using a RPI2, the adafruit HAT and the matrix was bought on aliexpress and I compiled with the option HARDWARE_DESC=adafruit-hat...

root@pi:~/rpi-rgb-led-matrix-master# sudo make clean
rm -f *.o  led-matrix minimal-example text-example rgbmatrix.so
make -C lib clean
make[1]: Entering directory '/root/rpi-rgb-led-matrix-master/lib'
rm -f gpio.o led-matrix.o framebuffer.o thread.o bdf-font.o graphics.o librgbmatrix.a
make[1]: Leaving directory '/root/rpi-rgb-led-matrix-master/lib'
root@pi:~/rpi-rgb-led-matrix-master# sudo make 
g++ -Iinclude -Wall -O3 -g -fno-strict-aliasing -DADAFRUIT_RGBMATRIX_HAT -c -o demo-main.o demo-main.cc
make -C lib
make[1]: Entering directory '/root/rpi-rgb-led-matrix-master/lib'
g++ -I../include -Wall -O3 -g -DADAFRUIT_RGBMATRIX_HAT -c -o gpio.o gpio.cc
g++ -I../include -Wall -O3 -g -DADAFRUIT_RGBMATRIX_HAT -c -o led-matrix.o led-matrix.cc
g++ -I../include -Wall -O3 -g -DADAFRUIT_RGBMATRIX_HAT -c -o framebuffer.o framebuffer.cc
g++ -I../include -Wall -O3 -g -DADAFRUIT_RGBMATRIX_HAT -c -o thread.o thread.cc
g++ -I../include -Wall -O3 -g -DADAFRUIT_RGBMATRIX_HAT -c -o bdf-font.o bdf-font.cc
g++ -I../include -Wall -O3 -g -DADAFRUIT_RGBMATRIX_HAT -c -o graphics.o graphics.cc
ar rcs librgbmatrix.a gpio.o led-matrix.o framebuffer.o thread.o bdf-font.o graphics.o
make[1]: Leaving directory '/root/rpi-rgb-led-matrix-master/lib'
g++ -Wall -O3 -g -fno-strict-aliasing demo-main.o -o led-matrix -Llib -lrgbmatrix -lrt -lm -lpthread
g++ -Iinclude -Wall -O3 -g -fno-strict-aliasing -DADAFRUIT_RGBMATRIX_HAT -c -o minimal-example.o minimal-example.cc
g++ -Wall -O3 -g -fno-strict-aliasing minimal-example.o -o minimal-example -Llib -lrgbmatrix -lrt -lm -lpthread
g++ -Iinclude -Wall -O3 -g -fno-strict-aliasing -DADAFRUIT_RGBMATRIX_HAT -c -o text-example.o text-example.cc
g++ -Wall -O3 -g -fno-strict-aliasing text-example.o -o text-example -Llib -lrgbmatrix -lrt -lm -lpthread
g++ -Iinclude -Wall -O3 -g -fno-strict-aliasing -DADAFRUIT_RGBMATRIX_HAT -c -o rgbmatrix.o rgbmatrix.cc
g++ -s -shared -lstdc++ -Wl,-soname,librgbmatrix.so -o rgbmatrix.so rgbmatrix.o -Llib -lrgbmatrix -lrt -lm -lpthread

But I got the same error...

root@pi:~/rpi-rgb-led-matrix-master# ./minimal-example 
mmap error -1

root@pi:~/rpi-rgb-led-matrix-master# sudo ./text-example -f fonts/8x13B.bdf -y1 -x1 -c1 -r16 -C0,0,255
mmap error -1

Any another(s) idea(s) ?

hzeller, 2018-01-05:
what operating system image are you using?

fxw, 2018-01-05:
uname -a
Linux pi 4.9.0-2-rpi2 #1 SMP Raspbian 4.9.13-1+rpi3 (2017-05-18) armv7l GNU/Linux

hzeller, 2018-01-05:
Can you add a line `perror("mmap: ")` right at the beginning of the `if` block here

https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/gpio.cc#L191

.. re-compile and run again ? It should then output a more detailed message. Please update the bug here with that.

Maybe there is a weird reason that your Pi is not properly recognized ? Can you tell me what the output of
`cat /proc/cmdline`
is printing for you ?

hzeller, 2018-01-05:
Also, don't use the -DADAFRUIT_RGBMATRIX_HAT thing, that is an ancient compile option. It should actually have output a warning in the compile.

Either use HARDWARE_DESC=adafruit-hat in the [Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L26) or simply use the `--led-gpio-mapping=adafruit-hat` command line flag, no need to mess with it at compile time anymore.

Looking at your compile output, you are still compiling a very old version. There is for instance the hardware-mapping.o compilation missing or options-initialize.o. Make sure to have a fresh checkout of this library.

The minimal example and the text-example are in the [examples-api-use/](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use) directory, so you have to go in there first before executing the binary; not sure what you are doing above.

So long story short: make sure to have a current checkout of this library, otherwise it is hard to debug what is going on.

fxw, 2018-01-05:
root@pi:~/rpi-rgb-led-matrix/examples-api-use# cat /proc/cmdline
bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline root=/dev/mmcblk0p2 rootfstype=ext4 rootwait

fxw, 2018-01-05:
I checkout the lastest version of the project, compile with HARDWARE_DESC=adafruit-hat in the Makefile...
Compile the example project, i got now : 

root@pi:~/rpi-rgb-led-matrix/examples-api-use# ./demo -t 10 -D 1 runtext.ppm
mmap: : Operation not permitted
mmap error 0xffffffff

fxw, 2018-01-05:
root@pi:~/rpi-rgb-led-matrix/lib# ls -lrth
total 2,4M
-rw-r--r-- 1 root root 3,9K janv.  5 22:26 graphics.cc
-rw-r--r-- 1 root root 5,3K janv.  5 22:26 framebuffer-internal.h
-rw-r--r-- 1 root root  17K janv.  5 22:26 framebuffer.cc
-rw-r--r-- 1 root root 4,8K janv.  5 22:26 content-streamer.cc
-rw-r--r-- 1 root root 6,4K janv.  5 22:26 bdf-font.cc
-rw-r--r-- 1 root root 1,9K janv.  5 22:26 utf8-internal.h
-rw-r--r-- 1 root root 6,7K janv.  5 22:26 transformer.cc
-rw-r--r-- 1 root root 1,9K janv.  5 22:26 thread.cc
-rw-r--r-- 1 root root  17K janv.  5 22:26 options-initialize.cc
-rw-r--r-- 1 root root 7,8K janv.  5 22:26 led-matrix-c.cc
-rw-r--r-- 1 root root  14K janv.  5 22:26 led-matrix.cc
-rw-r--r-- 1 root root 1,4K janv.  5 22:26 hardware-mapping.h
-rw-r--r-- 1 root root 6,9K janv.  5 22:26 hardware-mapping.c
-rw-r--r-- 1 root root 6,4K janv.  5 22:27 Makefile
-rw-r--r-- 1 root root   60 janv.  5 22:27 compiler-flags
-rw-r--r-- 1 root root  19K janv.  5 22:31 gpio.cc
-rw-r--r-- 1 root root 112K janv.  5 22:31 gpio.o
-rw-r--r-- 1 root root 137K janv.  5 22:31 led-matrix.o
-rw-r--r-- 1 root root 105K janv.  5 22:31 options-initialize.o
-rw-r--r-- 1 root root  98K janv.  5 22:31 framebuffer.o
-rw-r--r-- 1 root root  12K janv.  5 22:31 thread.o
-rw-r--r-- 1 root root  82K janv.  5 22:31 bdf-font.o
-rw-r--r-- 1 root root  44K janv.  5 22:31 graphics.o
-rw-r--r-- 1 root root 113K janv.  5 22:31 transformer.o
-rw-r--r-- 1 root root  70K janv.  5 22:31 led-matrix-c.o
-rw-r--r-- 1 root root 3,9K janv.  5 22:31 hardware-mapping.o
-rw-r--r-- 1 root root  57K janv.  5 22:31 content-streamer.o
-rw-r--r-- 1 root root 844K janv.  5 22:31 librgbmatrix.a
-rwxr-xr-x 1 root root 529K janv.  5 22:31 librgbmatrix.so.1

fxw, 2018-01-06:
@hzeller any suggestions ? :(

hzeller, 2018-01-07:
It sounds like you have an unusual set-up in which your Pi does not allow to memory map the right area in memory. Can you try a current Raspbian image on your Pi to see if that changes ?

fxw, 2018-01-07:
Hey Henner ! You rock ! It's working very well with a fresh install 👍 💯 

Now i will play all the night with that :) Thanks !!!

closed 2018-01-20 by hzeller