hzeller/rpi-rgb-led-matrix #464: Led Matrix display doesn't display blue color on lower half

dannielum, 2018-01-07:
Hi, I am having problem getting the display board to display blue color on the lower half of the led matrix board. Please see the image below:
[image]

The bottom half also seem to always display wrong color except red and green. For example, white color (255,255,255) will be displayed as yellow on the bottom half. I checked my soldering and connection and everything seem to be connected just fine.
[image]

Please help! Thanks.

hzeller, 2018-01-07:
So it must be the wiring. The lower blue is controlled by GPIO23, which is pin 16 on the PI. So you might want to trace that in particular. Also make sure that your IDC cable is working (it is on Pin 7 there).

If you have a scope, you can also see if something comes out of pin 7 on the IDC on the Adafruit Hat.

dannielum, 2018-01-07:
Wow, you are awesome! I checked pin 16 and found I have the soldering lead connected to pin 8. Pin 7  on the IDC also connected with its neighbor pin. Disconnecting them fixed the issue. Thanks a lot!

One other issue I have is flickering. If I use a darker setting, there will be a lot of flickering with the display. Only 100 brightness can avoid the flickering. I tested with connecting GPIO4 and GPIO18 with a jumper wire (but I didn't solder it just for testing out), and used `--led-gpio-mapping=adafruit-hat-pwm` setting, but that didn't help resolving the issue. Do you think I have to solder the jumper wire to test it correctly? Any other suggestion? Thanks again.

hzeller, 2018-01-07:
the PWM change should reduce brightness glitches, but it won't change the overall refresh rate (you can see that with the `--led-show-refresh` option). Permanently soldering in the wire for the Adafruit PWM option and using the `--led-gpio-mapping=adafruit-hat-pwm` is in general a good idea though.

If there is some heavy work going on the Pi with network traffic and memory intensive stuff, it will also start to flicker as the internal memory bandwidth in the Pi seems to be limited.

dannielum, 2018-01-07:
I have tested a few different settings and didn't get to the best result yet. Although I got it to an acceptable configuration. Tweaking it seems to be quite challenging. Thanks for your responsive replies. and awesome project!

hzeller, 2018-01-07:
If you just display text like in the example above `--led-pwm-bits=1` might get you the highest refresh rate; but then you only get 8 colors...

dannielum, 2018-01-07:
`--led-pwm-bits=1`, `2`, and `3` doesn't seem to display anything at all for me. However, after playing with the numbers, `--led-pwm-bits=4` seems to keep the flickering to minimal for me.

hzeller, 2018-01-07:
Maybe you have color values below that threshold, so they turn out to be black. You can also play with the `--led-pwm-lsb-nanoseconds` value to get more refresh, but often that pushes the limit of what the RGB panels can do. Also, if your panel is fast, you can try to set `--led-slowdown-gpio=1` or even zero, but often they can't deal with it (nothing breaks if you try it, so worth a shot).

All you can watch with `--led-show-refresh` how your refresh rate changes.

closed 2018-01-20 by hzeller