hzeller/rpi-rgb-led-matrix #465: No qound from video-player

tibo2noix, 2018-01-08:
Hi Heller,
I have a 128x128 Px display (4 panels of 64x64 Px) and drive them with your active board.
I got as well a USB sound adaptator to be able to display the sound of videos.
As the sound wasn't working, I followed [this tutorial](https://superuser.com/questions/989385/how-to-make-raspberry-pi-use-an-external-usb-sound-card-as-a-default) to install the usb adaptater, which is not really plug & play.
To test, I downloaded a video on youtube in .mp4.
When I want to launch a video with your video-viewer, the image is displayed correctly on the panels but no sound comes out. When I open this same video on Youtube, I have the image on my PC screen (not the LED display) and the sound. 

When I run the audio test with 
speaker-test -c2 -D plughw:0,0
it also works fine.
The on-board sound is completely unistalled, no more need of "--led-no-hardware-pulse", i have only the USB board active.

Do you have any idea where this could come from ?

Thank you in advance

hzeller, 2018-01-08:
The video viewer does not do any sound for simplicity, so if you want sound you need to add that to the code.

tibo2noix, 2018-01-08:
Hi Heller, thanks for the fast answer.

I am beginner so I won't be able to add anything to your library.

Is there another way to do that ? It seems that someone did in in [#379](https://github.com/hzeller/rpi-rgb-led-matrix/issues/379)

hzeller, 2018-01-08:
They just added sound to the Pi, not played videos.

Playing video is just a demo app that uses the library, so there is nothing that needs to be added to the library.
It is essentially a copy from a tutorial how to use the video decoding libs. If you want to add audio, have a look at the link at the top of the source code pointing to the tutorial, then follow the tutorial along to add audio.

I am usually just playing remotely video on the RGB matrix, then some other, more beefy, computer can take care of the decoding (using the FlaschenTaschen protocol and the vlc-player that natively supports FlaschenTaschen).

closed 2018-01-20 by hzeller