hzeller/rpi-rgb-led-matrix #466: Control matrix 64x32 with mbi5153 (icn2053) led driver ic

kbateam, 2018-01-08:
Does anybody sucessfull make control  led matrix 64x32 with mbi5153 (icn2053) led driver ic
New matrix P5 64x32 comes with new IC led driver (Same as P2.5 matrix)
P5 Matrix with mbi5038 (icn2038)  ic led driver works fine on this project (only transform passed)
Any idea ?
Thanks.

hzeller, 2018-01-08:
I have not seen such panels yet. Do you have a link to a source ?

kbateam, 2018-01-08:
I can make photo pcb of this matrix.
So. i parsed some datatsheet of mbi IC
Signal OE is not use. It's replaced by GCLK,
this ic led driver has hardware 16bit (or 14bit) PWM on each other output (0..15)
As I understand chip need to be preload configuration every time as power up. (load some paramters such scan rate 1/4, 1/8 e.t.c)
also every time after load data on SDI cycle, need load PWM data for each other output(0..15) 
and after that need send vsync command to latch data to outout pins.
P.S Can upload some datasheet which i found on internet.

kbateam, 2018-01-09:
[http://www.xlix.ru/datasheet/MBI5038.pdf](url)
[http://www.xlix.ru/datasheet/icn2038s.pdf](url)
[http://www.xlix.ru/datasheet/MBI5153.pdf](url)
[http://www.xlix.ru/datasheet/mbi5152.pdf](url)
[http://www.xlix.ru/datasheet/mbi5152_appnote.pdf](url)

raphy1234, 2018-01-10:
Hi @kbateam this is off topic but I would like to know if the old p5 display you bought was [this one](https://www.aliexpress.com/store/product/P5-led-video-panel-rgb-full-color-320mm-160mm-1-8-scan-outdoor-SMD-3in1-led/103659_32784851966.html?spm=2114.12010612.0.0.bdce635SM0Tr9) or an equivalent? This is because I want to buy this one. The drivers are the same so I think they are the same but I am not sure. 

Also, what does only "transform passed" mean in your first post?

I am sorry this is off topic but I could not find your email or any other way of contacting you :|

Thank you for your help in advance

kbateam, 2018-01-10:
To @raphy1234 
Follow by your link is different hardware P5 outdor matrix module that i have.
With china production is difficult to buy right matrix module.
[On offfical site macroblock owner MBI tm is not present MBI5154 ic chip.](http://www.mblock.com.tw/products.php?search=mbi5154)
It seems to be MBI5153 ic chip.

> Also, what does only "transform passed" mean in your first post?

It means for right working this led matrix module on project which wrote hzeller need wrtie
logical transofrmation procedure, cause matrix module splited on logical modules (for expample  2x(32x32), 4x(32x16) e.t.c)

Here is my transform for modules that i have P5 64x32 1/8scan outdoor based on icn2038 ic led driver (MBI5038)
[transformer.cc](http://www.xlix.ru/src/transformer.cc)
[led-matrix.cc](http://www.xlix.ru/src/led-matrix.cc)

kbateam, 2018-01-10:
Countinue for my theme
If someone have any idea, will very grateful.

hzeller, 2018-02-27:
Interestingly, on this document http://pmod86499.pic35.websiteonline.cn/upload/Q4F32V1.pdf there is a
`CS` mentioned on pin 16. It doesn't say if it is high or low logic. According to Google translate, the 消隐信号 label next to it mean "Blanking Signal".

Can someone who has such a panel try to pull 16 high, by connecting that pin with 5V ?

hzeller, 2018-02-27:
@birdtechstep can you try if changing the logic level on Pin 16 changes anything ?

birdtechstep, 2018-03-02:
@hzeller chip select (cs pin) active low

birdtechstep, 2018-03-02:
@hzeller  My friend dev. on stm32 said SetBits strobe hi on columns_-4 (led_width-4)

hzeller, 2018-03-02:
can you elaborate? The only place where things are strobed is already high logic.

birdtechstep, 2018-03-05:
add lib/framebuffer.cc (line 587 ) work.
``` edit
582      for (int col = 0; col < columns_; ++col) {
583        const gpio_bits_t &out = *row_data++;
584        io->WriteMaskedBits(out, color_clk_mask);  // col + reset clock
585        io->SetBits(h.clock);               // Rising edge: clock color in.
586
587        if (col == columns_-4) io->SetBits(h.strobe); // ICN2038S : Bird Techstep
588      }
589      io->ClearBits(color_clk_mask);    // clock back to normal.
```

hzeller, 2018-03-06:
Interesting. This looks like it requires further digging in the datasheet what they actually need there because it _looks_ like if it works that it is by accident. If we have decoded the full thing we can recreate it by making a new RowAddressSetter for this kind of panel.

Can others confirm that this improves the situation ?

hzeller, 2018-03-07:
Is there a place where I an buy a panel with that chip, so that I can look into supporting it with the library ?

hoangphan, 2018-03-07:
Great! @hzeller  and @birdtechstep : i did a quick check and above patch seems to work with mine (same 2038s led driver).

shall perform more test and let you know the result!

hoangphan, 2018-03-07:
> Is there a place where I an buy a panel with that chip, so that I can look into supporting it with the library ?

@hzeller : you can find the P3 that has ICN2038s led driver as following link,
https://www.alibaba.com/product-detail/GKGD-small-pitch-indoor-P3-RGB_60445963111.html

kbateam, 2018-03-08:
> what i'm curious is some body on the top comment of this topic, said with P5 64x32 the same code works with the same chip (2038s)

hzeller, 2018-03-10:
Without further datasheets that explain the protocol to the displays it is hart to tell, maybe what is happening underneath is that the strobe pulse actually is negative logic, not positive logic, so pulling up the signal before the strobe will create that situation.

hzeller, 2018-03-10:
In that case, swapping the lines in 
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer.cc#L595

from 

```
   io->SetBits(h.strobe);
   io->ClearBits(h.strobe);
```
to
```
   io->CleartBits(h.strobe);
   io->SetBits(h.strobe);
```

would help as well. Can someone who has such a display try if this works as well ?
(_without_ the `if (col == columns_-4) io->SetBits(h.strobe);` patch)

I need to get to the bottom of this so that we can support this in the standard library; the columns-specific thing looks very specific and probably not necessary.

Also, I still would like to know where to buy such a panel to do experiments with.

@hoangphan the ali-express shop you mentioned: their minimum order is 40 items which I don't have the money for.

hoangphan, 2018-03-11:
@hzeller : i checked but unfortunately the panel is not lighted up. stay dark without the line `if (col == columns_-4) io->SetBits(h.strobe);`

from the datasheet as link http://www.xlix.ru/datasheet/icn2038s.pdf I captured the following diagram, and trying to map it with the way you programmed software,

but just want to ask a curious question: is dual latching approach of led driver supported by your library?

[image]

hoangphan, 2018-03-11:
@kbateam : how did you get the datasheet for these led drivers? i looked into the icn2038s datasheet, but it seems too many typo and confusion. is it possible to get the official datasheet from the Chinese company (i.e. Chipone)?

hoangphan, 2018-03-11:
@hzeller : i think i know the reason why `if (col == columns_-4) io->SetBits(h.strobe);` patch works.

with the attached snapshot of datasheet, the STROBE pin of panel (that has ICN2038s led driver) - is not simply a STROBE pin, it serves as a command parser, which also cover the LATCH data in.

as you can see, in order to latch a data out, we need to pulse the strobe pin high in the 3rd clk pulse.

that's why the patch above works.

let me know if you agree with this. and if yes, kindly share your thought about how to make this patch generic into your standard library.

[image]

raphy1234, 2018-03-11:
@hzeller, here are some panels that have icn2038 make sure to pick the icn2038 option though https://m.aliexpress.com/item/32784851966.html?trace=wwwdetail2mobilesitedetail&productId=32784851966&productSubject=P5-led-video-panel-nationstar-rgb-full-color-320mm-160mm-1-8-scan-outdoor-SMD-3in1&spm=a2g0s.9042311.0.0.ZfkLbb

kbateam, 2018-03-11:
I got it form china resaler
Look MBI5038.pdf , ICN2038 hard copy from original driver
> how did you get the datasheet for these led drivers

hoangphan, 2018-03-12:
I see there is a key of similarity between the icn2038s (pay attention to the S) and icn2053: they have LE pin functions as a command parser, as below snapshot. 

If LE is raised after 12 CLK pulse, it means enable all output channels. 

The same for icn2038S, if LE is raised after 12 CLK pulses, data is copy to reg2 - which relevant to the dual latch feature of the ASIC

[image]

hoangphan, 2018-03-12:
Attached the snapshot for dual latch feature on icn2038s. By accident, the value 12 is the key to functional of both ASICs 
[image]

hoangphan, 2018-03-13:
@kbateam : are you able to make your board (with ICN2053) working with the provided patch ?`if (col == columns_-4) io->SetBits(h.strobe);`

kbateam, 2018-03-14:
@hoangphan 
Tested  both variants, result negative :(

hoangphan, 2018-03-15:
@kbateam : would you try to tweak around the number of CLK rising edge when STROBE pin is high?
because from the datasheet of ICN2053, it looks like if you want to deliver the data to buffer of the IC, this number should be 2 or 12.

can you just tweak this around from 1 to 15 and to just see if the panel lights up?

birdtechstep, 2018-03-21:
http://www.chiponeic.com/content/details11_293.html
Top-Level PWM - Leading a Small Pitch LED "Core" Standard with the ICN2053

arktrin, 2018-04-03:
Hello everyone. I also have these ugly 64x32 Qiang Li LED modules with ICN2053 drivers. I even managed to connect one of them to Huidu HD-C10 controller (almost), but I still cannot understand how to control these LED modules without HD-C10 controller. What should I do to solve the problem?

hzeller, 2018-04-03:
Did you try the proposed change in lib/framebuffer.cc ( https://github.com/hzeller/rpi-rgb-led-matrix/issues/466#issuecomment-370287035 ) @arktrin ?

DonDalle, 2018-08-29:
I also have a ICN2053 Panel. I applied both patches and it did not resolve the problem. I get some outputs on the panel but they are glitched. If re reboot the Pi the outputs won't show again. I have to unplug the power supply again.

pathosDev, 2018-12-15:
Hi everyone,

I have the same problem like @foliastar. I also have a 64x64 ICN2053 led panel and nothing worked so far. I tried every "patch" mentioned above but no results. For the most time the panel stays completely blank. After playing a long time with parameters and stuff I managed to get some bugy and flickering output (random squares and lines, mostly red) as you can see here:

[image]

Did anyone manage to get these panels working?

_PS: I have the Adafruit RGB Matrix Bonnet._

Greetings, Marcel

arktrin, 2018-12-16:
Here the output of Huidu HD-C10 controller that somehow can work with ICN2053:
[image]
Hope this will help someone.

g3gg0, 2019-01-29:
@arktrin do you happen to still have access to the signal?

need some information about the OE tick count.

SebiTimeWaster, 2019-03-14:
is this thread dead? did someone find a signal diagram or something? the picture above is not helpful with timings / counts...

VLVOZ, 2019-03-17:
> @hzeller chip select (cs pin) active low

``

> @hzeller My friend dev. on stm32 said SetBits strobe hi on columns_-4 (led_width-4)

Hello, can someone help with my problem? I have the driver ICN2053 on the LED panel, I need to know the contents of the configuration registers and the algorithm for displaying the image on the LED panel using these ICN2053

ElectronicsInFocus, 2019-03-28:
icn2053 need 138 clocks on OE pin in each scan period.
I successfully run it on ESP32.
Video (sources in description): https://www.youtube.com/watch?v=nhCGgTd7OHg

SebiTimeWaster, 2019-03-28:
@ElectronicsInFocus AWESOME! i am ordering an ESP32 now. 

I hope google translate can translate your comments ok ^^

SebiTimeWaster, 2019-04-02:
so, i was able to build it, had to fix some include paths from windows to linux notation, and in a newer version of ESP-IDF the mrt package moved from /mrt to /driver, but all in all easy.
flashing only works by holding the boot button on my SHC-NodeMCU-ESP32 module.

i think the pinout is totally different on my module, thats why i got this out:

[image]

i have to read up further about it, but i can already see the intended pattern, kinda... will continue tomorrow.

arktrin, 2019-04-03:
@SebiTimeWaster can you share your ESP-IDF linux project with pinout, please?

SebiTimeWaster, 2019-04-03:
so, it looks now like this:
[image]

the upper block has an issue with green (some stuck on full, most on off), but this seems to be an initialization problem, i swapped panels and rgb lines around and it is always green that shows like this.

the code from @ElectronicsInFocus is by no means complete, there is for instance an 8 to 16 bit mapping missing. but as a proof-of-concept it works nice!

@arktrin sure, here you go:
follow these two setup guides:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html
in https://docs.espressif.com/projects/esp-idf/en/latest/get-started/#get-started-get-packages  you do not need Step 7 if your ESP32 does use /dev/ttyUSB0 and Step 5 is replaced with the download:
unzip [nadyrshin_ryu-icn2053_esp32_demo.zip](https://files.timewaster.de/nadyrshin_ryu-icn2053_esp32_demo.zip) into the /esp/ directory
then in the "nadyrshin..." directory run "make flash"

that should be it.

SebiTimeWaster, 2019-04-21:
Here you go: [ICN2053_ESP32_LedWall](https://github.com/SebiTimeWaster/ICN2053_ESP32_LedWall)

much more information will follow, for now i'll watch Aquaman ^^

SebiTimeWaster, 2019-04-22:
Here the finished library (although i have a couple of additional feature ideas): [ICN2053_ESP32_LedWall](https://github.com/SebiTimeWaster/ICN2053_ESP32_LedWall)

Most information is in the repo's readme, just a few comments here:

The ESP32 is not fast enough, partially because of how the ICN2053 chips work. Since the chip is saving 16 x 32 pixels of gray scale 16 bit data in its internal memory and there is no pin to control how the data is outputted it depends on the incoming data stream to be uniform and exactly as specified to use this as cues to output its saved data. the generation of the clock data on the OE pin (At 12 Mhz) takes a lot of power away from outputting data, which reduces real FPS. so i think on one hand the chips have the advantage over the passive chips used in older panels to generate much better picture quality, but on the other hand it takes a lot of processing power to generate all the signals needed (in the quality needed), MUCH more than with the old style of chips. so overall i think these panels are much better suited for professional systems (Like the video processing systems one can buy from china), but not so much for private tinkering, although with my code it is absolutely possible to do graphics and text where the picture quality is not as important.

Finding all this stuff out was a pain in the but. taking over the code from @ElectronicsInFocus it looked all easy and i thought hey, ESP32 can do hardware PWM generation that makes it much faster! but after a lot of reading and trying out finding out that turning the hardware generation on and off has a massive delay and that the ICN2053 needs an EXACT clock cycle (meaning just putting out a couple more cycles produces a garbled image) meant i could not utilize that at all and was pretty frustrating. 

Finding out what pins i could use was also frustrating, my board has a GPIO13 that does not exist on any pin mapping i found, GPIO's 34 to 39 cannot be used for output even though they are named GPIO, GPIO 12 + 13 can be used, but do something special on programming, and pin 4 failed for me for no apparent reason.

What is really nice about the ICN2053 chips is that the output refresh "fps" is at 16 times the real fps, so even if you only reach 50 real FPS the refresh rate is actually 800 fps, which makes the output absolutely flicker free.

Buying a Logic Analyzer and to see what is happening on the lines was the only thing that enabled me to finish this project.

SebiTimeWaster, 2019-04-23:
so @hzeller will you adapt the code to your rPi library even though it works completely different to the other panels you support?

kimyunil, 2019-06-04:
Any update for driving this 128x128 panel with rpi?
https://www.aliexpress.com/item/P1-875mm-Indoor-SMD-Full-Color-Led-Module-240x240mm-High-Resolution-Monitoring-Room-LED-Video-Display/32989607179.html

kimyunil, 2019-06-11:
As @hzeller said,  
```
   io->CleartBits(h.strobe);
   io->SetBits(h.strobe);
```
seems to works, without 
```if (col == columns_-4) io->SetBits(h.strobe);```

But still led is glitching.

kimyunil, 2019-06-11:
Do I have to apply @kbateam patch file?

andrei0807, 2019-08-21:
@kbateam , @hzeller 
How to fix library for ICN2053?

andrei0807, 2019-08-23:
@ElectronicsInFocus 
How to fix library for ICN2053?
I have 30x30 1/3 scan mode.
How to set strobe pin?

daveythacher, 2020-04-14:
A lot of these are similar to the TLC5958. They are conceptually different from the standard SPI shift register. I recommend using PIC24/PIC32 controller to act as bridge interface. Have it handle all the interfacing stuff. Build simple SPI protocol, and use SPI-dev.

They use a command set which could be worked out with a PIC24 SPI using variable SPI. I want to say it counts the cycles of SCK (up to 25 MHz)  while LAT asserted. Change the word length of SPI and write.

For the GCLK (up to 33MHz) use SPI or PWM that supports counting cycles. Use DMA to feed if need be.

Command set SPI/DMA could be used for data transfer. Closer inspection of E-PWM/S-PWM reveals a small hazard. If using reduced PWM bit count say 8 bit instead of 16 since 2^16 would eat the FPS of high multiplexed configurations. (Which is why macroblock also sells them in 1:8 configurations.) Then you will need to issue reset PWM counter command. Which would conflict with data transfer.

Supposedly they claim the E-PWM/S-PWM makes up for this and the trick can pass off. Even though the FPS is 15.7. However they claim its 4,028.3. The benefit of 16 bit is look up tables can be used to smooth out LED steps. (Small table too 256x16.)

This would need to be worked through. One simple approach is to make controller stupid and correctly time transfer. Controller could use shallow memory copy, which would place limit on chain length. However bit banging would do that too.

This really is a job for a controller not processor. Since the work of PWM has moved over to hardware. I would not recommend patching this to the Pi. Personally I think this project should have used an offload controller all along.

I looked into this a few years back. Never was able to do much about it due to cost and lack of available of these panels. Plus I was annoyed about having to keep a shallow copy, since the dual buffers cannot talk. Serial interface nice to work with but also slow in terms of bandwidth. More than enough for this case. Just interesting they keep the high single clock, but I guess this was for HUB75 controller backwards compatibility. (Most are FPGAs.)

daveythacher, 2020-04-14:
Small correction: You would not be able to use SPI for data on HUB75 using these chips without rewiring the panel. Since it could be using 6 chips in parallel. You would be able to use SPI for the commands and GCLK. You would get into performance constraints of converting 6 shorts into bit streams. Coprocessors inside the processor could handle this or external conversion logic could to. (Assuming FPGA.)This could get a little weird,

daveythacher, 2020-04-16:
There is a small amount of variation between the SCLK and GCLK usage for commands. All use the LAT pin. These are the only two that are able to be used in parallel reliably. TLC5958 is centered around SCLK, while MBI5153 is centered around GCLK. The step counts per scan also varies. Some chips place min "dead time".

These are all things that you could take into account. I am not sure I would port in the ESP32 code, however it does appear pretty close. However some sections of it do not seem quite right. It is built for the ICN2053, which may be the MBI5153 rebranded. I looked over the original C code, and its structured well. However I am it would need a few corrections and modularizations for it to work in the model you have been following. Mostly with configuration flags, however some of the routines of the original C code could probably be ported.

There are app notes for TLC5958 and MBI5153, which would allow some prototyping. For larger panels these will likely be very common. Unfortunately the Pi will have much of the same issues it does with soft PWM. However the advantage here for the Pi is if the frames do not need to change the Pi can idle down.

If the images do not need to be redrawn then the Pi could conceptually support very long chains. I am still against bit banging. However for large frame rates generated by Pi the bit banging will likely shorten the change length of the Pi. 

I have never seen these panels. Do they have six data pins? I am guessing they just made OE -> GCLK and kept everything else the same. Conceptually the commands would be in parallel. Meaning the HUB75 connector structure still holds and the FPGA controllers that currently exist get reprogrammed assuming they do not have memory limitations.

I am curious if they will switch over to DDR3 and possibly chase after the Zynq. Its effectively a display so you could treat it as a video stream. Granted that would be for larger panel configurations. Security will possible be an issue at that point. However you could do direct stream forwarding.

0x1cedd1ce, 2020-05-14:
I have panels with icn2153 chips and ported the esp32 code from @SebiTimeWaster to the raspberry. Appart from some flickering it's working. My source is not optimized for speed, just a first working attempt.
[image]

[test.c.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/4629409/test.c.txt)

SebiTimeWaster, 2020-05-14:
Awesome! i have to try it out...

marcmerlin, 2020-05-19:
@birdtechstep adding
        if (col == columns_-4) io->SetBits(h.strobe); // ICN2038S : Bird Techstep
now at line 799, did not help.

I then tried @hzeller 's suggestion in
https://github.com/hzeller/rpi-rgb-led-matrix/issues/466#issuecomment-372053749

At line 809, I tried to swap those 2 lines as suggested by @hzeller 
      io->SetBits(h.strobe);   // Strobe in the previously clocked in row. 
      io->ClearBits(h.strobe);

This actually got the panel to light up and display garbage, then again garbage is better than dark :)
On a Pi4, adjusting --led-slowdown-gpio=0 made a measureable difference on what garbage was being dispayed (0 and 1 display more, 2 displays nothing, 3 displays a bit (3 squares), 4 and 5 display only a colum of 16 blue pixels on half the height,

I'm not sure if https://github.com/hzeller/rpi-rgb-led-matrix/issues/1070 is exactly a dupe from this bug since this bug is 2 years old and I just got a similar looking chip on a 128x64 panel.

marcmerlin, 2020-05-19:
[image]
[image]

daveythacher, 2020-05-20:
For reference:
https://www.neumueller.com/download/MBI5051%20MBI5052%20MBI5053%20Anwendungshinweis%20-%20Application%20Note.pdf

Player-1-TR, 2020-06-18:
> I have panels with icn2153 chips and ported the esp32 code from @SebiTimeWaster to the raspberry. Appart from some flickering it's working. My source is not optimized for speed, just a first working [...]

@0x1cedd1ce Sorry, I quite new to this matrix display thing. I looked your text.c code and found the lookup table named Translate8To16Bit[256]. It looks like icn2153 IC can process 16 bit depth of color but we using it only for 8 bit depth.
I'm using MBI5051 which also have 16bit PWM data input. I have to calculate the same lookup table like yours. I'm curious on how you calculate this table. Also Translate8To16Bit[1] = 46 but MBI5152 App note writes SDI value must be greater than the 64. Is this value outputs anything?

Another question is, If we already upscale colors from 8 bit why don't we use IC in 14 bit mode instead? 14 bit mode requires low cycle numbers for displaying it thus lower freq for GCLK. What is the benefit of using IC on 16 bit mode?

[113 more replies]