hzeller/rpi-rgb-led-matrix #467: rare flickers

mzdunek92, 2018-01-09:
Hello,
I got your app running on clean instalation of Raspbian Stretch Lite on RPi3. My matrix consists of 2 parallel chains of 7 32x32 modules (p5). With PWM bits set to 4 I got stable framerate number of 320-360Hz.
Unfortunately there is random amount of flickers after different portions of time. For example everything can be fine for 4 minutes, after this there are 4 flickers in 3-4 seconds interval. In most cases flickers are very fast, lasts for 0,05 second. But sometimes flicker is longer like 0,1-0,3 second and it looks like total black display.
Main loop of my app is very simple:
```
void DrawBuffer::update(float dt)
{
	_frameBuffer->SetBrightness(_brightness);

	if (_backgroundColor.r() == 0 && 
		_backgroundColor.g() == 0 &&
		_backgroundColor.b() == 0)
	{
		_frameBuffer->Clear();
	}

	_frameBuffer->Fill(_backgroundColor.r(), _backgroundColor.g(), _backgroundColor.b());

	for (auto o : _objects)
	{
		o->draw(_frameBuffer);
		o->update(dt);
	}

        _frameBuffer = _canvas->SwapOnVSync(_frameBuffer);
}
```

What can I do to solve this issue?

hzeller, 2018-01-09:
Can you remove the comment in [`#DEFINES+=-DDISABLE_RT_THROTTLE` in the lib/Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L121), recompile, and see if that changes anything ?

hzeller, 2018-01-21:
So I was testing with a recent version of Raspbian Lite and it looks like it indeed induces some brightness fluctuations (see #483), which was less with older versions of Raspbian. Still need to investigate what the reason is.

giuseppedevito, 2018-02-02:
First of all many many thanks for your great job hzeller.
I was using your library since 6 months almost and I have noticed a constant flicker from the first
time I used it (I use an Adafruit HAT and all your recommendation).
Investigating different issues (this one and the #483 look very similar to the mine) and debugging
my system (Raspbian Lite 2017-04-10-raspbian-jessie-lite.img) I have noticed that problems arise
when the cron daemon does its jobs (every 1 minute), writing stuff in the syslog.
To be more accurate, running

# tail -f /var/log/syslog

every time the cron job sends something on syslog (every 1 minute in my case) the display
flickers (I'm using "flicker" referring to a brightness down - brightness up process that results
in a flicker for my eyes).

using the command line loop

# while [ 1 ]; do logger "test1111111111111111111"; sleep 1; done

I see a very short flicker (when the cron is executed the flicker is more heavy).

Disabling the cron

# systemctl stop cron

the flicker stops definitively :-) .
But the cron daemon is very important for me and I'm notso sure that the issue is
related to cron, to rsyslogd or both.
Do you have more information about this?

hzeller, 2018-02-04:
Yes it is known that the default kernel in the Raspbian distribution does mess a lot with flicker in the LED display, in particular if it does some IO operations, such as writing to a file or doing something in a terminal logged in remotely (even running `top` in a terminal generates a faint flicker every time it updates). So probably the syslog writing is part of the issue you see. Maybe have your job that runs in cron limit writing to syslog ?

It is not clear it this is due to some limited memory bandwidth on the Raspberry Pi hardware, or due to the non-realtime scheduling of the default kernel that comes with the distribution. Unfortunately, the Raspbian distribution does not provide a default realtime kernel to switch to (with the usual PREEMPT_RT, NOHZ, disable RCU tweaks), so this is up to the users at this point to compile that for themselves.
(It would actually be good if someone poked them and ask if they can provide a RT kernel as an alternative to switch to. Many projects using the Raspberry Pi - which is after all a hardware playground - can benefit from having such choice).

Anyway, I suggest to play with compiling an RT kernel (and report back). It might not work however, if this is due to some legitimate hardware limitation such as memory bandwidth of course.

closed 2018-02-17 by hzeller

hzeller, 2018-03-17:
Check out the new compile-time option in `FIXED_FRAME_MICROSECONDS` in [lib/Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L105). It will allow to time-buffer glitches created by the rest of the system.