hzeller/rpi-rgb-led-matrix #472: USB Audio Out Issue - Python bindings and asoundrc

irataxy, 2018-01-11:
Hello,

I've got mostly everything working on my project but the matrix library seems to be interfering with USB sound playback. I am using a .asoundrc file in the /root directory on a Pi 3.

When I run `speaker-test` from the terminal, everything works fine. I can also record with the USB mic. 

When I run a python [file](https://github.com/googlesamples/assistant-sdk-python/blob/master/google-assistant-sdk/googlesamples/assistant/library/hotword.py) and use the bindings in the file, I get the following error when trying to play back:

```
ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
[ERROR:audio_output_stream.cc(61)] Device could not be opened: default
[894:907:ERROR:audio_output_stream.cc(121)] An error occurred when playing audio output
```

The microphone is working fine; I can see my words being displayed on screen. I've tried two different USB audio outputs - a set of USB-only speakers and an external USB to mic/speaker adapter with no luck.

Any help would be greatly appreciated!

Hardware:
Adafruit RGB Matrix HAT
Raspberry Pi 3 Model B
Adafruit RGB 16x32 Matrix
Hardware hack (HARDWARE_DESC=adafruit-hat-pwm)
One core refresh (isolcpus=3)
Disabled on-board sound (dtparam=audio=off)

Software:
Raspian Stretch Desktop
SDL2 + SDL ttf

hzeller, 2018-01-11:
Not sure how I can help with that, that is somewhat outside the Matrix realm. I suspect that the alsa functions you call in the Phyton code ignores your .asoundrc.

hzeller, 2018-01-11:
also, never use a desktop distribution, only headless; some desktop distros do funny stuff with sound.

irataxy, 2018-01-12:
OK, I tried the Raspian stretch lite and I am still getting the same error on playback. In fact, when I
first create the matrix:

`matrix = RGBMatrix(options = options)
`
I am getting the following error:

[ERROR:file_watcher.cc(103)] poll() failed: Interrupted system call

and then later when I try to play back sound, I get the same error as before:

```
ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
[ERROR:audio_output_stream.cc(61)] Device could not be opened: default
[9408:9420:ERROR:audio_output_stream.cc(121)] An error occurred when playing audio output
```

It works fine when I comment out the line that creates the matrix.

Can anyone point me to a very simple sample that uses the matrix and mic + speaker on the rpi3?

Thanks!
Nick

hzeller, 2018-01-21:
There is probably not much we can help on this forum as this has nothing to do with the matrix but with the fact that the Alsa configuration still wants to access the BCM audio device even though the module is not installed; it looks like it tries to side-step the kernel driver and then blocks the resource the library needs.

Can you check with the people packaging the Rasbpian and ask them if you can use alsa also just with your USB sound-card without relying on the BCM device ?
Maybe that BCM device needs to be explicitly unconfigured and only left your USB device in there.

If you find a solution, please post it here, so that others know what to do.

Until then, you can disable the hardware pulsing (`disable_hardware_pulsing` in the options), to get things working, but then of course, you'll see more brightness flicker glitches.

hzeller, 2018-02-17:
Also see #493

closed 2018-02-17 by hzeller