hzeller/rpi-rgb-led-matrix #473: using python 3?

marwellit, 2018-01-12:
hi

I have a successful project running with 4 signs all on python 2 ..... I thought I'd try and migrate to python 3 so have tested but I get 

Traceback (most recent call last):
  File "sign_io3.py", line 3, in <module>
    from samplebase import SampleBase
  File "/home/pi/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/home/pi/rpi-rgb-led-matrix/bindings/python/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: /home/pi/rpi-rgb-led-matrix/bindings/python/rgbmatrix/core.so: undefined symbol: _Py_ZeroStruct

I'm afraid this is beyond my Python skills to resolve - what is the status of the library with Python 3?

thanks for any advice !

dave

hzeller, 2018-01-12:
You have to compile the Python library for version 3 first. See the Python binding page, it has a section for the different versions.

 It is slower than Python2, so only use if you have to.

marwellit, 2018-01-12:
ah....

thanks very much.....for prompt response

i'll leave it alone.....as I need the speed 8-)

dave

brianprats, 2018-01-16:
I'm having a similar issue. I have a program on Python 3 and have compiled as shown in the Python binding page. When I try "sudo ./counter.py" it claims no such file exists. When I run "sudo python ./counter.py" I get the error: 
File "./counter", line 6, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions, graphics
File "/home/pi/rpi-rgb-led-matrix/bindings/python/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: No module named core

I run into the same issue when I try to run any of the sample files. I've hit a dead-end and don't really know how to troubleshoot this. Any help would be much appreciated.

hzeller, 2018-01-20:
If you have to use Python3 and it doesn't compile, please send a pull request that fixes it.

hzeller, 2018-01-21:
So I have tried the compile instructions again on the README and they work very well
```
sudo apt-get update && sudo apt-get install python3-dev python3-pillow -y
make build-python PYTHON=$(which python3)
sudo make install-python PYTHON=$(which python3)
```
Make sure when you invoke your scripts to explicitly invoke them with python3

```
 python3 ./counter ...
```

The default python on the Pi is Python2, so if you don't explicitly invoke it with python3, it will silently try with python2 and fail to fine the libraries that you only compiled for 3.

closed 2018-01-21 by hzeller

dzshanno, 2018-06-03:
> Make sure when you invoke your scripts to explicitly invoke them with python3
> 
>  python3 ./counter ...

Thanks this saved me tearing my hair out - could it be added to the install pages?