hzeller/rpi-rgb-led-matrix #475: using led-image-viewer with a gif on a 16 by 32 LED Matrix

dumes20, 2018-01-13:
Hello,

I have been trying to upload a gif and get the utility to output based on the instructions. In the terminal my system appears to be successful, but there is no output on the display. Here is the output:

pi@raspberrypi:~/display16x32/rpi-rgb-led-matrix/utils $ sudo ./led-image-viewer -C -l4 --led-rows=16 music.gif --led-no-hardware-pulse
Size: 32x16. Hardware gpio mapping: regular
Loading 1 files...
Loading took 6.538s; now: Display.

The interesting point here is that when I compile the code for the test python script I am successful. So I believe that the actual code is indeed working. My physical connections appear to be in tact as well. 

Also I know that you mention the snd module on the pi. I have also taken care of this based on the troubleshooting document and the readme file.

Let me know if you have any ideas of what I can try. I will continue to play around

hzeller, 2018-01-13:
What documentation did you use to wire it up ? Are you using the [wiring diagram](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md) or use some adapter board ? E.g. if you are using the Adafruit HAT, don't forget to also give `--led-gpio-mapping=adafruit-hat`

dumes20, 2018-01-13:
I used the following: https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/experimental-python-code

I also tried this command with what you stated above
sudo ./led-image-viewer -l4 --led-rows=16 music.gif --led-gpio-mapping=adafruit-hat --led-no-hardware-pulse

hzeller, 2018-01-13:
Yea, that documentation is only for their python demo.

Please only use the documentation [within this project](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md) to use this project.

closed 2018-01-13 by hzeller

dumes20, 2018-01-13:
Thank you for the response. I will take a look and try again!