hzeller/rpi-rgb-led-matrix #476: 16x32 P10 Functionality

speedtek99, 2018-01-15:
Greetings - I have two P-10 16x32 1/8 scan panels that I've connected one on top of the other.  I'm using the pi 3 with the adafruit hat.

When running the demo (or any other example) the LED panel is only displaying the first 8 rows on each matrix.  There is no other activity happening.  I've checked all of my solder joints, read through the troubleshooting section, and searched here - but could not find any similar issues.

The command I am running is this:
sudo ./examples-api-use/demo -D5 --led-rows=32
Size: 32x32. Hardware gpio mapping: adafruit-hat

Any clues would be appreciated, thanks!

[image]

speedtek99, 2018-01-15:
Actually it has to be something related to the pi setup - I replaced the hat, and bought the adafruit 32x32 P6 display, and have the exact same issue - only the bottom half of the matrix lights up - and does not respond properly to the demos.  I'll report back when I find out more in case anyone encounters the same issue.

angelogoncalve, 2018-01-15:
Try do that for one panel 16x32: sudo ./examples-api-use/demo -D5 --led-rows=16 --led-chain=1 --led-parallel=1

For two panels 16x32 in paralllel:  sudo ./examples-api-use/demo -D5 --led-rows=16 --led-chain=1 --led-parallel=2

angelogoncalve, 2018-01-15:
Hi Dear Sir. Where did you buy your panels 16x32 P10 with scan rate 1/8, can you give me the link or the manufacturer contact please? Thank you for your atention and help.

speedtek99, 2018-01-15:
I've tried that - and sadly i get the same result, I've replaced the matrix hat, reinstalled and reconfigured the pi os,  and even used a different panel (the 32x32 adafruit) - and it does the exact same thing.  I think there must be an issue with the pi itself now.

As far as the panels, these were obtained on Amazon - if i can get things to work I can order the cheapo chinese ones for half the price though.   This is more of an experimental phase - https://www.amazon.com/gp/product/B074MZGS65/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

angelogoncalve, 2018-01-15:
Hello Dear Sir. Thank you for the link to obtain the panels 16x32 P10. But the one panel 32x32 you tried, his scan rate is 1/16 or 1/8? And the interface is hub75 for both (16x32 and 32x32 panels)? Thank you for your atention.

speedtek99, 2018-01-15:
That is correct, the 16x32 P10 is 1/8 scan, the 32/32 P6 is 1/16 scan - they are both hub75

speedtek99, 2018-01-16:
I tried with a new pi and all works as advertised, my old pi must have been overcooked!

closed 2018-01-20 by hzeller