hzeller/rpi-rgb-led-matrix #477: Split the led matrix to show timer and scrolling image

DebaGracias, 2018-01-15:
Hi All,
I have a requirement where I need to display a countdown timer on the left side and a scrolling image on the right side of a led matrix.How can I split the canvas(horizontally) to show a timer in the first part and a scrolling image in the second half.Can somebody provide me with a sample example?

Thanks,
Debasish

hzeller, 2018-01-16:
You essentially need to structure your code in a way that whenever you update your image to scroll (I suppose you have a big loop in which you do that), you then also update your timer text, possibly by first cleaning its background with a loop that sets black pixels. After everything is updated, then do the `SwapOnVSync()`.

The rest is left as an exercise for the reader :)

DebaGracias, 2018-01-20:
Thank u hzeller. I was able to split the screen to show the timer at one end and running text using runtext.py sample.But however when the text scrolls ,it runs over the timer which is placed at one end.I tried to run the text from canvas.width to 30(assuming thats the width occupied by the timer),but it just overlaps the timer.
Please guide me on how to scroll a long text from pt A to pt B

hzeller, 2018-01-20:
In the initial example you said that you want to split the screen horizontally, so it should always be simple to draw something on the top and something on the bottom.

If you want to split it verticallyi, yhou might only want to show a limited characters at a time. So if you have string ABCDE, then you can show ABC, scroll a little, start to show ABCD ... etc.

In general, if you want an arbitrary windowing thing, you need to manage each pixel individually, but I don't suggest to do that in Python as Python is pretty slow with that; it will be easier if you do that in C++ and e.g. write a delegating canvas that clips out areas.

anandrajgupta, 2021-08-17:
Hi all,
I am also trying to achieve a similar thing with partial success,
I want to achieve a vertically splitted screen.
My Application is: Draw a Static image in 1st half and scroll a car in 2nd half.
When i try to go with the idea suggested by Mr. @hzeller, it partially works (by scrolling one step then black background for the static image then redraw image) but it's not clear on screen i partially see a mixed pixels in static image (probably because of the small duration of scrolling image present just before writing black background)

Please enlighten me if there is any way to split the screen in such a way that the Right Half of the Screen is completely independent of the Left Half i mean no interference between those splitted areas.

Note: I am using C language to accomplish this, will that be slower as compared to C++?

Thank you,
Raj