hzeller/rpi-rgb-led-matrix #478: Python3 with Adafruit Hat and 32x64 matrix - flickering and pulsing panel

AdmPicard, 2018-01-16:
Hi!

After I was using your library just fine with my Raspberry PI 3 (Adafruit Hat with PWM mod) and my python3.5 script, I am now facing strange issues on the same device, on which I now installed Raspbian Lite for a fresh start.

I installed all dependencies (at least I think so) and followed your instructions to compile the library for Python 3. Whereas your demo examples work perfectly fine and with nearly no flicker, my python script does not properly work any longer: the panel will flicker rhythmically (and I can hear a subtle beeping sound in the same rhythm as well) and shows artefacts. Here is a [short video](https://drive.google.com/open?id=1iKEw4VM1K0bW2COhLphyLIuIWhG7SC5T).

I have set up everything again and again, but do not have a clue anymore, where the issue might be. My script/program is an exact copy of the version, which runs perfectly fine on my prior full Raspbian installation (just with python3.4 instead of 3.5) and which continues to do so if I recover this Raspbian installation. But of course, I'd like to stick with the new, fresh installation. Except from installing some python packages for my script, the Raspbian Lite installation is brand new.

I did not change any option except the `adafruit-hat-pwm` wiring and the chain length of two. If I change the `gpio_slowdown` to `2` though, I can prevent the flickering and the artefacts. However, the display will continue to rhythmically pulse. The custom slowdown was not necessary before - the display ran fine without additional modifications.

Thanks for any ideas! I hope, that I just made a stupid mistake somewhere.

hzeller, 2018-01-16:
Some people reported, that for whatever reason the 1-wire communication seems to be switched on in some Pis, that need to be removed, maybe it is something like that. Also, do all the usual things, such as the  isolcpus settings.

How long is your panel ? Depending on the length, you might need to [switch off RT-throttle](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L121); but that only manifests in regular (every couple of seconds or so) darker display and only really if you have more than like 8 panels connected so this might not be the reason for your observations.

Note, that Python as scripting language is much more taxing to the CPU and memory subsystem as C/C++ code; in particular if you have the garbage collector kick in, you might get more flicker. Note that Python2 is faster than 3 last time I checked, so if you use that, you might get more out of it.

AdmPicard, 2018-01-16:
Thanks for the quick advice! The 1-wire communication was already disabled and the `isolcpus` setting also did not show an effect. Whereas the flickering and the artefacts were gone after the `gpio_slowdown` increase, the strange and rhythmic pulsing remained. As I am just using a single 32x64 pixel panel, I don't expect a benefit from the throttle setting. However, I will try it as soon as I received a second SD card to speed up switching between the old/ new system and will post an update.

As I want to have a working display for now, I reverted to my initial setup with the full Raspbian installation. It does not require a modified `gpio_slowdown` and delivers a great and stable image even with my unchanged python3 script.

Thanks a lot for this library and the comprehensive instructions - I've had and have fun using it for my status display and I may expand with additional panels in the future.

hzeller, 2018-01-16:
Interesting. It it disconcerning that they apparently messed up the Raspbian Lite distribution; will have to look some time to figure out what is wrong.

hzeller, 2018-01-21:
Did you try playing with the `--led-gpio-slowdown` option ? Some panels can't keep up with the Pi.

claytonketner, 2018-01-21:
Hey @AdmPicard - saw this issue because I was having the same rhythmic brightness changes, but not the ghosting in your video, with nearly the same setup. I'm running raspbian stretch lite on a Pi 3 with python 2.7 and an unmodified adafruit HAT. 

I realized I had a `time.sleep(0.2)` in my code - removing that fixed the brightness pulsing. I still have some random line flashing, but I think that's a matter of reducing CPU usage for me to sort out (perhaps using `threading` or `multiprocessing`). I also already downloaded the full raspbian stretch, so I'll try that to see if I can confirm that the "lite" version is faulty.

Unrelated, thank you @hzeller for this repo and your continued efforts maintaining it!!! I really wish adafruit would just link to this repo rather than their fork. I stupidly used their fork for a while before noticing it is almost 500 commits behind!

hzeller, 2018-01-21:
The line flashing you can fix with the [PWM hardware hack](https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker) @claytonketner

AdmPicard, 2018-01-21:
@hzeller Using `--led-gpio-slowdown 2` nearly entirely eliminates the flickering and the artefacts - just the pulsing remains.

However, using my original Raspbian environment doesn't even require this setting for a perfectly stable image. Just the fresh Raspbian installations (no matter whether full or lite) require it and still the pulsing remains. As I am using the exact same hardware, there have to be software or configuration issues, which I am missing. I am running the display absolutely fine now since several days and even additional tasks like other scripts won't noticeably harm the display performance - it may yield a short flicker here and there but it's not worth mentioning.

@claytonketner I applied the PWM mod to my Adafruit hat, which may help you, too. I also have lots of short `sleep`s in my script for animation purposes, but these don't matter while running on my "old" Raspbian installation. I am just unable to get it working on a fresh Raspbian installation.

hzeller, 2018-01-21:
Could there be one of the GPIO-daemon things or PiGPIO running on the Pi that interferes with it ?

hzeller, 2018-01-21:
So after some experimenting it looks like the newer Raspbians are indeed worse than older (also see #483 ).

I think it is time to start a no frills Linux distribution that is suited for near realtime tasks...

AdmPicard, 2018-01-21:
Okay, interesting. So the current Raspbian distributions themselves are responsible for the decreased performance of applications like driving the display. Thanks for pointing this out as I'd probably have started further attempts to get it running with a fresh installation otherwise.

Unfortunately, I am not more than a hobby programmer and therefore always depend on the great work of other people. So I will stick around and have a look at the future developments. I am glad, that my older Raspbian continues to work very fine with your library and the python bindings.

Thanks a lot for your research and the effort!

hzeller, 2018-01-22:
I have expanded a bit the allowance for kernel induced jitter. Can you try with a fresh git pull and recompile everything to see if occasional brightness flicker is reduced ?

AdmPicard, 2018-01-24:
I have been able to test your new version now - sorry for the delay. Unfortunately, I had to set up a new Raspbian Lite environment as my backed up version refused to recover.

So I am not able to compare while using the exact same environment, but with the new fresh installation and your current library, the flicker, as well as the pulsing, are completely gone even with the default `--led-gpio-slowdown` setting. The picture is perfectly stable while I am still using my unmodified script python3.5 script.

Compared to my old working setup, I'd even say the quality got slightly better as the rare brightness changes are nearly entirely gone. So your modification seems to work very well in my environment.

However, I am seeing small scaling differences, which I cannot understand yet. A bitmap, for instance, is lacking some of its original height and some font characters seem to be compressed. I will try to attach an image for comparison.

Anyways, I will be able to run my setup with a fresh installation now - thanks a lot!

hzeller, 2018-01-24:
To test if there are issues with scaling etc., I'd suggest to run the demos. For instance `-D3` shows a rectangle that goes all to the edges.

closed 2018-02-17 by hzeller

hzeller, 2018-03-17:
Check out the new compile-time option in `FIXED_FRAME_MICROSECONDS` in [lib/Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L105). It will allow to time-buffer glitches created by the rest of the system.