hzeller/rpi-rgb-led-matrix #482: Using LINSN LED

GOBish, 2018-01-20:
Hi,

The code works great on most LED matrices I have, but I recently got this one so I could do an outdoor installation:

http://www.ledcontrolcard.com/outdoor-led-module-c-49/latest-outdoor-p5mm-18scan-64x32dots-320mmx160mm-smd-led-module-p-578.html

and the formatting is all weird - jumbled letters, line breaks, etc.

Any ideas on how to setup the code for this screen?

Thanks!

GOBish, 2018-01-20:
The board is a 32x64 made up of two 32x32 panels ribbon-ed together. I think it is 1:8 scan. They seem to be set up differently, though, in that each 32x32 has some kind of routing in it.
I tried the -L option (for U shape) for both the whole 32x64 board and for just one of the 32x32 panels, but that didn't help.

angelogoncalve, 2018-01-20:
Try use one panel 32x32 1:8 scan rate. The transformer for your case is:

void P5outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
int odd8_block =(((y % 16) / 8) +1) % 2; //to know if the pixel is in an odd 8pixel row
int num_mod_x = (x / 32); //num of daisy chainned modules
int new_x = (x + 32*(num_mod_x + odd8_block));
int new_y = (y % 8) + 8 *( y / 16);
delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

Use for one P5outdoorTransformer with:
examples-api-use/demo -D0 --led-rows=16 --led-chain=2 --led-parallel=1

GOBish, 2018-01-20:
Thanks very much for the advice. I'll try it out.

GOBish, 2018-01-21:
This issue has been solved in #311:
https://github.com/hzeller/rpi-rgb-led-matrix/issues/311#issuecomment-359214629

closed 2018-01-21 by hzeller

Saurav-K-Aryal, 2018-12-22:
@GOBish 
I have a Linsn LED panel as well. It's a quirky P2.5 screen with 1/20 scan. If you have any advice please drop some comments on https://github.com/hzeller/rpi-rgb-led-matrix/issues/738

Thank you!