hzeller/rpi-rgb-led-matrix #484: Compatible 64*64 Matrix for Adafruut HAT

Einhornkot, 2018-01-21:
Hello,

I plan to build my own LED-Matrix with an  Adafruit Hat.

Now I wanted to ask if somebody knows a compatible P4 64*64 or 64*32 Matrix.

Regards 
Einhornkot

hzeller, 2018-01-21:
If they are 64x64, then you might need to do [some hardware modding on the HAT]( https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hat), depending on how they address the lines.

There are some other 64x64 panels out there, that only use the  [A, B] address lines instead of [A, B, C, D, E]. They should work without change on the Adafruit HAT; you'd need to set the `--led-row-addr-type=1` command line flag to get these working (Cave, this has only been tested on a single type of panels with [A, B] address line, there might others out there that require some tweaks to the code).

So, now how to identify these panels on Aliexpress and such ? That might be hard, the descriptions are often pretty vague and typically don't explain what address mode there is. If  they have a picture of the connector, you can sometimes sortof see if they have [A, B] or [A, B, C, D, E].
Though _typically_ if you get any 64x64 RGB panel, it should be possible to get working with this library.

(Note, the adapter whose [plans are published in this project](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) already allows to do 64x64 panels with [A, B, C, D, E] addressing.)

hzeller, 2018-01-21:
If you get a 64x32 (instead of a 64x64) they are typically just arranged internally as 32 pixel high panel, so they will have 1:16 multiplexing (or, if they are 'outdoor panels', 1:8), so they will work without modification to the HAT as well.

It is more desirable to actually rather get two of these smaller 64x32 panels instead of one 64x64, because you can drive them with higher refresh rate. You can e.g. use them in a [U-configuration]( https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates) to get one 64x64 display.

Einhornkot, 2018-01-21:
@hzeller
Thank you for your help.

I think I will do use two 64x32.

Does anyone know some who work well with the Adafruit Hat?