hzeller/rpi-rgb-led-matrix #485: P5 64x32 16S

HummingBuddha, 2018-01-22:
Hello HZeller

Thanks for solving P5 32x32 8S (#66). I am having the same issue with P5 64x32 16S in parallel mode. I have even updated my rgb-matrix and Flaschen-Taschen but still I am having the issue. I am giving
server fields as sudo ./ft-server  --led-rows=32 --led-cols=64 -D64x64 --led-parallel=2 
and 
client fields as ./send-text -h localhost "  " -f fonts/10x20.bdf -s 0 -g 64x64+0+0+1
I have attached the images for your reference. Please let me know your inputs or should i need to write any transformer or modify the existing transformer.
[image]

[image]

hzeller, 2018-01-22:
It is hard to tell, in particular as this only uses the top-half of the display.
Can you run the [demo from this project with -D3 (simple square)](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#running-some-demos), which makes things a little easier to see ?

Do you have a picture of the IDC connector on the back of the panel to see the markings ? It might help to figure out what flavor that is.

Did you wire these panels yourself or are you using an [adapter]( https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) ? If you wired it yourself, double-check the address lines [A, B, C, D].

HummingBuddha, 2018-01-23:
Hello HZeller

Tried with D3 demo and have attached the images. 
Image:1 Single Panel, 
[image]

Image2: With parallel & without Chain
[image]

Image3: With parallel=2 & chain=2
[image]

Image4:The IDC 
[image]

Image5:A,B,C,D lines
[image]

Image6:P5-64x32-16S
[image]

Wiring is done by me, I havent used any adapters as such. Have checked the A,B,C,D lines and all are looking great. 
Hope these might help you to help me. 

Thanks for your support.

hzeller, 2018-01-23:
So looks like the address decoding in the bottom half of the bottom board is not working.

It is not your wiring, because then this would happen in the top and bottom half of that board, as the A, B, C, D are shared between the top and bottom half of each board. The fact that it only shows up on one of these means that it is not the wiring to the matrix but that the matrix itself is faulty. Nothing we can work around in software.

You can double-check if you make the bottom board the top board, then the same artifact will show up in the top board (that tells you that it is not the second parallel line, but just the top issue).

 I suspect they have a solder short there somewhere in some address decoder, you might maybe find something if you closely inspect all the chips they have on there. It could also be a faulty address decoder chip.

hzeller, 2018-01-23:
Another thought: do you level convert the 3.3V logic level to 5V for the matrix board ? Some boards really want 5V logic levels, so if no conversion took place, it is possible to see such artifacts.

You might be able to see if that is the case by reducing the voltage on the board to, say 3.5V - 4V and see if it starts working (the colors will be off, as it can't switch on all the LEDs, in particular blue, but this is only for experimenting).

closed 2018-02-17 by hzeller