hzeller/rpi-rgb-led-matrix #486: advice with buying parts for active adapter?

ckirmse, 2018-01-25:
Hi--I'd like to assemble the active board to get good quality output with some LED panels. I have done lots of through-hole soldering but no surface mount, but I trust your advice that it's doable. I've ordered the boards from OSH Park, but could use help on how to buy the parts. Looking on Mouser/Arrow/Digikey/etc., there are a ton of parts that match 100nF capacitor in 0805 package, or the driver chip, etc. I'm guessing a more experienced hardware person needs no more detail, but could you document more specifically every detail about the parts to put on the board? 

Thanks for any help you can give!

hzeller, 2018-01-25:
Literally any 0805 100nF capacitor will do, so just get the cheapest one (it should be voltage rated above 6.3V, but typically they are around 50).

Same for the driver chip. You essentailly only need to make sure to get a 74HCT245 or 74AHCT245 and get the cheapest in that SOIC-20 / 7.5mm size (e.g. Digikey part 1727-6542-1-ND)

I usually specify in Digikey the thing  I am looking for, then choose some smallish number and order by reverse prices (and possibly slide down the list until every criteria is fulfilled).h

ckirmse, 2018-01-25:
Thanks so much, I'm off to order now :)

closed 2018-01-25 by ckirmse