hzeller/rpi-rgb-led-matrix #489: Python 3: ImportError: No module named core - Python 2: Python.h: No such file or directory

bommels, 2018-01-26:
I followed the steps in the Python readme, all execute fine except when I want to run a sample. See my output below.

**make build-python PYTHON=$(which python3):**

```
(venv) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python $ make build-python PYTHON=$(which python3)
make -C ../../lib
make[1]: Entering directory '/home/pi/Desktop/display/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/home/pi/Desktop/display/rpi-rgb-led-matrix/lib'
/home/pi/virtualenvs/venv/bin/python3 setup.py build --build-lib .
running build
running build_py
running build_ext
```

**sudo make install-python PYTHON=$(which python3):**

```
(venv) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python $ sudo make install-python PYTHON=$(which python3)
/home/pi/virtualenvs/venv/bin/python3 setup.py install
running install
running build
running build_py
running build_ext
running install_lib
running install_egg_info
Removing /home/pi/virtualenvs/venv/lib/python3.6/site-packages/rgbmatrix-0.0.1-py3.6.egg-info
Writing /home/pi/virtualenvs/venv/lib/python3.6/site-packages/rgbmatrix-0.0.1-py3.6.egg-info
```

**The rgbmatrix package is sucessfully installed:**

```
(venv) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python $ pip freeze
certifi==2018.1.18
future==0.16.0
Pillow==5.0.0
rgbmatrix==0.0.1
RPi.GPIO==0.6.3
```

**When I want to run a sample script:**

```
(venv) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python/samples $ python --version
Python 3.6.0
(venv) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python $ cd samples/
(venv) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python/samples $ sudo ./runtext.py
Traceback (most recent call last):
  File "./runtext.py", line 3, in <module>
    from samplebase import SampleBase
  File "/home/pi/Desktop/display/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/home/pi/Desktop/display/rpi-rgb-led-matrix/bindings/python/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: No module named core
```

I am currently stuck on how to continue! I also tried **Python 2.7**(.3) but get an error on the make build-python command. See the output of that here:

```
(py2) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python $ python --version
Python 2.7.3
(py2) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python $ make build-python
make -C ../../lib
make[1]: Entering directory '/home/pi/Desktop/display/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/home/pi/Desktop/display/rpi-rgb-led-matrix/lib'
python setup.py build --build-lib .
running build
running build_py
running build_ext
building 'core' extension
gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I../../include -I/usr/include/python2.7 -c rgbmatrix/core.cpp -o build/temp.linux-armv6l-2.7/rgbmatrix/core.o -O3 -Wall
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for Ada/C/ObjC but not for C++ [enabled by default]
rgbmatrix/core.cpp:4:20: fatal error: Python.h: No such file or directory
compilation terminated.
error: command 'gcc' failed with exit status 1
Makefile:67: recipe for target 'build-python' failed
make: *** [build-python] Error 1
```

bommels, 2018-01-26:
So tried out many different things and got it working. I have no idea why, but running it with python3 instead of python works. Although 'python' and 'python3' both use the same configuration:

```
(venv) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python $ python --version
Python 3.6.0
(venv) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python $ python3 --version
Python 3.6.0
(venv) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python $ echo -e "import sys\nprint(sys.path)" | python
['', '/home/pi/virtualenvs/venv/lib/python36.zip', '/home/pi/virtualenvs/venv/lib/python3.6', '/home/pi/virtualenvs/venv/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6', '/home/pi/virtualenvs/venv/lib/python3.6/site-packages']
(venv) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python $ echo -e "import sys\nprint(sys.path)" | python3
['', '/home/pi/virtualenvs/venv/lib/python36.zip', '/home/pi/virtualenvs/venv/lib/python3.6', '/home/pi/virtualenvs/venv/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6', '/home/pi/virtualenvs/venv/lib/python3.6/site-packages']
```

**Running the sample with just 'python':**

```
(venv) pi@raspberrypi ~/Desktop/display/rpi-rgb-led-matrix/bindings/python $ sudo python samples/runtext.py
Traceback (most recent call last):
  File "samples/runtext.py", line 3, in <module>
    from samplebase import SampleBase
  File "/home/pi/Desktop/display/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/home/pi/Desktop/display/rpi-rgb-led-matrix/bindings/python/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: No module named core
```

**Running the sample with 'python3':**
```
sudo python3 samples/runtext.py
```
works!

Thank you for your fast reply.

closed 2018-01-26 by bommels

jroelofs, 2019-06-16:
> rgbmatrix/core.cpp:4:20: fatal error: Python.h: No such file or directory

This means you're missing `python2.7-dev`.