hzeller/rpi-rgb-led-matrix #490: Issue with 64 x 32 RGB LED Panel - Bottom half not lighting up

adamwth, 2018-01-27:
Hi, I tried to display some of the demos and kept getting only one half of the 64 x 32 LED panel lit up. The problem is not with faulty LED lights on the other half because I was able to turn them on when I set --led-inverse. 

I tried to run the simple square demo with this command:  
sudo ./demo -D 3 --led-cols=64
This was the result on a single 64 x 32 panel:
[image]

May I know what is the most likely problem for this? And how can I fix this?

Thanks in advance!

hzeller, 2018-01-27:
Good question, this looks like they have some different kind of multiplexing going on inside.
Can you post a picture of the connector on the back ? Hopefully it has some silk-screen around that helps understand the inputs.
If you had a compiled version lying around and recently updated the code to get the `--led-cols` option, make sure to make clean everything before (there could be some mismatching struct sizes in the compiled binary).

How does the rotating square (`-D0`) look like ?

adamwth, 2018-01-27:
(Back of my LED panel)
[image]

(What happens when I ran rotating square -D0)
[image]

Oh and I just cloned the repository today.

Thank you for replying!

hzeller, 2018-01-27:
mmh, this looks strange. And apparently, there are no markings on the input port.
Could it be that this is a 1:32 multiplexing ? Do you have any documentation from the vendor ?
Can you try to also connect the 'E' line to either pin 4 or pin 8 (you have to try) to the panel and choose `--led-cols=64 --led-rows=64` and see if that makes a difference ? If so we might be able to get it to work.

adamwth, 2018-01-27:
Ahh maybe it's because I got it from eBay.. The vendor didn't provide any documentation unfortunately.
What implications does 1:32 multiplexing have?

I'm not sure where the E pin is, is it the GND that is to the right of B2? If so, I tried connecting it to both pin 4 and 8, but the results didn't make a significant difference - one half is still not lighting up.

hzeller, 2018-01-27:
Are you using logic level voltage converters to 5V logic level ? These are needed for most panels as they are 5V logic and the Pi only outputs 3.3V. You can temporary try to reduce the voltage on the panel to maybe 4V or so and see if that makes a difference.

What is strange is, that in the example above with `-D3`, it looks like some parts are displayed side-by-side, while with the `-D0` demo top/bottom is overlayd on top of each other but otherwise perfectly centered. This is puzzling. Both of these are with exactly the same options, right ?

(1:32 would mean that it is slower, but apparantly we don't have this here.)

adamwth, 2018-01-27:
I am not using 5V logic converters as I don't currently have them on hand, but I will try sourcing them when I have the time! How do I reduce the voltage to 4V?

Yes, I ran both D3 D0 with the same option (--led-cols=64). I find it strange too, it seems like the logical levels for the LEDs in rows 17 - 32 are overflowing to those in rows 1-16.

hzeller, 2018-01-27:
the the `-D0` looks like they are overflowing, while the `-D3` examples looks like they are streting every other pixel in the length.

If you have the LED panel on a bench power supply, you can reduce the voltage there (while you still operate your Raspberry Pi with an independent e.g. USB supply).

adamwth, 2018-01-27:
Ah I don't have a bench power supply, but I will try to get the logic converters. I'll update here when I try them out.

closed 2018-02-17 by hzeller

Ashish-Jog, 2024-09-02:
@adamwth Did you find any work around, I have an exact panel from Waveshare and bottom 16 rows (or exactly lower half) is not working.