hzeller/rpi-rgb-led-matrix #492: Issue with 64 x 32 RGB LED Panel -- Interlaced display and top half always on

594crab, 2018-01-28:
sudo ./demo -D 0
[image]
sudo ./demo -D 0 --led-cols=64
[image]
sudo ./demo -D 3
[image]
sudo ./demo -D 0 --led-cols=64
[image]

The pcb back:
[image]
The hub75 connector:
[image]

I tried all the options of demo,the issue is not resolved.
I reduce the power supply from 5V to 4V,it is the same.
I was beated.

594crab, 2018-01-28:
row 1,3,5,7 and so on are on.And row 2,4,6,8 and so on are never on.

hzeller, 2018-01-28:
double-check your address lines A, B, C, D. At least A is not properly connected.
If you continue to have issues make sure to first get logic level converters.

594crab, 2018-01-29:
u are right. A connect to 17 and G1 connect to 15.
thanks a lot.

closed 2018-01-29 by hzeller