hzeller/rpi-rgb-led-matrix #493: USB Audio Out Issue - I Can't play sound

Phasor1, 2018-01-29:
Hello
I'm using the text-example.cc demo with some audio playback on Raspberry Pi 3.  I'm using a Behringher UCA-202 as external USB audio Codec to which are connected headphones. I've disabled the built in audio codec changing the /boot/config.txt file (dtparam=audio=off) as you write on the READMe.  I'm also using the "sox" tool for audio playback. Every time a use "play" command, provided by sox, to play some sounds while the text-example.cc is running, i can't ear anything from the USB sound card. Furthermore, after a "play" call, the console gives me these errors:

``` sh
No protocol specified
xcb_connection_has_error() returned true
Home directory not accessible: Permission denied
W: [pulseaudio] core-util.c: Failed to open configuration file '/root/.config/pulse//daemon.conf': Permission denied
W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
play FAIL formats: can't open output file `default': snd_pcm_open error: No such file or directory
```
I tried to modify all ALSA setting related with the external USB audio codec, but everything seems ok in that part, audio playback works perfectly outside the text-example.cc code. I really don't know what problem could be. 
Sorry for my english, i'm not a mother tongue.

hzeller, 2018-02-17:
Please ask the maintainers for the Alsa lib or the Raspbiean distribution, this is not something really that can be solved on the code level here. You can always use `--led-no-hardware-pulse` on the rpi-rgb-led-matrix library and leave the sound enabled, but that will reduce image quality.

hzeller, 2018-02-17:
Please update this bug once you've got feedback from the maintainers there, so that we can point people to that if they run into similar issues.

closed 2018-02-17 by hzeller