hzeller/rpi-rgb-led-matrix #494: Thanks Again

BNNorman, 2018-01-29:
I'd just like to says thanks again for your library and the  active driver board, which I bought from OSH Park.

You said you like to see what people are doing with your code etc so take a look at this (about 7 minutes long):-

https://www.youtube.com/watch?v=hF6wfx8zTg0&feature=youtu.be

It's a bunch of graphics and image animations running on a hub75 panel 64x64 pixels and driven from a Pi3. I have buffered output to the Panel , using numpy images, in order to handle an alpha channel. And no, I haven't driven a bigger panel (can't afford it).

The code is available here if anyone is interested:-

https://github.com/BNNorman/rpi-rgb-led-matrix-animator

Regards

Brian

angelogoncalve, 2018-01-29:
Dear Brian. I use your code but the simulator does not work because the opencv library. Can you help me please? Thank you for your atention.

BNNorman, 2018-01-29:
I need more detail. Can you write a small program which uses openCV to read
an image with cv2.imread and display it with cv2.imshow followed by
cv2.waitKey(0). If that works the simulator should work.

angelogoncalve, 2018-01-29:
Dear Brian. Yes openCV reads and show a picture:

import cv2

# Load an color image in grayscale
img = cv2.imread('messi.jpg', 0)

cv2.imshow('image', img)
cv2.waitKey(0)

But this not works why because the opencv version? It is another way to fix this problem? How can I start a simulator panel example in python? Thank you for your help. 

# opencv line types

LINE_AA=cv2.LINE_AA
LINE_8=cv2.LINE_8
LINE_4=cv2.LINE_4
FILLED=cv2.FILLED

BNNorman, 2018-01-29:
Try editing Constants.py and change to.

LINE_AA=16
LINE_8=8
LINE_4=4

Yes it sounds like an opencv version issue. I found it to be a pain.

angelogoncalve, 2018-01-29:
Dear Brian but gives an error in this line: FILLED=cv2.FILLED
How can I fix that? Any sugestion or value please? Thank you for your atention.

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Simulator/MatrixTest.py
Traceback (most recent call last):
  File "C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Simulator/MatrixTest.py", line 11, in <module>
    from Simulator.RGBMatrix import *
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\Simulator\RGBMatrix.py", line 15, in <module>
    from LEDAnimator.NumpyImage import *
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\NumpyImage.py", line 23, in <module>
    import ImageCache
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\ImageCache.py", line 21, in <module>
    from LEDAnimator.Constants import *
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Constants.py", line 73, in <module>
    FILLED=cv2.FILLED
AttributeError: 'module' object has no attribute 'FILLED'

Process finished with exit code 1

angelogoncalve, 2018-01-29:
Dear Brian. I do that also and change filled constant to -1 and changed cv2.FILLED to FILLED but give an error in the line above. How can I resolve this error in cv2.IMREAD_UNCHANGED or what value constant can I give -1? Thank you for your help.

# must include any Alpha channel so read unchanged
# openCV reads in BGR order. See Constants.py
srcBGR=cv2.imread(imagePath, cv2.IMREAD_UNCHANGED)

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Simulator/MatrixTest.py
Traceback (most recent call last):
  File "C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Simulator/MatrixTest.py", line 18, in <module>
    img=NumpyImage(imagePath="../Images/SmallRedBox.png")
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\NumpyImage.py", line 114, in __init__
    self.rgba_orig = ImageCache.loadImage(self.imagePath)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\ImageCache.py", line 57, in loadImage
    srcBGR=cv2.imread(imagePath,cv2.IMREAD_UNCHANGED)
AttributeError: 'module' object has no attribute 'IMREAD_UNCHANGED'

Process finished with exit code 1

BNNorman, 2018-01-29:
Change to  -1

BNNorman, 2018-01-30:
I have added the openCV constant values (as comments) in Constants.py and changed ImageCache.py to use my definitions. You'll experience similar problems trying to use HERSHEY fonts unless you download the new Constants.py plus a couple of other other updates to avoid your problem.

According to my research on the internet, the problem you are seeing with cv2.XXX constants not being recognised is (possibly) due to python picking up the wrong openCv library (previous V2 leftovers?).  If you built openCV on your PC you will find the library python needs in the build/python folder. I had to copy that to my python site-packages folder (save the old one if it exists). If you installed openCV using PIP then I don't know what version you got but cv2.__version__ will tell you.

angelogoncalve, 2018-01-30:
Dear Brian. My opencv version is 3.4.0.12 >= 3.0. I set the constants python file with the constants numbers but gives me an error when compile MatrixTest.py file with te compiler python 2.7.13. How can I fix this error please to run the simulator? Thank you for your help.

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Simulator/MatrixTest.py
RGBMatrix: Not recording video
RGBMatrix.run() starting the simulator window.
RGBMatrix.run() setting up window failed (keyboard key stuck?).
Image display took 0.048910

BNNorman, 2018-01-30:
It isn’t really an error – the MatrixTest.py just opens a simulator window – it does nothing with it except display the red box image and wait for you to press a key.

 

It thinks a key is already pressed. Possibly it need more time on your machine.

 

If you look at lines 173 to 176 in RGBMatrix.py you will see the error message. It is saying waitKey(1) returned a value other than 255 (-1) after waiting 1ms. This is just to allow the openCV imshow() code to display the image.

 

255 means it timed out waiting for a key press. Any other value indicates it thinks a key is pressed before it enters the “while self.running” loop.

 

This explains waitKey :-

 

https://docs.opencv.org/2.4/modules/highgui/doc/user_interface.html?highlight=waitkey#waitkey

 

If you print the value of r it will tell you which key openCV thinks is pressed.

 

You can increase the delay from 1 to any number you like. Possibly your system hasn’t created the window in time and you need to increase the value on line 173 – did the window appear at all? If not then try increasing the delay slightly.

 

Regards

angelogoncalve, 2018-01-30:
Dear Brian. For the question:

Possibly your system hasn’t created the window in time and you need to increase the value on line 173 – did the window appear at all? - The windows appears for a second and disappears after. I will increase the delay to see what happens. Thank you for your atention.

angelogoncalve, 2018-01-30:
Dear Brian. The windows from the simulator appears for a second and disappears after. I increase the delay and happens the same. I can not see the simulator display the red box image . How can fix this? Thank you for your help.

BNNorman, 2018-01-30:
I’ve just uploaded a patched RGBMatrix.py – apparently some systems don’t return -1 (255) on timeout. I have added masking to waitKey() which might fix your problem – the patch works on my machine (it worked ok anyway).

 

Basically do this:-

 

r=0xFF & waitKey(1)       on line 173

 

and also in the while loop – I’ve closed PyCharm so I can’t say which line it was – you could just download the updated RGBMatrix.py.

 

Let me know how you get on.

 

Regards

angelogoncalve, 2018-01-31:
Dear Brian. It works the simulator thank you for your atention. I have only an example that not works. How can fix these errors? Thank you for your help.

PanelDemo.py - used to showcase panel animations

def antiAlias(self,method=None,coordList): gives an error in file AntiAlias.py

def startVideo(self,fname):
fourcc = cv2.VideoWriter_fourcc(*'DIVX') gives an error in file RGBMatrix.py

BNNorman, 2018-01-31:
You haven’t told me what the errors are. Grab a screen shot of all the messages, please.

 

Fourcc isn’t mine, I assume you want to record the video in a different format –I’m guessing (tired and going to bed) it should be :-

 

def startVideo(self,fname):
self.video = cv2.VideoWriter_fourcc(*'DIVX') 

 

if you can provide the full error messages I’ll reply in the morning.

BNNorman, 2018-02-01:
I made an error in my last email – I should have looked at my code again first.

 

This may be the problem –  (you didn’t send me the actual error message so I was guessing) – I’m guessing it’s the same problem you had with openCV on your machine

 

http://answers.opencv.org/question/29648/attributeerror-module-object-has-no-attribute-videowriter_fourcc/

 

Try this…

 

fourcc=cv2.cv.CV_FOURCC(*'DIVX')       

 

it’s doesn’t give an error on my system.

 

Instead of DIVX you might also specify XVID if it’s the DIVX encoder that’s the problem.

 

 

Let me know if this fixes your problem – if so I’ll add something to Constants.py.

angelogoncalve, 2018-02-01:
Dear Brian. I have an error when I run PanelDemo.py for the simulator. How can fix? Thank you for your support my friend.

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Examples/PanelDemo.py
Traceback (most recent call last):
  File "C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Examples/PanelDemo.py", line 33, in <module>
    from LEDAnimator.Chain import *
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Chain.py", line 37, in <module>
    import Helpers.AntiAlias as AA
  **File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Helpers\AntiAlias.py", line 42
    def antiAlias(self,method=None,coordList):
SyntaxError: non-default argument follows default argument**
Doing imports, please wait...
PathSetter.py: Parent folder [ C:\Users\User\Desktop\rpi-rgb-led-matrix-animator ] already exists in sys.path
Running on Windows simulator

Process finished with exit code 1

BNNorman, 2018-02-01:
I saw that error before. Thought i had fixed it. Check you have latest
copy. Meanwhile add =None before the end bracket.

angelogoncalve, 2018-02-01:
Dear Brian. I have already put that but the PanelDemo.py not show any video in the simulator, it shows a black screen in the simulator. How can fix? Thank you for your help my friend.

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Examples/PanelDemo.py
Doing imports, please wait...
PathSetter.py: Parent folder [ C:\Users\User\Desktop\rpi-rgb-led-matrix-animator ] already exists in sys.path
Running on Windows simulator
Panel.init() starting..
RGBMatrix: Recording video to  ./PanelDemo.avi
VideoWriter started ok
Panel.init() creating frameBuffer width 64,height 64

Creating the Animator and adding animations
Animations have been created
Running
RGBMatrix.run() starting the simulator window.
RGBMatrix.run() entering the run loop frameDuration=10.00ms
Animator.run() Animation frame interval exceeded - check animation durations. Total loopTime= 0.499041030408 frameInterval= 0.01
*** exception:
Traceback (most recent call last):
  File "C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Examples/PanelDemo.py", line 77, in <module>
    A.run()
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Animator.py", line 98, in run
    animInfo.nextFrame(self.debug)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\AnimInfo.py", line 51, in nextFrame
    if self.animFunc.nextFrame(debug=self.debug,id=self.animFunc.id):
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\AnimBase.py", line 295, in nextFrame
    self.step()
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\PanelAnimations.py", line 87, in step
    self.refreshCanvas()
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\AnimBase.py", line 441, in refreshCanvas
    pasteWithAlphaAt(self.layerBuffer.getImageData(),X, Y, self.fgImage.getImageData())
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\UtilLib.py", line 102, in pasteWithAlphaAt
    assert fg is not None, "fg (foreground) image cannot be None."
AssertionError: fg (foreground) image cannot be None.

BNNorman, 2018-02-01:
Ok, thats a bug.  Replace line 102 with

If fg is None: return bx

Never saw this on my machine - odd

angelogoncalve, 2018-02-01:
Dear Brian. This condition fix the bug and show a first image in the simulator, then gives an error. How can fix? Thank you for your atention Dear friend.

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Examples/PanelDemo.py
Doing imports, please wait...
PathSetter.py: Parent folder [ C:\Users\User\Desktop\rpi-rgb-led-matrix-animator ] already exists in sys.path
Running on Windows simulator
Panel.init() starting..
RGBMatrix: Recording video to  ./PanelDemo.avi
VideoWriter started ok
Panel.init() creating frameBuffer width 64,height 64

Creating the Animator and adding animations
Animations have been created
Running
RGBMatrix.run() starting the simulator window.
RGBMatrix.run() entering the run loop frameDuration=10.00ms
Animator.run() Animation frame interval exceeded - check animation durations. Total loopTime= 0.296768829676 frameInterval= 0.01
*** exception:
Traceback (most recent call last):
  File "C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Examples/PanelDemo.py", line 77, in <module>
    A.run()
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Animator.py", line 98, in run
    animInfo.nextFrame(self.debug)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\AnimInfo.py", line 51, in nextFrame
    if self.animFunc.nextFrame(debug=self.debug,id=self.animFunc.id):
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\AnimBase.py", line 295, in nextFrame
    self.step()
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\PanelAnimations.py", line 352, in step
    p=PoissonLib()
NameError: global name 'PoissonLib' is not defined

BNNorman, 2018-02-01:
Looks like the bugs you found were fixed but not committed to Github as I was in the middle of trying to improve the code speed.

 

I’ve just done that. 

 

You can download the version I have just committed.

 

Many thanks for finding the bugs – it’s been useful for me.

angelogoncalve, 2018-02-01:
Dear Brian. In your code, from the HZeller library which funtions did you use from rgbmatrix, and where are they in your files which python file have these methods below? Thank you for your help my friend.

SetPixel
Clear
Fill
SetImage
CreateFrameCanvas
SwapOnVSync
graphics

BNNorman, 2018-02-01:
Panel.py uses the zeller setImage() only. SetPixel writes to the numpy
image (frameBuffer) which is sent to the panel.

angelogoncalve, 2018-02-01:
Dear Brian. Thank you for your support. I change the two files PoissonLib.py and UtiLib.py with your changes but I have this error in the PanelDemo.py when execute after showing at first an image into the simulator. How can fix? Thank you for your kind help Dear friend.

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Examples/PanelDemo.py
Doing imports, please wait...
PathSetter.py: Parent folder [ C:\Users\User\Desktop\rpi-rgb-led-matrix-animator ] already exists in sys.path
Running on Windows simulator
Panel.init() starting..
RGBMatrix: Recording video to  ./PanelDemo.avi
VideoWriter started ok
RGBMatrix.run() starting the simulator window.
RGBMatrix.run() entering the run loop frameDuration=10.00msPanel.init() creating frameBuffer width 64,height 64

Creating the Animator and adding animations
Animations have been created
Running
Animator.run() Animation frame interval exceeded - check animation durations. Total loopTime= 0.0392869606529 frameInterval= 0.01
*** exception:
Traceback (most recent call last):
  File "C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Examples/PanelDemo.py", line 77, in <module>
    A.run()
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Animator.py", line 98, in run
    animInfo.nextFrame(self.debug)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\AnimInfo.py", line 51, in nextFrame
    if self.animFunc.nextFrame(debug=self.debug,id=self.animFunc.id):
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\AnimBase.py", line 295, in nextFrame
    self.step()
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\PanelAnimations.py", line 364, in step
    self.fgImage.setPixel(int(x),int(y),color)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Image.py", line 204, in setPixel
    self.image.setPixel(x,y,color)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\NumpyImage.py", line 489, in setPixel
    self.out[y,x]=alphaBlendPixels(color,self.out[y,x])
ValueError: cannot convert float NaN to integer

angelogoncalve, 2018-02-01:
Dear Brian. Did you use the function SetImage from the HZeller library to do the scroll text in the simulator/panel? Thank you for your atention my friend.

BNNorman, 2018-02-01:
I got the same error but when i ran it a second time it worked ok. Until i
can reproduce the error every time it will be tough to fix. I'll take a
look in the morning.

angelogoncalve, 2018-02-01:
Dear Brian. Ok my friend I wait for your help and thank you for your generous to gives your kindly support.

BNNorman, 2018-02-01:
Text is rendered to an image first. The image is then positioned over the
frameBuffer then sent to the panel.

BNNorman, 2018-02-01:
Do you get the NaN error every time? If so it might be helpful if i send
you code with extra debugging in it.

BNNorman, 2018-02-02:
Looking at where this error originated it seems that the PoissonLib *might* have generated Nan values. 

 

Since I didn’t write the original code I don’t know why it is possible. However, I’ve added code to filter the samples and remove any NaN values.

 

If the filtering results in an empty list you should see this message

 

"PoissonLib() failed to generate any samples."

 

I have run the code in a loop 100x and it hasn’t once failed.

 

I have pushed the new PoissonLib.py to Github – it will be easier for you to download that rather than me explain where to add the extra 8 lines of code.

 

Regards

 

Brian

angelogoncalve, 2018-02-03:
Dear Brian. The filter not solve the problem yet. In the simulator appears alternaty two type of images but after a time the simulator crash. How can fix this error? Thank you for your atention and help my friend.

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Examples/PanelDemo.py
Doing imports, please wait...
PathSetter.py: Parent folder [ C:\Users\User\Desktop\rpi-rgb-led-matrix-animator ] already exists in sys.path
Running on Windows simulator
Panel.init() starting..
RGBMatrix: Recording video to  ./PanelDemo.avi
VideoWriter started ok
RGBMatrix.run() starting the simulator window.
RGBMatrix.run() entering the run loop frameDuration=10.00ms
 Panel.init() creating frameBuffer width 64,height 64

Creating the Animator and adding animations
Animations have been created
Running
Animator.run() Animation frame interval exceeded - check animation durations. Total loopTime= 0.0389856330519 frameInterval= 0.01
*** exception:
Traceback (most recent call last):
  File "C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Examples/PanelDemo.py", line 77, in <module>
    A.run()
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Animator.py", line 98, in run
    animInfo.nextFrame(self.debug)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\AnimInfo.py", line 51, in nextFrame
    if self.animFunc.nextFrame(debug=self.debug,id=self.animFunc.id):
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\AnimBase.py", line 295, in nextFrame
    self.step()
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\PanelAnimations.py", line 364, in step
    self.fgImage.setPixel(int(x),int(y),color)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Image.py", line 204, in setPixel
    self.image.setPixel(x,y,color)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\NumpyImage.py", line 489, in setPixel
    self.out[y,x]=alphaBlendPixels(color,self.out[y,x])
ValueError: cannot convert float NaN to integer

BNNorman, 2018-02-03:
I am able to repeat this now. I still don’t understand why it happens. The values of x, y and color are correct and they are/contain integers

 

There is a simple workaround. I have put a try-except around the offending code to ignore the error – it doesn’t affect the Twinkle animation – but more importantly the animation continues to run.

 

I have pushed an updated NumpyImage.py to Github – I suggest you download it.

 

Regards

 

Brian

angelogoncalve, 2018-02-03:
Dear Brian. Thank you it works fine. I have in the university a paneol 16x32 with scan rate 1:8. I want to test in home with the simulator to set a pixel first using the code below in your simulator. How can I do it? Thank you for your generous help my friend.

from rgbmatrix import RGBMatrix, RGBMatrixOptions

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.hardware_mapping = 'adafruit-hat'
options.inverse_colors = False
options.led_rgb_sequence = "BGR"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0

display = RGBMatrix(options=options) 

while True:
    x=0
    y=0
    red = 0
    green = 255
    blue = 0
    display.SetPixel(x, y, red, green, blue)

BNNorman, 2018-02-03:
This code drives the panel directly through the hzeller library. It has
nothing to do with the simulator. It needs a loop to keep the panel
refreshed. You need to call SwapOnVSync. Look at zellers python examples
for ideas.

angelogoncalve, 2018-02-05:
Dear Brian. I see that you have a test for see if the HZeller library was been correctly instaled with the file test PanelCheck.py to test a square in the panel. Have you any way to do the test PanelCheck.py in the simulator that you created? Thank you for your atention my friend.

BNNorman, 2018-02-06:
SimulatorCheck.py added - also some other bug fixes.

 

SimulatorCheck.py should run on the Pi and live panel as well.

angelogoncalve, 2018-02-06:
Dear Brian. I get an error when I run the SimulatorCheck.py file. How can solve this error please? Thank you for your help. How can I send the same message for two servers panels in the same IP address using UDP multicast in python? Thank you for your attention my friend.

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Tests/SimulatorCheck.py
Running on Windows simulator
Panel.init() starting..
RGBMatrix: Recording video to  ./PanelDemo.avi
VideoWriter started ok
RGBMatrix.run() starting the simulator window.
RGBMatrix.run() entering the run loop frameDuration=20.00ms
 Panel.init() creating frameBuffer width 64,height 64

Traceback (most recent call last):
  File "C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Tests/SimulatorCheck.py", line 55, in <module>
    Panel.DrawPixel(x,y,b)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Panel.py", line 214, in DrawPixel
    frameBuffer.setPixel(x, y, color)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\NumpyImage.py", line 502, in setPixel
    assert type(x) is np.ndarray and type(y) is np.ndarray, "If color is a numpy ndarray then x and y must also be ndarrays."
AssertionError: If color is a numpy ndarray then x and y must also be ndarrays.

BNNorman, 2018-02-06:
Check you have the latest version of NumpyImage.py and Panel.py I think I
fixed that recently.

angelogoncalve, 2018-02-07:
Dear Brian. Yes I have the latest version of NumpyImage.py and Panel.py and I get the same error again when I run the SimulatorCheck.py file. How can solve this error please? Thank you for your help my friend.

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Tests/SimulatorCheck.py
Running on Windows simulator
Panel.init() starting..
RGBMatrix: Recording video to ./PanelDemo.avi
VideoWriter started ok
RGBMatrix.run() starting the simulator window.
RGBMatrix.run() entering the run loop frameDuration=20.00ms
Panel.init() creating frameBuffer width 64,height 64

Traceback (most recent call last):
File "C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Tests/SimulatorCheck.py", line 55, in
Panel.DrawPixel(x,y,b)
File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Panel.py", line 214, in DrawPixel
frameBuffer.setPixel(x, y, color)
File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\NumpyImage.py", line 502, in setPixel
assert type(x) is np.ndarray and type(y) is np.ndarray, "If color is a numpy ndarray then x and y must also be ndarrays."
AssertionError: If color is a numpy ndarray then x and y must also be ndarrays.

BNNorman, 2018-02-08:
I am unable to replicate this – though I can think of a workaround for it, but before I add extra code could you please insert the following code before line 489 ,at start of NumpyImage.setPixel() method then let me know what is printed please.

 

print "type(x)=", type(x), "type(y)=", type(y),"type(color)=",type(color)

 

I get the following showing that x and y are both integers and color is a tuple  e.g. (r,g,b,a)

 

type(x)= <type 'int'> type(y)= <type 'int'> type(color)= <type 'tuple'>

 

Regards

Brian

angelogoncalve, 2018-02-08:
Dear Brian. I get this error when I test the PanelDemo.py file. At start of NumpyImage.setPixel() method the result of the print was:

type(x)= <type 'int'> type(y)= <type 'int'> type(color)= <type 'numpy.ndarray'> 

How can fix please? Thank you so much for your help my friend.

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Tests/SimulatorCheck.py
Running on Windows simulator
Panel.init() starting..
RGBMatrix: Recording video to  ./PanelDemo.avi
VideoWriter started ok
RGBMatrix.run() starting the simulator window.
RGBMatrix.run() entering the run loop frameDuration=20.00ms
 Panel.init() creating frameBuffer width 64,height 64

UtilLib.alphaBlend() images not the same shape. Ignored. fg (64, 63, 4) bg (63, 63, 4)
type(x)= <type 'int'> type(y)= <type 'int'> type(color)= <type 'numpy.ndarray'>
Traceback (most recent call last):
  File "C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Tests/SimulatorCheck.py", line 55, in <module>
    Panel.DrawPixel(x,y,b)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Panel.py", line 214, in DrawPixel
    frameBuffer.setPixel(x, y, color)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\NumpyImage.py", line 502, in setPixel
    assert type(x) is np.ndarray and type(y) is np.ndarray, "If color is a numpy ndarray then x and y must also be ndarrays."
AssertionError: If color is a numpy ndarray then x and y must also be ndarrays.

BNNorman, 2018-02-08:
There are two errors in there. The main one causing the AssertionError I have just fixed getPixel() wasn’t returning a tuple.

 

This error will take more investigation.

 

UtilLib.alphaBlend() images not the same shape. Ignored. fg (64, 63, 4) bg (63, 63, 4)

 

I have recently uploaded a new version of UtilLib.py would be worth downloading it to see if it still happens.

angelogoncalve, 2018-02-08:
Dear Brian. Yes I have the recently UtiLib.py file and I test the PanelDemo.py and have the same errors. I wait for your help my friend when you can. Thank you for your kindly attention.

C:\Python27\python.exe C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Tests/SimulatorCheck.py
Running on Windows simulator
Panel.init() starting..
RGBMatrix: Recording video to  ./PanelDemo.avi
VideoWriter started ok
RGBMatrix.run() starting the simulator window.
RGBMatrix.run() entering the run loop frameDuration=20.00ms
 Panel.init() creating frameBuffer width 64,height 64

UtilLib.alphaBlend() images not the same shape. Ignored. fg (64, 63, 4) bg (63, 63, 4)
Traceback (most recent call last):
  File "C:/Users/User/Desktop/rpi-rgb-led-matrix-animator/Tests/SimulatorCheck.py", line 55, in <module>
    Panel.DrawPixel(x,y,b)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\Panel.py", line 214, in DrawPixel
    frameBuffer.setPixel(x, y, color)
  File "C:\Users\User\Desktop\rpi-rgb-led-matrix-animator\LEDAnimator\NumpyImage.py", line 504, in setPixel
    assert type(x) is np.ndarray and type(y) is np.ndarray, "If color is a numpy ndarray then x and y must also be ndarrays."
AssertionError: If color is a numpy ndarray then x and y must also be ndarrays.

angelogoncalve, 2018-02-08:
Dear Brian. Sorry for my mistake. The test that gives 2 errors was made in the file SimulatorCheck.py. Sorry again my friend.

BNNorman, 2018-02-08:
Yes I can see that but I uploaded a new NumpyImage.py tonight because that
was causing it.

angelogoncalve, 2018-02-09:
Dear Brian. Thank you my friend. The SimulatorCheck.py is working but have the warning UtilLib.alphaBlend() images not the same shape. Ignored. fg (64, 63, 4) bg (63, 63, 4) with the new file UtiLib.py. How can I use multicast UDP using your simulator? Thank you for your help Dear friend.

BNNorman, 2018-02-10:
Good stuff. 

I think the 'images not same shape' is still an issue in UtilLib.py which I need to work on. It's to do with the calculation of the slice of the foreground image which overlaps the background. Getting the background overlap region is the easy bit but the foreground slice can slide within the foreground image and I think it's that calculation which may be 1 pixel wrong.

UDP:

Check this out: https://stackoverflow.com/questions/22878625/receiving-broadcast-packets-in-python

In RGBMatrix you would add code to send the video frames (self.frameBuffer) at the end of the setImage() method (After line 155). I've not played with UDP so can't help more. Sounds interesting though, what do you plan to do?

angelogoncalve, 2018-02-10:
Dear Brian. Thank you for your attention my friend. I want to use UDP multicast to send the same pixels/data at the same time to a pair of panels instead of using the same entrance from an adafruit hat with a flat cable for both, instead using hardware stuff solve using by software. How can do by only using software? Thank you for your help Dear friend.

BNNorman, 2018-02-10:
Not really enough HDW information for me. You need an UDP listener on the second panel which sends it's received image to the screen. If you have a spare networked RPi you should be able to do that using Python and the information on StackOverflow . 
I must add that this goes beyond the purpose of my project BUT you could use Panel with the UDP Listener - you need to make sure the Pi doesn't have the HZeller RGBMatrix software installed or remove the test for it at the start of Panel.py
As I said, this is beyond the scope of my project and I have problems with my central heating to deal with right now.

hzeller, 2018-02-17:
To come back to the original message on this thread: Thanks @BNNorman for the kudos and you have a nice LED panel and video there!

closed 2018-02-17 by hzeller

angelogoncalve, 2018-02-17:
Change in the TextDemo.py 

cv2msg=Text(text=cv2Text,fontSize=12,fontFace=FONT_HERSHEY_SIMPLEX,fgColor=Palette.XMAS,bgColor=(125,0,125,255),
lineType=LINE_AA) 

to

cv2msg=Text(text=cv2Text,fontSize=12,fontFace=FONT_HERSHEY_SIMPLEX,fgColor=Palette.XMAS,bgColor=(125,0,125,255))

BNNorman, 2018-02-17:
You really should continue issues with my code on my Github project page not on HZellers site please.

https://github.com/BNNorman/rpi-rgb-led-matrix-animator