hzeller/rpi-rgb-led-matrix #495: LED display flickering

DebaGracias, 2018-01-30:
Hi All,
I am using Python 2.7 on my Raspberry Pi 3 .I was successfully able to display a count up timer on the left of the LED display and a scrolling text on the right.I am able to see flickers.I have switched off on-board sound (dtparam=audio=off in /boot/config.txt),although there is minor improvement  the led still flickers after every 2 secs.I am also using python threads to carry out some tasks.Is that causing the flickering?
Is there any setting or option that I can enable so that I can eliminate the flickering of LED? 

Thanks,
Debasish

hzeller, 2018-01-30:
How do you do the scrolling/counter thing ? Make sure to not accidentally blank the other half of the screen.

If there is a lot of processing going on that requires memory access, you can saturate the memory bus thus reducing the throughput the matrix update needs. So if you do something CPU and memory intensive, you will cause the matrix to flicker. So try to offload that (e.g. by pre-calculating things; the [`led-image-viewer`](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#image-viewer) precalculates animations in off-screen buffers beforehand so that it then can stream it later.
In general it is a good idea to move to C++ if you want to do something CPU intensive.

Finally, make sure to run a version of this library that is fresher than a week, because about a weekend ago I tweaked some defaults that might reduce flicker.

hzeller, 2018-01-30:
Also make sure to do the `isolcpus` setting.

DebaGracias, 2018-02-03:
Thank you for your reply.I tried using the latest version of the library and also setting isolcpus to 3.There is a minor improvement with this but the LED still flickers.
Here is the piece of code that i'm using to display the timer and the scrolling message:
 while True:
                self.offscreen_canvas.Clear()
                strTime = self.strTime           
                self.message = self.response
                if self.message != '':                    
                    length = graphics.DrawText(self.offscreen_canvas, font, self.pos, 15, textColor, self.message)
                    self.offscreen_canvas.SetImage(self.image, -35,0)
                    graphics.DrawText(self.offscreen_canvas, fontTimer,posTimer, 10, textColor, strTime)
                    
                    self.pos -= 1
                    if (self.pos + length < 0 or length == 0):
                        self.pos = self.offscreen_canvas.width
                    
                    mins,secs=divmod(int(self.t),60)
                    timeformat='{:02d}:{:02d}'.format(mins,secs)
                    self.strTime = timeformat
                    self.t+=0.02
                  
                    time.sleep(0.02)
                    self.offscreen_canvas= self.matrix.SwapOnVSync(self.offscreen_canvas)
   
                if(self.t >= 60):
                   self.SwitchOffLED()

Is something wrong with my code?
The LED model is http://arduino.ru/forum/apparatnye-voprosy/led-matrix-32x16-c5-4x8v10-dual-color-rg and I bought it from https://www.aliexpress.com/item/40-inch-101cm-ios-And-Android-Wifi-wireless-remote-Programmable-Advertising-LED-Display-Board-Bright-Red/32752577115.html?spm=a2g0s.9042311.0.0.F3UzUv.
How can I completely eliminate the flickering?

hzeller, 2018-02-17:
Make sure you don't have anything else running on the Pi (a shell running top, some cron-job doing stuff regularly etc.), this will mess up timings inside the Pi. If there are other things running on the Pi, slight brightness variations are to be expected.
Essentially follow [some recommendations in the documentation](https://github.com/hzeller/rpi-rgb-led-matrix#use-minimal-raspbian-distribution).

Also, the panel you're using looks more like a HUB12; did you verify the logical level of the output-enable ? Often they need to be reversed.

closed 2018-02-17 by hzeller

hzeller, 2018-03-17:
Check out the new compile-time option in `FIXED_FRAME_MICROSECONDS` in [lib/Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L105). It will allow to time-buffer glitches created by the rest of the system.