hzeller/rpi-rgb-led-matrix #496: LEDs flickering around Text

sKeLeTr0n, 2018-01-30:
Hey there,

I didn't really found any issues listed here about this kind of flickering so I am hoping that someone can lead me to the right direction. I am using this library with Python 3.6 and no matter what sample application I run or if I run my own, i get LED flickering on LEDs that shouldn't light up. It is only on the far left side of the Panel 32x64 panel though. Here is a short video that shows the problem:
https://drive.google.com/file/d/0BzWe6fDGEkCRY28zbUtsQTdxWDQtNU5jM0d0TWZ5eUJETVFj/view

Does anybody know what could be causing this/how to fix it? Thank you very much!

sKeLeTr0n, 2018-01-30:
Panel I am using : https://ru.aliexpress.com/item/P5-indoor-full-color-RGB-led-display-2-in-1-64-32-pixel-320mm-160mm-fant/32582246175.html

hzeller, 2018-01-30:
more gpio slowdown might help.

 Also, are you using the latest version of the library? Recent kernels and Raspian distros seem to increase artifacts, so I was tweaking things a little last week. Make sure to have a fresh compile of that.

Also make sure to apply the usual optimizations, such as the `isolcpus` setting.

sKeLeTr0n, 2018-01-30:
Okay, I will take a look, thanks. I am using the latest library build though. Gpio slowdown didn't really help when I tried it. I will take a look at isolcpus.

angelogoncalve, 2018-01-30:
Set pwm bits to 1.

sKeLeTr0n, 2018-01-31:
Setting isolcpus to 3 didn't help, nor slowing down gpio. Any other ideas?

hzeller, 2018-01-31:
How is your panel connected, did you use a digital level converter from the 3.3V to 5V ?

sKeLeTr0n, 2018-01-31:
It gets power separately from the raspberry pi, with a 5V 2100 mA power adapter

sKeLeTr0n, 2018-02-04:
Thanks for your help. I used shorter wiring between the led panel pins and the raspberry pi pins and it stopped the flickering.

closed 2018-02-04 by sKeLeTr0n