hzeller/rpi-rgb-led-matrix #498: Calling Led-image-viewer in Python Function

Sakthi-apat, 2018-01-31:
I have been trying to call led-image-viewer in my Python GUI code to display any image that is picked up from a folder.  I am able to call the led-image-viewer, but nothing is  displayed in matrix.  Can anyone suggest me a solution for this.

hzeller, 2018-01-31:
If you want someone to help you, you need to supply more information.
Is the image viewer running, but not displaying, or is it just exiting ?
Does it work if you manually call the program with the `led-image-viewer` ?
Are you providing the absolute path to the image ?
Is the image world-readable ? (the program switches to be running as user daemon, so that needs to be able to read the file).
What is your hardware set-up, i.e. how did you wire the Pi to the Matrix ?

In general, it is a good idea to not call some external executeable from Python, but just display the image  directly in Python or just write the program in C++. Calling some external executable means you need to have a way to stop it for instance which you only can do by sending a nasty kill to the process.

Sakthi-apat, 2018-02-02:
Thanks.  I understand.  I am working on it now.  Will get back to this.  For now I am closing this issue.

closed 2018-02-02 by Sakthi-apat