hzeller/rpi-rgb-led-matrix #500: Can't run python samples

iFEARR, 2018-01-31:
Hello, like the title says, I can't run the python scrips, all I get is 
`Traceback (most recent call last):
  File "samples/runtext.py", line 3, in <module>
    from samplebase import SampleBase
  File "/home/pi/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/usr/local/lib/python2.7/dist-packages/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: cannot import name RGBMatrixOptions
`

hzeller, 2018-01-31:
make sure to compile and install the Python wrapper first.

iFEARR, 2018-02-01:
I did by running

> sudo apt-get update && sudo apt-get install python2.7-dev python-pillow -y
make build-python
sudo make install-python

Do I need to do something else ?

hzeller, 2018-02-01:
yep, that should be it. If it doesn't work, there is something funky with your Python installation. Please ask a local Python person to help figuring it out.

dayne, 2018-02-01:
Just a note: It appears you have a `/usr/local/lib/python2.7` in your development path but you are trying to apt-install the development libraries.  The apt installed python and python development libraries will not (should not) show up in `/usr/local`.  As hzeller pointed out you should double check your python environment and verify which python you are building against.  Once you get a clean python setup the build should work fine again.

closed 2018-02-17 by hzeller

dkyy, 2019-07-30:
I'm having a similar problem when I try to import from my own directory. 

When I do:  
  ```
$make build-python PYTHON=$(which python3)
  make: *** No rule to make target 'build-python'.  Stop.
```

I tried adding this to the top of my python code:
`sys.path.insert(0, "/home/pi/rpi-rgb-led-matrix/bindings/python")`

Which allows me to do this successfully:
  `from rgbmatrix import RGBMatrix, RGBMatrixOptions`

But if I'm not running as root I get a segmentation fault when I do:
   `rgb_matrix = RGBMatrix(rgb_options = rgb_options)`

And, if I don't run as root, I get this error when I try to import RGBMatrix,etc...

```
File "/home/pi/rpi-rgb-led-matrix/bindings/python/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: No module named core
```

Any help would be appreciated.

Thanks,

david,

hzeller, 2019-07-30:
did you follow the instructions to install it ?

dkyy, 2019-07-30:
I thought I did, but perhaps I did something wrong. Could you point me to the instructions so I can make sure I started in the right place?

david,

hzeller, 2019-07-30:
build instructions are in bindings/python/README.md

dkyy, 2019-07-31:
Ok - my first dumb mistake was that I wasn't issuing the make commands root directory for the matrix library (ie. ~/rpi-rgb-led-matrix/bindings/python )

But I'm still having problems.   If I don't run as root here's what happens:

```
$ python3
>>> import sys
>>> from tkinter import *
>>> from rgbmatrix import RGBMatrix, RGBMatrixOptions
>>> from PIL import ImageTk,Image

>>> rgb_options = RGBMatrixOptions()
>>> rgb_options.rows = 64
>>> rgb_options.cols = 64
>>> rgb_options.chain_length = 1
>>> rgb_options.parallel = 1
>>> rgb_options.hardware_mapping = 'regular'  # If you have an Adafruit HAT: 'adafruit-hat'

>>> rgb_matrix = RGBMatrix(rgb_options = rgb_options)

Segmentation fault
```

If I run as root a different problem:

```
$ sudo python3

>>> import sys
>>> from tkinter import *
>>> from rgbmatrix import RGBMatrix, RGBMatrixOptions
>>> from PIL import ImageTk,Image

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'ImageTk'
```
And the same happens if I run as:   $ sudo -H -E python3

Sorry if these are basic questions... but I feel like I'm so close to getting this working... it must be something simple

Thanks,

david,

Beersatron, 2019-07-31:
You might need GraphicsMagick? https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/utils/README.md

Beersatron, 2019-07-31:
And you definitely need to run as root.

dkyy, 2019-07-31:
OK... I'm able to run as root and get past the ImportError. But I still am getting a Segmentation fault when I call RGBMatrix.

I found that I needed to run Python as follows to make sure all my environment variables were carried over...

`sudo -HE env PATH=$PATH PYTHONPATH=$PYTHONPATH python3`

But the Segmentation fault?

dkyy, 2019-07-31:
Ok - I think I solved most of it. It was a total newbie mistake -- I had renamed some variables and was calling:

`rgb_matrix = RGBMatrix(rgb_options = rgb_options)`
when I should have called it like this:
`rgb_matrix = RGBMatrix(options = rgb_options)`

So now things are looking good. And I'm able to run almost all of the demos in the samples directory. BUT - for some reason image-viewer.py doesn't work. I've updated options.rows and .cols to 64 (i'm using the 64x64 led matrix). But no image appears on the LED. No errors. Just no image.

petike779, 2020-06-01:
Hi all
Using all the instructions , successfully working examples on folders:
- examples-api-use folder
-utils folder
On python folder I have a problem...

I am fighting with this error like most of beginners I suppose:
$ sudo ./simple-square.py
Traceback (most recent call last):
  File "./simple-square.py", line 2, in <module>
    from samplebase import SampleBase
  File "/home/pi/display16x32/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
ImportError: cannot import name 'RGBMatrixOptions' from 'rgbmatrix' (/usr/local/lib/python3.7/dist-packages/rgbmatrix/__init__.py)**

Steps done as in readme instructions:

$ sudo apt-get update && sudo apt-get install python3-dev python3-pillow -y
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'python3-pil' instead of 'python3-pillow'
python3-dev is already the newest version (3.7.3-1).
python3-pil is already the newest version (5.4.1-2+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 32 not upgraded.

$ make build-python PYTHON=$(which python3)
make -C ../../lib
make[1]: Entering directory '/home/pi/display16x32/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/home/pi/display16x32/rpi-rgb-led-matrix/lib'
/usr/bin/python3 setup.py build --build-lib .
running build
running build_py
running build_ext

$ sudo make install-python PYTHON=$(which python3)
/usr/bin/python3 setup.py install
running install
running build
running build_py
running build_ext
running install_lib
running install_egg_info
Removing /usr/local/lib/python3.7/dist-packages/rgbmatrix-0.0.1.egg-info
Writing /usr/local/lib/python3.7/dist-packages/rgbmatrix-0.0.1.egg-info

And here I cannot do nothing... Tried also a clean reinstall with downloading again with git, doing all the steps again. I tried also with python 2.7 usage, the same error.
Reading other comments I realize that I am missing simple  step...
Thanks in advance
Peter